espressif/vscode-esp-idf-extension #695: [Bug Report]: 'idf.py.exe' is not recognized as an internal or external command, operable program or batch file issue

Nistp, 2022-04-13:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [Yes] Make sure you have searched for existing bugs and features request before you post an issue.
- [Yes] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [Yes] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I can't get the VSC extension to work with ESP stuff., I'm getting the same errors as in the closed #386 issue. I follow the instructions on GitHub/video and I get errors saying that I'm missing the tools needed to compile or run idf.py commands. I have had ESP-IDF 4.3/4.4 running some time ago, but I used the ESP-IDF 4.3 CLI.

**To Reproduce**
Install the extension and copy an example to a different folder.

**Expected behavior**
The tools are found and I can compile and flash the example onto the board.

**Environment (please complete the following information):**

- OS Name & Version: Windows 10
- VSCode Version: 1.66.2
- ESP-IDF Version: 4.4
- Python Version: the one that comes with the extension

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**ESP-IDF Extension for Visual Studio Code report**
```
OS win32 x64 10.0.19043 
System environment variable IDF_PYTHON_ENV_PATH 
 C:\Users\Nistp\.espressif\python_env\idf4.4_py3.8_env\Scripts 
System environment variable PATH 
 C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files\OpenCV\opencv\build\x64\vc14\bin;C:\Windows\SysWOW64;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\WINDOWS\System32\OpenSSH\;M:\cygwin64\bin;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\CMake\bin;C:\Program Files\Git\cmd;C:\Program Files\usbipd-win\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\National Instruments\Shared\LabVIEW CLI;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;;C:\Program Files\Microsoft VS Code\bin;C:\Users\Nistp\.espressif/tools/xtensa-esp32-elf-clang/esp-12.0.1-20210914-x86_64-pc-windows-msvc/bin/;M:\Toolchains\Arm\11.2 2022.02\bin;C:\Users\Nistp\.cargo\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Users\Nistp\AppData\Local\Programs\Python\Python37\Scripts\;C:\Users\Nistp\AppData\Local\Programs\Python\Python37\;C:\Users\Nistp\AppData\Local\Microsoft\WindowsApps;M:\cygwin64\bin;C:\Users\Nistp\AppData\Local\GitHubDesktop\bin;M:\Intel\ModelSim\modelsim_ase\win32aloem;C:\Program Files\OpenCV\opencv\build\x64\vc14\bin;C:\Users\Nistp\AppData\Local\Microsoft\WindowsApps;";C:\Users\Nistp\Desktop\esp-idf\tools" 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.66.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\Nistp\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\Nistp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin;C:\Users\Nistp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\Nistp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\Nistp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin;C:\Users\Nistp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Nistp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Nistp\.espressif\tools\cmake\3.20.3\bin;C:\Users\Nistp\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Users\Nistp\.espressif\tools\ninja\1.10.2;C:\Users\Nistp\.espressif\tools\idf-exe\1.0.3;C:\Users\Nistp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\Nistp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Nistp\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\Nistp\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\Nistp\.espressif
Git Path (idf.gitPath) C:\Users\Nistp\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\Nistp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\Nistp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\Nistp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\Nistp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\Nistp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\Nistp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\Nistp\.espressif\tools\cmake\3.20.3\bin: true
Access to C:\Users\Nistp\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:\Users\Nistp\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\Nistp\.espressif\tools\idf-exe\1.0.3: true
Access to C:\Users\Nistp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Users\Nistp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... 157 lines cut ...]
{"message":"Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (node:child_process:406:12)\n\tat ChildProcess.emit (node:events:390:28)\n\tat maybeClose (node:internal/child_process:1064:16)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)","level":"error","timestamp":"2022-04-13T02:14:30.269Z"}
{"message":"Error checking xtensa-esp32-elf version","stack":"Error: Command failed: xtensa-esp32-elf-gcc --version\n'xtensa-esp32-elf-gcc' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n\n\tat ChildProcess.exithandler (node:child_process:406:12)\n\tat ChildProcess.emit (node:events:390:28)\n\tat maybeClose (node:internal/child_process:1064:16)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:301:5)","level":"error","timestamp":"2022-04-13T02:14:30.270Z"}
{"message":"Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal or external command,\r\noperable program or batch file.\r\n","stack":"Error: Command failed: esp32s2ulp-elf-as --version\n'esp32s2ulp-elf-as' is not recognized as an internal o [... cut ...] espressif\\tools\\ninja\\1.10.2;C:\\Users\\Nistp\\.espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\Nistp\\.espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Users\\Nistp\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\Nistp\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}"

Had to trim down the log as this post going over the char limit.

brianignacio5, 2022-04-13:
Based on the doctor command output the extension seems to be configured correctly. What is the build output ?

Nistp, 2022-04-14:
@brianignacio5 Hi! Thank you for your help.

I'm a bit concerned about the following in the doctor command:

```
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
```

I don't even have to start building for that output to appear, it happens between steps 3 and 4 of the basic tutorial. Whenever I try to make sure the project is targeted for the right platform, I get the following output:

```
[Error: non zero exit code 2
To use idf.py, either the 'ninja' or 'GNU make' build tool must be available in the PATH
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\users\nistp\documents\esp stuff\bleeweeclockesp\bleprph_wifi_coex\build' is empty. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.

	at ChildProcess.<anonymous> (\dist\extension.js:1:4499)
	at ChildProcess.emit (node:events:390:28)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)]
```

If I do try building after this by pressing on the icon next to the garbage can on the bottom left, I get the following:

`Something went wrong while trying to build the project'

brianignacio5, 2022-04-15:
Could you try a folder that doesn't contain spaces? We don't support path spaces in the extension yet.

Nistp, 2022-04-16:
@brianignacio5 yeah, I just renamed the "ESP Stuff" to "ESP_Stuff" and the issue persists. 

When trying to target the example to esp32 ->
```
[Error: non zero exit code 2

To use idf.py, either the 'ninja' or 'GNU make' build tool must be available in the PATH
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\users\nistp\documents\esp_stuff\bleeweeclockesp\bleprph_wifi_coex\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.

	at ChildProcess.<anonymous> (\dist\extension.js:1:4499)
	at ChildProcess.emit (node:events:390:28)
	at Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)]
```

brianignacio5, 2022-04-20:
I can't seem to reproduce your issue locally.

Could you share any related error in `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log` ?

By the way i noticed your env PATH has some characters that might break the parsing `...";C:\Users\Nistp\Desktop\esp-idf\tools"` Is this correct ?

Nistp, 2022-04-23:
@brianignacio5 no worries, the errors that I see in that log I've included with my opening post.

Yes, PATH does include ';' to separate values, that seems to be a common thing in Windows. I'm thinking that I have "PATH" as a system env. variable and then I also have a user-defined env. variable "Path" which includes some of the ESP-IDF stuff (maybe from the previous installations)? Could that be an issue? I might have to investigate that further on my end.

brianignacio5, 2022-04-24:
We've made some changes in the extension  related to python packages. Probably you can try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/216432431.zip) ?

The log might give some indication where the issue is, also there might  be a conflicted cache in  the build  folder already. I would suggest delete  the build folder and make sure that  python packages are set.

Nistp, 2022-04-30:
@brianignacio5 uninstalling the extension in VSC and then installing via the vsix installer resulted in the same thing. It seems to be my own configuration. I'll have to check out the path. Maybe I'll use the official ESP-IDF installer and see if I can get VSC to recognize it on my desktop.

guivo, 2022-05-01:
I would like to add that I appear to be having the same issue. I can add that everything was ok with the SDK 4.3.1, after update to 4.4.1 nothing works.

All tools don't appear in the paths and the extensions cannot do anything.

brianignacio5, 2022-05-03:
What is the result of the `ESP-IDF: Doctor command` ? What errors do you see ?

Check the readme's troubleshooting section to understand your errors.

guivo, 2022-05-13:
Hi @brianignacio5  the doctor output will follow.

The project path doensn't contain spaces

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python38\Scripts\;C:\Python38\;C:\Program Files\Microsoft MPI\Bin\;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Microsoft\Web Platform Installer\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\dotnet\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\ProgramData\chocolatey\bin;C:\Program Files\TortoiseSVN\bin;C:\Program Files\Go\bin;C:\Program Files\Git\cmd;C:\Program Files (x86)\dotnet\;C:\Program Files\nodejs\;C:\Program Files\PowerShell\7\;C:\Users\lugbu\AppData\Local\Microsoft\WindowsApps;C:\Program Files\Microsoft VS Code\bin;C:\Users\lugbu\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\heroku\bin;C:\Users\lugbu\AppData\Local\Google\Cloud SDK\google-cloud-sdk\bin;C:\Users\lugbu\.dotnet\tools;C:\Users\lugbu\.dotnet\tools;C:\Users\lugbu\AppData\Local\Microsoft\WindowsApps;C:\Users\lugbu\go\bin;C:\Users\lugbu\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.66.2 
Visual Studio Code language it 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\lugbu\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\lugbu\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Users\lugbu\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\lugbu\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\lugbu\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\lugbu\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\lugbu\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\lugbu\.espressif\tools\cmake\3.20.3\bin;C:\Users\lugbu\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Users\lugbu\.espressif\tools\ninja\1.10.2;C:\Users\lugbu\.espressif\tools\idf-exe\1.0.3;C:\Users\lugbu\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\lugbu\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\lugbu\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\lugbu\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\lugbu\.espressif
Git Path (idf.gitPath) C:\Users\lugbu\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\lugbu\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\lugbu\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\lugbu\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\lugbu\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\lugbu\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\lugbu\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\lugbu\.espressif\tools\cmake\3.20.3\bin: true
Access to C:\Users\lugbu\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:\Users\lugbu\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\lugbu\.espressif\tools\idf-exe\1.0.3: true
Access to C:\Users\lugbu\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Users\lugbu\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
ESP-IDF version 4.3.2
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
[... cut ...]
Check Debug AdapterPython packages Python requirements from c:\Users\lugbu\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\Users\lugbu\Documents\Sviluppo\IIIMediaEnigma\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

guivo, 2022-05-13:
after yet another removal and reinstallation the compilation seems to be working.

londonlove, 2022-05-15:
> 在再次删除和重新安装之后，编译似乎正在工作。

I have the same problem too. I upgrade my idf version from v4.4.0 to v4.4.1, and reinstall the tools, but the problem is also exis.

brianignacio5, 2022-05-26:
Closing this issue since there has not been a response from the original user in a long time and can't reproduce the issue.

closed 2022-05-26 by brianignacio5

Nistp, 2023-05-30:
Believe it or not, I'm still having the same issue...

I came back to the VSC and tried to install the Espressif IDF 5.02 thru the extension and I'm getting the same errors. Can we try this again? I know it works, since once I manually run the export.bat file, I can use idf-py thru the CLI. I'd at least like to be able to do that with VSC.

Here is my `Doctor command` report:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\OpenSSH\;C:\Windows\SysWOW64;C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files\OpenCV\opencv\build\x64\vc14\bin;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Program Files\usbipd-win\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\Microsoft VS Code\bin;C:\mozilla-build\;C:\Program Files\TortoiseHg\;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\WireGuard\;C:\CMake\bin;C:<HOMEPATH>\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:\texlive\2022\bin\win32;M:\Toolchains\Arm\11.2 2022.02\bin;C:<HOMEPATH>\.cargo\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;M:\Intel\ModelSim\modelsim_ase\win32aloem;C:\Program Files\OpenCV\opencv\build\x64\vc14\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\AppData\Local\Programs\Julia-1.8.5\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.78.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.3 
Workspace folder c:<HOMEPATH>\Documents\Projects\ESP32Projects\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:\Espressif\frameworks\esp-idf-v5.0.2
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:\Espressif\tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;c:\Espressif\tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;c:\Espressif\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;c:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;c:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;c:\Espressif\tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;c:\Espressif\tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:\Espressif\tools\tools\cmake\3.24.0\bin;c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;c:\Espressif\tools\tools\ninja\1.10.2;c:\Espressif\tools\tools\idf-exe\1.0.3;c:\Espressif\tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;c:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64;c:\Espressif\tools\tools\esp-rom-elfs\20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:\Espressif\tools\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
    ESP_ROM_ELF_DIR: c:\Espressif\tools\tools\esp-rom-elfs\20220823/
Virtual env Python Path (idf.pythonBinPath) c:\Espressif\tools\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM3
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\Espressif\tools
Git Path (idf.gitPath) c:\Espressif\tools\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:\Espressif\tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to c:\Espressif\tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to c:\Espressif\tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
[... cut ...]
--------------------------------------------------------------------------------------------------------------------------------------------
```

It seems like the issue is that I don't have IDF_PYTHON_ENV_PATH or PYTHON in the system env. variables; however, how do I get them there? I added IDF_PYTHON_ENV_PATH and ran the installation again and then the Doctor command again and nothing changes. I also have Python installed already, aside from what gets installed by ESP's installer, so...

Any advice?