espressif/vscode-esp-idf-extension #697: [Bug Report]: Unable to continue after installation of framework was interrupted，info:".../esp-idf already exists. Delete it or use another location. (ERROR_EXISTING_ESP_IDF)"

xiaguangbo, 2022-04-14:
info:".../esp-idf already exists. Delete it or use another location. (ERROR_EXISTING_ESP_IDF)"

There are many reasons for the installation process to be interrupted. For example, a python module is not installed, for example, 'click' cannot be found.
'/.espressif' directory will not cause similar errors
This causes me to re-download esp-idf every time, but my network is very poor, and many times I can't download it all at once, and the experience is extremely poor
The core function of this plugin should be to manage the toolchain, and downloading 'esp-idf' should not block the framework installation process

# about
vscode v1.66.2
plugin 'Espressif IDF' v1.4.0

brianignacio5, 2022-04-15:
There is an option in the setup wizard to use an existing esp-idf directory and a `Install ESP-IDF Python Packages` command to  just install python packages in the configured virtual environment.

xiaguangbo, 2022-04-15:
@brianignacio5 It always times out when python downloads the module and stops, but after several downloads, the toolchain is successfully downloaded, and the environment is successfully configured. But it is really a headache to download esp-idf repeatedly in the process, it is obviously there

ankostis, 2022-04-15:
@brianignacio5 which particular section is important to avoid git-cloning _esp_idf_ repo again?

xiaguangbo, 2022-04-16:
@brianignacio5 I think the logic of the current configuration environment is a bit strange. The options for providing local configuration should be displayed directly, rather than need what conditions are met. Now, I have to cooperate with the plugin to do some work that I don't need to do. I should be able to specify the location of 'esp-idf' and '.espressif' directly in the plugin.

xiaguangbo, 2022-04-16:
@brianignacio5 in the drop-down box of the selected version, use the local configuration 😆

closed 2022-04-16 by xiaguangbo