espressif/vscode-esp-idf-extension #699: [Bug Report]: When the plugin uses 'esp-idf master', there is a problem with the monitoring function, but 'esp-idf v4.4' is fine

xiaguangbo, 2022-04-18:
Hi
When the plugin uses 'esp-idf master', there is a problem with the monitoring function, but 'esp-idf v4.4' is fine
I've tried back and forth several times
I don't know if the plugin version has not kept up or there is a problem with esp-idf...

**Info**

```
PS C:\Users\xiaguangbo\esp\project\sample_project> C:\Users\xiaguangbo\esp\master\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\xiaguangbo\esp\master\esp-idf\tools\idf.py -p COM6 monitor
ERROR: C:\Users\xiaguangbo\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

ESP-IDF v5.0-dev-2483-g3aeb80acb6-dirty
```

**Environment**

- OS Name & Version: win11
- VSCode Version: 1.66.2
- ESP-IDF Version: 1.4
- Python Version: 3.8.7

brianignacio5, 2022-04-19:
Sorry for late reply @xiaguangbo I think i found the issue.

The extension was not following the constraints given by ESP-IDF. This should be fixed in #700 and this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/216432431.zip).

xiaguangbo, 2022-04-19:
@brianignacio5 thanks

PR #700 (2022-04-21): use constrainst in py reqs install
Use the constraints file in `ESP-IDF v5.0`  python packages install.

Fix #688
Fix #699
closed 2022-04-19 by xiaguangbo