espressif/vscode-esp-idf-extension #707: [Bug Report]:

Steven35700, 2022-04-26:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**

Unable to flash a program when enabling memory security.

**To Reproduce**

Follow [the tutorial](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html#flash-enc-development-mode) to protect your flash memory in dev mode using ESP32 Generated Key.
In step 3, go directly through the VScode application to flash the firmware.

**Expected behavior**

The program should be flash but an error appears.
**By going directly through the CLI ESP-IDF, it is possible to flash the program (idf.py flash)**

**Error**

````
> Executing task: C:\Users\Steven\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe d:\ESP-IDF\esp-idf\components\esptool_py\esptool\esptool.py -p COM7 -b 460800 --before default_reset --after no_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0xf000 partition_table/partition-table.bin --encrypt-files 0x20000 main.bin 0x1000 bootloader/bootloader.bin <

esptool.py v3.2-dev
Serial port COM7
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:c9:a3:c5:78:64
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Unexpected FLASH_CRYPT_CONFIG value: 0x0
Flash encryption key is not programmed

A fatal error occurred: Can't perform encrypted flash write, consult Flash Encryption documentation for more information
The terminal process "C:\Users\Steven\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe 'd:\ESP-IDF\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM7', '-b', '460800', '--before', 'default_reset', '--after', 'no_reset', '--chip', 'esp32', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '40m', '--flash_size', 'detect', '0xf000', 'partition_table/partition-table.bin', '--encrypt-files', '0x20000', 'main.bin', '0x1000', 'bootloader/bootloader.bin'" terminated with exit code: 2.
````

**"Unexpected FLASH_CRYPT_CONFIG value: 0x0" but this value will be modified by the bootloader after boot**
**"Flash encryption key is not programmed" but I would like to use the key generated by the ESP32!**

**Environment (please complete the following information):**

- OS Name & Version: Windows
- VSCode Version: 1.66.2
- ESP-IDF Version: 4.4
- Python Version: 3.9.4

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\Git\cmd;C:\Program Files\Amazon\AWSCLIV2\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\CMake\bin;C:\Users\Steven\AppData\Local\Programs\Python\Python39\Scripts\;C:\Users\Steven\AppData\Local\Programs\Python\Python39\;C:\Users\Steven\AppData\Local\Microsoft\WindowsApps;D:\VisualStudio\Microsoft VS Code\binD:\Espressif\tools\amazon-corretto-11-x64-windows-jdk\jdk11.0.14_10\bin\;;D:\VisualStudio\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.66.2 
Visual Studio Code language en 
[... 146 lines cut ...]
{
    "workbench.colorTheme": "Monokai Dimmed",
    "python.defaultInterpreterPath": "C:\\Users\\Steven\\AppData\\Local\\Programs\\Python\\Python39\\python.exe",
    "git.enableSmartCommit": true,
    "workbench.editorAssociations": {
        "*.kv": "default"
    },
    "idf.gitPath": "D:/Espressif/tools/idf-git/2.34.2/cmd/git.exe",
    "idf.espIdfPathWin": "d:\\ESP-IDF\\esp-idf",
    "idf.pythonBinPathWin": "C:\\Users\\Steven\\.espressif\\python_env\\idf4.4_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "C:\\Users\\Steven\\.espressif",
    "idf.customExtraPaths": "C:\\Users\\Steven\\.espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32-elf\\bin;C:\\Users\\Steven\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Users\\Steven\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch2-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Users\\Steven\\.espressif\\tools\\riscv32-esp-elf\\esp-2021r2-patch2-8.4.0\\riscv32-esp-elf\\bin;C:\\Users\\Steven\\.espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Users\\Steven\\.espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Users\\Steven\\.espressif\\tools\\cmake\\3.20.3\\bin;C:\\Users\\Steven\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220\\openocd-esp32\\bin;C:\\Users\\Steven\\.espressif\\tools\\ninja\\1.10.2;C:\\Users\\Steven\\.espressif\\tools\\idf-exe\\1.0.3;C:\\Users\\Steven\\.espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Users\\Steven\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Users\\\\Steven\\\\.espressif\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.adapterTargetName": "",
    "idf.portWin": "COM7",
    "idf.openOcdConfigs": [
        "interface/ftdi/esp32_devkitj_v1.cfg",
        "target/esp32.cfg"
    ],
    "idf.flashType": "UART",
    "cmake.configureOnOpen": true,
    "json.schemas": [],
    "aws.profile": "profile:default"
}
```

aravindtn, 2022-05-22:
I see the same issue on eclipse-ide too. Is there any solution or work around identified?

aravindtn, 2022-05-22:
Here is my espefuse.py output:
PS C:\Espressif\frameworks\esp-idf-v4.4> espefuse.py -p COM4 summary
Connecting......................................

A fatal error occurred: Failed to connect to Espressif device: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
PS C:\Espressif\frameworks\esp-idf-v4.4> espefuse.py -p COM4 summary
Connecting........................
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP32
espefuse.py v3.2-dev
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1149 R/W (0b00111)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 51 R/W (0x33)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 0 R/W (0x0000)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 0 R/W (0x0)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address
   = 70:b8:f6:90:61:2c (CRC 0xa7 OK) R/W
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 167 R/W (0xa7)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                             Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 0 R/W (0b0000000)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 0 R/W (0x0)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = False R/W (0b0)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = False R/W (0b0)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = False R/W (0b0)
BLOCK1 (BLOCK1):                                   Flash encryption key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK2 (BLOCK2):                                   Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                                   Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

aravindtn, 2022-05-22:
Since the FLASH_CRYPT_CNT is 0, the expectation, is that the encryption key will be generated and increment the FLASH_CRYPT_CNT to 1, but getting the following error in eclipse-ide when tried to flash

Configuring flash size...
Unexpected FLASH_CRYPT_CONFIG value: 0x0
Flash encryption key is not programmed

A fatal error occurred: Can't perform encrypted flash write, consult Flash Encryption documentation for more information
CMake Error at run_serial_tool.cmake:56 (message):
  python
  C:/Espressif/frameworks/esp-idf-v4.4/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

aravindtn, 2022-05-22:
@Steven35700 
This may be because for very first time while flashing the encrypted image, it should be flashed with "flash" command and not with "encrypted-flash". For subsequent flashing "encrypted-flash" should be used.

PR #719 (2022-05-26): add encrypt and refactor to single flash method
Fix #711 
Fix #707
Fix #709

Add `ESP-IDF: Encrypt and flash your project` command to separate from regular flash commands. This Encrypt and Flash uses esptool so it only works with UART.

Update template devcontainer.json and Dockerfile to run in privileged mode.

Refactor and simplify flash method.
closed 2022-05-26 by brianignacio5