espressif/vscode-esp-idf-extension #710: Select openOCD board configuration missing esp32-s3 usb-jtag option.... this was painful to get working

chmorgan, 2022-05-02:
esp-box dev kit here.

Ran through the installation steps, was able to program via openocd but kept getting invalid target errors, I think because I was using the esp32c3 openocd configuration as that was the closest one to the esp32s3 config with built in usb-jtag.

Started going through the bug reporting process and pulling variables and then, after verifying that the appropriate board file existed changed these entries from:

```
    "idf.openOcdConfigs": [
        "board/esp32c3-builtin.cfg"
    ],
    "idf.adapterTargetName": "esp32c3",
```

to:

```
    "idf.openOcdConfigs": [
        "board/esp32s3-builtin.cfg"
    ],
    "idf.adapterTargetName": "esp32s3",
```

then restarted vscode and its working, so I think this is what fixed it.

Is there a reason there is no esp32-s3 usb-jtag option for the openocd board config?

chmorgan, 2022-05-18:
Should we be looking to contribute the new entry to the plugin? Please don't mark this as stale :-(

brianignacio5, 2022-05-26:
Sorry for late reply.

The OpenOCD configuration files comes from a file in OpenOCD Scripts directory called `esp-config.json`. **Could you specify which version are  you using?**

Also what are you using to set the openOCD board value? The `ESP-IDF: Select OpenOCD Board Configuration` would use the previous file to show the boards for a given openOCD version, or use  a default set of boards found in the extension which should include the esp32s3.

closed 2022-05-26 by brianignacio5

dariomingarelli, 2022-10-12:
Hi. I'm using Visual Studio Code V1.72.1 (Universal) with Espressif IDF extension V1.5.0 and ESP-IDF V4.4.2.
I'm trying to debug on ESP32-S3-DevKitC-1 using the USB-JTAG (NO FTDI) of the ESP32-S3, but there is still no configuration in EDF-IDF.

I've tried 

`    "idf.openOcdConfigs": [
        "board/esp32s3-builtin.cfg"
    ],
    "idf.adapterTargetName": "esp32s3",`

with no success... any update?
I've also tried ESP-IDF V5 (beta) with no success and also I've tried Eclipse IDE with Espressif IDF extension with no success.
Do you plan to release the configuration to DEBUG code using the onboard ESP32-S3 USB-JTAG (NO FTDI)?

Thanks.

brianignacio5, 2022-10-13:
It should already work with built-in usb JTAG.

Could you share the [troubleshooting doc](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) What is happening when you start the debug session. What is the openOCD and Debug Adapter output? You might need to increment the logLevel of them to see the error.

PR #724 (2022-05-26): add built in esp32s3 default board
Fix #710