espressif/vscode-esp-idf-extension #712: [Bug Report]: Install ADF command won't pull submodules of adf.

RuofengX, 2022-05-04:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**

1. Using command ctrl+shift+p and run `ESP-ADF: Install ESP-ADF`
2. Use default setting, install adf under current IDF workfolder.
3. ADF projects' submodule won't get downloaded. This causes include error when running adf's example.
4. You could double check the .gitmodules file in adf path, all modules won't be downloaded.

**Expected behavior**

All of ADF's submodule **except ESP-IDF** will get init by git command.

**Environment (please complete the following information):**

- OS Name & Version: Windows 1
- VSCode Version:  1.66.2
- ESP-IDF Version: v1.4.0
- Python Version: 3.7

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 C:\Users\11986\AppData\Local\Programs\Python\Python37\python.exe 
System environment variable PATH 
 C:\Users\11986\AppData\Local\Programs\Python\Python37\Scipts;C:\Users\11986\AppData\Local\Programs\Python\Python37;C:\Program Files\Microsoft\jdk-17.0.1.12-hotspot\bin;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v11.0\libnvvp;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\PuTTY\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\GtkSharp\2.12\bin;C:\Program Files\NVIDIA Corporation\Nsight Compute 2020.1.1\;C:\Program Files\WorldPainter;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files\CMake\bin;C:\Program Files\dotnet\;C:\Program Files (x86)\Yarn\bin\;C:\Program Files\nodejs\;C:\Android;C:\Windows\System32;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\PowerShell\7\;C:\Program Files\usbipd-win\;C:\Program Files\Git\cmd;.;C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.20.3\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Users\11986\AppData\Local\Programs\Python\Python37\Scripts; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.66.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) C:/Espressif/esp-adf
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v4.4.1/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\cmake\3.20.3\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM8
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:\Espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) true
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\cmake\3.20.3\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... 43 lines cut ...]
pywin32 version: 303
PyYAML version: 6.0
reedsolo version: 1.5.4
requests version: 2.27.1
requests-toolbelt version: 0.9.1
schema version: 0.7.5
semantic-version version: 2.9.0
setuptools version: 62.1.0
six version: 1.16.0
tqdm version: 4.64.0
urllib3 version: 1.26.9
websocket-client version: 1.3.2
Werkzeug version: 0.16.1
wheel version: 0.36.2
windows-curses version: 2.3.0
xmlrunner version: 1.7.7
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.1\requirements.txt are satisfied.
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Python requirements from c:\Users\11986\.vscode\extensions\espressif.esp-idf-extension-1.4.0\requirements.txt are satisfied.
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements from c:\Users\11986\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\Espressif\esp-adf\examples\get-started\pipeline_tcp_client\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

closed 2022-05-25 by github-actions[bot]

brianignacio5, 2022-05-25:
Could you share the output of the `Install ESP-ADF` command ?

The executed code should be `${idf.gitPath} clone --recursive --progress -b master https://github.com/espressif/esp-adf.git` and the output should be shown in the ESP-IDF Output channel and the extension log.

Check log file which can be obtained from:

- Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
- Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

RuofengX, 2022-05-25:
Problem somehow disappeared, when I watching the output:
```
......
Cloning into 'C:/Espressif/esp-adf/esp-adf/esp-idf'...
remote: Enumerating objects: 332989, done.        
remote: Counting objects:   0% (1/1633)        
remote: Counting objects:   1% (17/1633)        
remote: Counting objects:   2% (33/1633)        
remote: Counting objects:   3% (49/1633)        
......
```

RuofengX, 2022-05-25:
Close