espressif/vscode-esp-idf-extension #713: [Bug Report]:IDF extension sdkconfigeditor slow start

arilink-tech, 2022-05-10:
IDF-Version  V4.4
ESP-idf-extension version v1.4
When trying to open sdkconfig with the vscode idf extension plugin, the startup is very slow and it has not been opened for 5 minutes. What is the problem?

brianignacio5, 2022-05-11:
There might be an error in your settings which makes the SDK Configuration editor confserver.py process fails.

Could you share the extension log and the doctor command output as described in the [troubleshooting section](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) or bug report template ?

HiFiPhile, 2022-05-27:
Hi,
I think I've the same issue.

My configuration:
esp-idf version: git master
IDF_TOOLS_PATH: `c:\Tool\esp-idf`
IDF_PATH: `D:\mcu\esp-idf`
`Use existing ESP-IDF Tools` is selected in advanced installation.

Observations:
- esp-idf already installed a python venv into `C:\Tool\esp-idf\python_env\idf5.0_py3.9_env`, this extension should use existing venv instead create a new `idf5.0_py3.8_env` venv.
- Form the log wrong version of Click installed.
- If I change Click version `C:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts>.\python.exe -m pip install Click==8.0` then menuconfig is successful.

Error log:
```
{"message":"Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:48)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:1:7563\n\tat ChildProcess.exithandler (node:child_process:397:7)\n\tat ChildProcess.emit (node:events:390:28)\n\tat maybeClose (node:internal/child_process:1064:16)\n\tat Socket.<anonymous> (node:internal/child_process:450:11)\n\tat Socket.emit (node:events:390:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2022-05-27T20:42:31.775Z"}
{"message":"The following Python requirements are not satisfied:\r\r\nclick<8.1,>=7.0 - (click 8.1.3 (c:\\tool\\esp-idf\\python_env\\idf5.0_py3.8_env\\lib\\site-packages), Requirement.parse('click<8.1,>=7.0'))\r\r\nTo install the missing packages, please run \"install.bat\"\r\r\nDiagnostic information:\r\r\n    IDF_PYTHON_ENV_PATH: c:\\Tool\\esp-idf\\python_env\\idf5.0_py3.8_env\r\r\n    Python interpreter used: C:\\Tool\\esp-idf\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe\r\r\n    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH\r\r\n    PATH: D:\\mcu\\esp-idf\\components\\esptool_py\\esptool;D:\\mcu\\esp-idf\\components\\espcoredump;D:\\mcu\\esp-idf\\components\\partition_table;c:\\Tool\\esp-idf\\tools\\xtensa-esp-elf-gdb\\11.1_20220318\\xtensa-esp-elf-gdb\\bin;c:\\Tool\\esp-idf\\tools\\riscv32-esp-elf-gdb\\11.1_20220318\\riscv32-esp-elf-gdb\\bin;c:\\Tool\\esp-idf\\tools\\xtensa-esp32-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32-elf\\bin;c:\\Tool\\esp-idf\\tools\\xtensa-esp32s2-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32s2-elf\\bin;c:\\Tool\\esp-idf\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32s3-elf\\bin;c:\\Tool\\esp-idf\\tools\\riscv32-esp-elf\\esp-2021r2-patch3-8.4.0\\riscv32-esp-elf\\bin;c:\\Tool\\esp-idf\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;c:\\Tool\\esp-idf\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;c:\\Tool\\esp-idf\\tools\\cmake\\3.23.1\\bin;c:\\Tool\\esp-idf\\tools\\openocd-esp32\\v0.11.0-esp32-20220411\\openocd-esp32\\bin;c:\\Tool\\esp-idf\\tools\\ninja\\1.10.2;c:\\Tool\\esp-idf\\tools\\idf-exe\\1.0.3;c:\\Tool\\esp-idf\\tools\\ccache\\4.3\\ccache-4.3-windows-64;c:\\Tool\\esp-idf\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64;c:\\Tool\\esp-idf\\python_env\\idf5.0_py3.8_env\\Scripts;D:\\mcu\\esp-idf\\tools;c:\\Tool\\esp-idf\\tools\\idf-git\\2.30.1\\cmd;C:\\Program Files\\riscv32-esp-elf\\bin;c:\\program files (x86)\\common files\\oracle\\java\\javapath;c:\\program files (x86)\\stmicroelectronics\\st_toolset\\asm;c:\\program files (x86)\\intel\\intel(r) management engine components\\icls\\;c:\\program files\\intel\\intel(r) management engine components\\icls\\;c:\\windows\\system32;c:\\windows;c:\\windows\\system32\\wbem;c:\\windows\\system32\\windowspowershell\\v1.0\\;c:\\program files\\intel\\wifi\\bin\\;c:\\program files\\common files\\intel\\wirelesscommon\\;c:\\program files (x86)\\intel\\intel(r) management engine components\\dal;c:\\program files\\intel\\intel(r) management engine components\\dal;c:\\program files (x86)\\intel\\intel(r) management engine components\\ipt;c:\\program files\\intel\\intel(r) management engine components\\ipt;c:\\windows\\system32\\openssh\\;C:\\Program Files\\Python39;c:\\program files\\git\\cmd;c:\\tool;c:\\program files\\dotnet\\;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\Wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;C:\\WINDOWS\\System32\\OpenSSH\\;C:\\Program Files\\TortoiseSVN\\bin;C:\\Program Files (x86)\\dotnet\\;C:\\Program Files\\PuTTY\\;C:\\Go\\bin;C:\\Python27;C:\\Program Files (x86)\\Microsoft SQL Server\\150\\DTS\\Binn\\;C:\\Program Files\\CMake\\bin;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\msys64\\mingw64\\bin;C:\\Program Files\\TortoiseGit\\bin;C:\\Program Files\\Git\\cmd;C:\\Program Files (x86)\\IVI Foundation\\VISA\\WinNT\\Bin\\;C:\\Program Files\\IVI Foundation\\VISA\\Win64\\Bin\\;C:\\Program Files (x86)\\IVI Foundation\\VISA\\WinNT\\Bin;C:\\Program Files\\tacoscript\\bin;C:\\Program Files (x86)\\Windows Kits\\10\\Windows Performance Toolkit\\;C:\\msys64\\usr\\bin;C:\\lscc\\iCEcube2.2017.08\\Aldec\\Active-HDL\\BIN;E:\\rust\\cargo\\bin;C:\\Program Files (x86)\\Atmel\\sam-ba_2.18\\drv\\;C:\\Program Files (x86)\\Atmel\\sam-ba_2.18;C:\\Users\\zli\\AppData\\Local\\Microsoft\\WindowsApps;C:\\Program Files (x86)\\COSMIC\\FSE_Compilers\\CXSTM8;C:\\Program Files (x86)\\COSMIC\\FSE_Compilers\\CXSTM32;;C:\\Users\\zli\\AppData\\Local\\Programs\\Microsoft VS Code\\bin;C:\\Users\\zli\\.dotnet\\tools;C:\\Users\\zli\\go\\bin;C:\\Users\\zli\\AppData\\Local\ [... cut ...] error","timestamp":"2022-05-27T20:43:46.644Z"}
{"message":"SDK Configuration editor confserver process exited with code: 1","stack":"Error: SDK Configuration editor confserver process exited with code: 1\n\tat m.printError (\\dist\\extension.js:1:30492)\n\tat ChildProcess.<anonymous> (\\dist\\extension.js:1:30098)\n\tat ChildProcess.emit (node:events:390:28)\n\tat Process.ChildProcess._handle.onexit (node:internal/child_process:290:12)","level":"error","timestamp":"2022-05-27T20:43:46.653Z"}
```

brianignacio5, 2022-06-06:
The issue you mentioned should be fixed in #700 which was not using the constraints file from v5.0. Please let me know if any issues persist. Will try to reuse the existing setup on the existing setup option.

Closing this issue since original user hasn't responded in a long time.

closed 2022-06-06 by brianignacio5

arilink, 2022-06-14:
Yes.  it work!   
Thankyou very much

***@***.***
 
From: Brian A. Ignacio
Date: 2022-06-06 13:16
To: espressif/vscode-esp-idf-extension
CC: arilink; Author
Subject: Re: [espressif/vscode-esp-idf-extension] [Bug Report]:IDF extension sdkconfigeditor slow start (VSC-900) (Issue #713)
The issue you mentioned should be fixed in #700 which was not using the constraints file from v5.0. Please let me know if any issues persist. Will try to reuse the existing setup on the existing setup option.
Closing this issue since original user hasn't responded in a long time.

HiFiPhile, 2022-06-15:
The issue is not resolved in my case.

I used this version from ci:
https://github.com/espressif/vscode-esp-idf-extension/actions/runs/2473536930

While initially requirement of ESP-IDF is correctly used:
```
Requirement already satisfied: setuptools in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r D:\mcu\esp-idf\tools\requirements\requirements.core.txt (line 3)) (51.3.3)
```
Then packages are overwritten while installing ESP-IDF Debug Adapter.
```
Collecting click
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Requirement already satisfied: colorama in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from click->-r c:\Users\xxx\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 3)) (0.4.4)
Installing collected packages: xmlrunner, pywin32, psutil, click
  Attempting uninstall: click
    Found existing installation: click 8.0.4
    Uninstalling click-8.0.4:
      Successfully uninstalled click-8.0.4
```

Full log :  
```
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
Using existing c:\Tool\esp-idf\dist\idf-git-2.30.1-win64.zip
Installing c:\Tool\esp-idf\tools\idf-git\2.30.1 ...
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\cmd\git.exe
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\dev\_._
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\bash.bash_logout
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\bash.bashrc
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\fstab
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\gitattributes
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\msystem
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\nsswitch.conf
                                        already exists and was not updated.
Warning File c:\Tool\esp-idf\tools\idf-git\2.30.1\etc\profile
                                        already exists and was not updated.
Extracted c:\Tool\esp-idf\tools\idf-git\2.30.1 ...
Using existing c:\Tool\esp-idf\tools\idf-python\3.8.7
Requirement already satisfied: virtualenv in c:\tool\esp-idf\tools\idf-python\3.8.7\lib\site-packages (20.4.0)
Requirement already satisfied: appdirs<2,>=1.4.3 in c:\tool\esp-idf\tools\idf-python\3.8.7\lib\site-packages (from virtualenv) (1.4.4)
Requirement already satisfied: filelock<4,>=3.0.0 in c:\tool\esp-idf\tools\idf-python\3.8.7\lib\site-packages (from virtualenv) (3.0.12)
Requirement already satisfied: distlib<1,>=0.3.1 in c:\tool\esp-idf\tools\idf-python\3.8.7\lib\site-packages (from virtualenv) (0.3.1)
Requirement already satisfied: six<2,>=1.9.0 in c:\tool\esp-idf\tools\idf-python\3.8.7\lib\site-packages (from virtualenv) (1.15.0)
WARNING: You are using pip version 20.3.3; however, version 22.1.2 is available.
You should consider upgrading via the 'c:\Tool\esp-idf\tools\idf-python\3.8.7\python.exe -m pip install --upgrade pip' command.

created virtual environment CPython3.8.7.final.0-64 in 241ms
  creator CPython3Windows(dest=C:\Tool\esp-idf\python_env\idf5.0_py3.8_env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, setuptools=bundle, wheel=bundle, via=copy, app_data_dir=C:\Users\xxx\AppData\Local\pypa\virtualenv)
    added seed packages: pip==20.3.3, setuptools==51.3.3, wheel==0.36.2
  activators BashActivator,BatchActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator

Python 3.8.7
pip 20.3.3 from c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages\pip (python 3.8)

Looking in indexes: https://pypi.org/simple, https://dl.espressif.com/pypi
Requirement already satisfied: setuptools in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r D:\mcu\esp-idf\tools\requirements\requirements.core.txt (line 3)) (51.3.3)
Collecting click
  Downloading https://dl.espressif.com/pypi/click/click-8.0.4-py3-none-any.whl (97 kB)
Collecting cryptography
  Downloading https://dl.espressif.com/pypi/cryptography/cryptography-36.0.2-cp36-abi3-win_amd64.whl (2.2 MB)
[... 114 lines cut ...]
Requirement already satisfied: requests>=2.21.0 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\xxx\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2.28.0)
Collecting psutil>=5.5.1
  Using cached psutil-5.9.1-cp38-cp38-win_amd64.whl (246 kB)
Collecting pywin32>=227
  Using cached pywin32-304-cp38-cp38-win_amd64.whl (12.3 MB)
Requirement already satisfied: idna<4,>=2.5 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\xxx\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (3.3)
Requirement already satisfied: charset-normalizer~=2.0.0 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\xxx\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requireme [... cut ...]
Installing collected packages: xmlrunner, pywin32, psutil, click
  Attempting uninstall: click
    Found existing installation: click 8.0.4
    Uninstalling click-8.0.4:
      Successfully uninstalled click-8.0.4
Successfully installed click-8.1.3 psutil-5.9.1 pywin32-304 xmlrunner-1.7.7
WARNING: You are using pip version 20.3.3; however, version 22.1.2 is available.
You should consider upgrading via the 'c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.

Python requirements has been installed
```

HiFiPhile, 2022-07-26:
@brianignacio5 This issue is still persist on HEAD:
```
[SDK Configuration Editor]
---------------------------ERROR--------------------------

The following Python requirements are not satisfied:

click<8.1,>=7.0 - (click 8.1.3 (c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages), Requirement.parse('click<8.1,>=7.0'))

To install the missing packages, please run "install.bat"

Diagnostic information:

    IDF_PYTHON_ENV_PATH: c:\Tool\esp-idf\python_env\idf5.0_py3.8_env

    Python interpreter used: c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe

Skipping the download of c:\Tool\esp-idf\espidf.constraints.v5.0.txt because it was downloaded recently.

Constraint file: c:\Tool\esp-idf\espidf.constraints.v5.0.txt

Requirement files:

 - D:\mcu\esp-idf\tools\requirements\requirements.core.txt

Python being checked: c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe

ESP-IDF v5.0-dev-4379-g36f49f361c-dirty

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 1
-----------------------END OF ERROR-----------------------
```

brianignacio5, 2022-07-26:
Could you share the doctor command output @HiFiPhile ?

You can observer there is a small difference in the `idf.pythonBinPath` (python version used is different.

SDK Configuration Editor `c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe`

Command line is using `C:\Tool\esp-idf\python_env\idf5.0_py3.9_env\Scripts\python.exe`

HiFiPhile, 2022-07-26:
[report.zip](https://github.com/espressif/vscode-esp-idf-extension/files/9189149/report.zip)

> You can observe there is a small difference in the idf.pythonBinPath (python version used is different)

Yes that's what I'm saying, esp-idf installed `idf5.0_py3.9_env` which works well, but this extension didn't use it but created `idf5.0_py3.8_env`

brianignacio5, 2022-07-26:
The extension will install embed python (version 3.8.7) by default  in `idf.toolsPathWin` and use that to create the virtual environment defined in `idf.pythonBinPathWin` configuration settings. I've just tested with current extension GitHub master and current esp-idf master and the SDK Configuration is working correctly. I did find an issue with doctor command and added a fix in #766 

You can also update the extension settings to use the same settings as the command line configuration you are using by [manual configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration).

HiFiPhile, 2022-07-27:
@brianignacio5 
I used this version https://github.com/espressif/vscode-esp-idf-extension/actions/runs/2740089050
But on anther PC doctor command still product incomplete result.
[doctor.zip](https://github.com/espressif/vscode-esp-idf-extension/files/9196389/doctor.zip)

I experienced the same thing, `click` package installed previously is overwritten while `Installing ESP-IDF Debug Adapter python packages`

In `esp_debug_adapter/requirements.txt` there is no version limit for `click` package.

```
Installing ESP-IDF Debug Adapter python packages in c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe ...

Ignoring typing: markers 'python_version < "3.5"' don't match your environment
Requirement already satisfied: setuptools>=21 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 1)) (63.2.0)
Requirement already satisfied: click in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 3)) (8.0.4)
Requirement already satisfied: psutil>=5.5.1 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 4)) (5.9.1)
Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 5)) (0.9.0.2)
Requirement already satisfied: pywin32>=227 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 6)) (304)
Requirement already satisfied: requests>=2.21.0 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2.28.1)
Requirement already satisfied: xmlrunner>=1.7.7 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 9)) (1.7.7)
Requirement already satisfied: idna<4,>=2.5 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (3.3)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (1.26.9)
Requirement already satisfied: charset-normalizer<3,>=2 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2.0.12)
Requirement already satisfied: certifi>=2017.4.17 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2022.5.18.1)
Collecting click
  Using cached click-8.1.3-py3-none-any.whl (96 kB)
Requirement already satisfied: colorama in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from click->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 3)) (0.4.4)
Installing collected packages: click
  Attempting uninstall: click
    Found existing installation: click 8.0.4
    Uninstalling click-8.0.4:
      Successfully uninstalled click-8.0.4
Successfully installed click-8.1.3
WARNING: You are using pip version 20.3.3; however, version 22.2 is available.
You should consider upgrading via the 'c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
```

HiFiPhile, 2022-07-27:
My actual workaround is use `click<=8.0.4`
https://github.com/espressif/esp-debug-adapter/blob/03244445c5edc18b8b3a57d630a22e9f09962035/requirements.txt#L3

brianignacio5, 2022-07-27:
Based on your latest doctor command output the idf.gitPath is not correct (`\Git\cmd\git.exe`). Could you update it in your settings.jsonwith the git executable full path in the .espressif directory ?

You can observe that we are using a constraint file from the .espressif directory as shown in:

https://github.com/espressif/vscode-esp-idf-extension/blob/0965aa631632029b1af056b189448dc69af0c79a/src/pythonManager.ts#L169

The click package was updated in ESP-IDF GitHub master some time ago in the constraint file generation. Which esp-idf master commit are you using ?

HiFiPhile, 2022-07-27:
> Could you update it in your settings.json with the git executable full path in the .espressif directory ?

Which `.espressif` directory you are referring ?
My `idf.toolsPathWin` is `c:\\Tool\\esp-idf` there is no `settings.json` inside.
In `C:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0` there are some `settings.json` in templates but they are unrelated.

My idf.gitPath is correct, while configurating the extension:
[image]
And in `C:\Users\zli\AppData\Roaming\Code\User\settings.json`:
```
    "idf.gitPath": "c:\\Tool\\esp-idf\\tools\\idf-git\\2.30.1\\cmd\\git.exe",
    "idf.espIdfPathWin": "D:\\mcu\\esp-idf",
    "idf.pythonBinPathWin": "c:\\Tool\\esp-idf\\python_env\\idf5.0_py3.8_env\\Scripts\\python.exe",
    "idf.toolsPathWin": "c:\\Tool\\esp-idf",
    "idf.customExtraPaths": "c:\\Tool\\esp-idf\\tools\\xtensa-esp-elf-gdb\\11.2_20220715\\xtensa-esp-elf-gdb\\bin;c:\\Tool\\esp-idf\\tools\\riscv32-esp-elf-gdb\\11.2_20220715\\riscv32-esp-elf-gdb\\bin;c:\\Tool\\esp-idf\\tools\\xtensa-esp32-elf\\esp-2022r1-RC1-11.2.0\\xtensa-esp32-elf\\bin;c:\\Tool\\esp-idf\\tools\\xtensa-esp32s2-elf\\esp-2022r1-RC1-11.2.0\\xtensa-esp32s2-elf\\bin;c:\\Tool\\esp-idf\\tools\\xtensa-esp32s3-elf\\esp-2022r1-RC1-11.2.0\\xtensa-esp32s3-elf\\bin;c:\\Tool\\esp-idf\\tools\\riscv32-esp-elf\\esp-2022r1-RC1-11.2.0\\riscv32-esp-elf\\bin;c:\\Tool\\esp-idf\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;c:\\Tool\\esp-idf\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;c:\\Tool\\esp-idf\\tools\\cmake\\3.23.1\\bin;c:\\Tool\\esp-idf\\tools\\openocd-esp32\\v0.11.0-esp32-20220706\\openocd-esp32\\bin;c:\\Tool\\esp-idf\\tools\\ninja\\1.10.2;c:\\Tool\\esp-idf\\tools\\idf-exe\\1.0.3;c:\\Tool\\esp-idf\\tools\\ccache\\4.3\\ccache-4.3-windows-64;c:\\Tool\\esp-idf\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"c:\\\\Tool\\\\esp-idf\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
    "idf.flashType": "UART",
```

> Which esp-idf master commit are you using ?

This issue exist at least from 36f49f361c001b49c538364056bc5d2d04c6f321 to b45723b92ee904c4463bf5d135243a9739cab3fe

brianignacio5, 2022-07-27:
Could you replace `"idf.gitPath"` for `"idf.gitPathWin"` ? There is a PR updating this setting because conflicts with dev containers and WSL

HiFiPhile, 2022-07-27:
I didn't see there is a `"idf.gitPathWin": "\\Git\\cmd\\git.exe"` at the end of settings.json.

So I corrected it's path by `"idf.gitPath"`

Now click package remains correct:
```
Installing ESP-IDF Debug Adapter python packages in c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe ...

Ignoring typing: markers 'python_version < "3.5"' don't match your environment
Requirement already satisfied: setuptools>=21 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 1)) (63.2.0)
Requirement already satisfied: click in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 3)) (8.0.4)
Requirement already satisfied: psutil>=5.5.1 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 4)) (5.9.1)
Requirement already satisfied: pygdbmi<=0.9.0.2 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 5)) (0.9.0.2)
Requirement already satisfied: pywin32>=227 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 6)) (304)
Requirement already satisfied: requests>=2.21.0 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2.28.1)
Requirement already satisfied: xmlrunner>=1.7.7 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from -r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 9)) (1.7.7)
Requirement already satisfied: certifi>=2017.4.17 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2022.5.18.1)
Requirement already satisfied: charset-normalizer<3,>=2 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (2.0.12)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (1.26.9)
Requirement already satisfied: idna<4,>=2.5 in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from requests>=2.21.0->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 7)) (3.3)
Requirement already satisfied: colorama in c:\tool\esp-idf\python_env\idf5.0_py3.8_env\lib\site-packages (from click->-r c:\Users\zli\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\requirements.txt (line 3)) (0.4.4)
WARNING: You are using pip version 20.3.3; however, version 22.2 is available.
You should consider upgrading via the 'c:\Tool\esp-idf\python_env\idf5.0_py3.8_env\Scripts\python.exe -m pip install --upgrade pip' command.
```

How could git path affect pip install ?