espressif/vscode-esp-idf-extension #718: [Bug Report]:

seedub158, 2022-05-20:
Example Project: HD_EEPROM
from SPI_master
from peripherals

I build successfully and have UART selected for flash method and com3 for port
I get an error when trying to flash "failed to flash because of some unusual error"
i am able to flash blink and hello world with the same settings
[image]
[image]
[image]

> Executing task: ninja  <

[1/4] cmd.exe /C "cd /D C:\Users\e729128\Documents\CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507\Espressif\Visua...nts/CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507/Espressif/VisualStudioCode/SPI/hd_eeprom/build/spi_eeprom.bin"spi_eeprom.bin binary size 0x31d00 bytes. Smallest app partition is 0x100000 bytes. 0xce300 bytes (81%) free.
[2/4] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\e729128\Documents\CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507\Espressif\VisualStudioCode\SPI\hd_eeprom\build\bootloader\esp-idf\esptool_py && C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/e729128/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/e729128/Documents/CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507/Espressif/VisualStudioCode/SPI/hd_eeprom/build/bootloader/bootloader.bin"
Bootloader binary size 0x62b0 bytes. 0xd50 bytes (12%) free.

> Executing task: C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\e729128\esp\esp-idf\tools\idf_size.py c:\Users\e729128\Documents\CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507\Espressif\VisualStudioCode\SPI\hd_eeprom\build\spi_eeprom.map <

Total sizes:
Used static DRAM:   14264 bytes ( 166472 remain, 7.9% used)
      .data size:   11936 bytes
      .bss  size:    2328 bytes
Used static IRAM:   58487 bytes (  72585 remain, 44.6% used)
      .text size:   57460 bytes
   .vectors size:    1027 bytes
Used stat D/IRAM:   72751 bytes ( 239057 remain, 23.3% used)
      .data size:   11936 bytes
      .bss  size:    2328 bytes
      .text size:   57460 bytes
   .vectors size:    1027 bytes
Used Flash size :  133475 bytes
      .text     :   95095 bytes
      .rodata   :   38124 bytes
Total image size:  274321 bytes (.bin may be padded larger)

> Executing task: C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py -p COM1 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x10000 spi_eeprom.bin 0x1000 bootloader/bootloader.bin 0x8000 partition_table/partition-table.bin <

esptool.py v3.2-dev
Serial port COM1
Traceback (most recent call last):
  File "C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 4930, in <module>
    _main()
  File "C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 4923, in _main
    main()
  File "C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 4370, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 122, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py", line 326, in __init__
    self._port = serial.serial_for_url(port)
  File "C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\__init__.py", line 90, in serial_for_url
    instance.open()
  File "C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM1': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)
The terminal process "C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe 'C:\Users\e729128\esp\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM1', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '40m', '--flash_size', 'detect', '0x10000', 'spi_eeprom.bin', '0x1000', 'bootloader/bootloader.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1.

seedub158, 2022-05-20:
I tried a cycled power and unplugged USB, replugged in and it downloaded/flashed successfully

brianignacio5, 2022-05-24:
Probably an issue with the USB connection. Closing this.