espressif/vscode-esp-idf-extension #721: [Bug Report]: "ESP-IDF Monitor device" doesn't work

xiaguangbo, 2022-05-24:
using the latest code，Windows OS

[image]

Just added some environment variables using the script
esp_full.ps1：
```
$env:IDF_PATH = "$PSScriptRoot\esp-idf"
$env:IDF_TOOLS_PATH = "$PSScriptRoot\.espressif"
$env:Path = "$PSScriptRoot\.espressif\tools\idf-python\3.8.7;$PSScriptRoot\.espressif\tools\idf-python\3.8.7\Scripts;$PSScriptRoot\.espressif\tools\idf-git\2.30.1\cmd;$PSScriptRoot\.espressif\tools\cmake\3.23.1\bin;$PSScriptRoot\.espressif\tools\ninja\1.10.2"
```

brianignacio5, 2022-05-25:
Did you configure the extension ? You shared the powrshell script you made but no information about the extension error as described in the issue template.

Could you share the troubleshooting information :

Use the `ESP-IDF: Doctor command` to generate a report of your configuration and it will be copied in your clipboard to paste anywhere.

Check log file which can be obtained from:
- Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
- Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

xiaguangbo, 2022-05-25:
@brianignacio5 

'ESP-IDF Build project' is normal

'Extension configuration report has been copied to clipboard with errors' pops up after running the command

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;D:\software\AOMEI\AOMEI Backupper\6.8.0;D:\software\node\;D:\software\Git\cmd;C:\ProgramData\chocolatey\bin;C:\Users\xiaguangbo\.cargo\bin;C:\Users\xiaguangbo\AppData\Local\Microsoft\WindowsApps;D:\software\Microsoft VS Code\bin;C:\Users\xiaguangbo\AppData\Roaming\npm;D:\software\mingw64\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.67.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell C:\windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) D:\software\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) D:\software\esp\.espressif\tools\xtensa-esp-elf-gdb\11.1_20220318\xtensa-esp-elf-gdb\bin;D:\software\esp\.espressif\tools\riscv32-esp-elf-gdb\11.1_20220318\riscv32-esp-elf-gdb\bin;D:\software\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;D:\software\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;D:\software\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;D:\software\esp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;D:\software\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;D:\software\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;D:\software\esp\.espressif\tools\cmake\3.23.1\bin;D:\software\esp\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin;D:\software\esp\.espressif\tools\ninja\1.10.2;D:\software\esp\.espressif\tools\idf-exe\1.0.3;D:\software\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;D:\software\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"D:\\software\\esp\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) D:\software\esp\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM6
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) D:\software\esp\.espressif
Git Path (idf.gitPath) D:\software\esp\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to D:\software\esp\.espressif\tools\xtensa-esp-elf-gdb\11.1_20220318\xtensa-esp-elf-gdb\bin: true
Access to D:\software\esp\.espressif\tools\riscv32-esp-elf-gdb\11.1_20220318\riscv32-esp-elf-gdb\bin: true
Access to D:\software\esp\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to D:\software\esp\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to D:\software\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to D:\software\esp\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to D:\software\esp\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to D:\software\esp\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to D:\software\esp\.espressif\tools\cmake\3.23.1\bin: true
Access to D:\software\esp\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin: true
Access to D:\software\esp\.espressif\tools\ninja\1.10.2: true
Access to D:\software\esp\.espressif\tools\idf-exe\1.0.3: true
Access to D:\software\esp\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to D:\software\esp\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
ESP-IDF version 5.0
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
certifi version: 2022.5.18.1
cffi version: 1.15.0
charset-normalizer version: 2.0.12
click version: 8.0.4
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'd:\xiaguangbo\project\dap\project\esp32s2\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

```

xiaguangbo, 2022-05-25:
@brianignacio5 This is the problem last time, you also sent me a fixed extension, why is this not released to the vscode extension?