espressif/vscode-esp-idf-extension #723: [Bug Report]: Blink        ESP-IDF Monitor device

seedub158, 2022-05-25:
I created a project from the blink example
I successfully built and flashed to my ESP32-DevKitM-1 v1.0
I went to MONITOR DEVICE and it kept scrolling and scrolling:
(previously it would "Turning the LED ON.....Turning the LED OFF...

Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\e729128\Documents\CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507\Espressif\VisualStudioCode\blink\blink> C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\e729128\esp\esp-idf\tools\idf.py -p COM12 monitor
Executing action: monitor
Running idf_monitor in directory c:\users\e729128\documents\cat_metal_debris_bluetooth_sensor_ewr-903507\espressif\visualstudiocode\blink\blink
Executing "C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\e729128\esp\esp-idf\tools/idf_monitor.py -p COM12 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 c:\users\e729128\documents\cat_metal_debris_bluetooth_sensor_ewr-903507\espressif\visualstudiocode\blink\blink\build\blink.elf -m 'C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\Users\e729128\esp\esp-idf\tools\idf.py' '-p' 'COM12'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM12 instead...
--- idf_monitor on \\.\COM12 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6604
ho 0 tail 12 room 4      
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (30) boot: ESP-IDF v4.4 2nd stage bootloader
I (30) boot: compile time 17:42:38
I (30) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0     
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length  
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08754h ( 34644) map
I (124) esp_image: segment 1: paddr=0001877c vaddr=3ffb0000 size=02368h (  9064) load
I (128) esp_image: segment 2: paddr=0001aaec vaddr=40080000 size=0552ch ( 21804) load
I (140) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=15cb4h ( 89268) map
I (172) esp_image: segment 4: paddr=00035cdc vaddr=4008552c size=05c38h ( 23608) load
I (182) esp_image: segment 5: paddr=0003b91c vaddr=50000000 size=00010h (    16) load
I (188) boot: Loaded app from partition at offset 0x10000
I (188) boot: Disabling RNG early entropy source...
I (203) cpu_start: Pro cpu up.
E (204) cpu_start: Running on single core variant of a chip, but app is built with multi-core support.
E (206) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig

abort() was called at PC 0x400d0df4 on core 0
0x400d0df4: start_other_core at C:/Users/e729128/esp/esp-idf/components/esp_system/port/cpu_start.c:214 (discriminator 3)

Backtrace:0x4008181e:0x3ffe3b600x40085051:0x3ffe3b80 0x40089e56:0x3ffe3ba0 0x400d0df4:0x3ffe3c10 0x40081101:0x3ffe3c40 0x400794d4:0x3ffe3c80  |<-CORRUPTED
0x4008181e: panic_abort at C:/Users/e729128/esp/esp-idf/components/esp_system/panic.c:402

0x40085051: esp_system_abort at C:/Users/e729128/esp/esp-idf/components/esp_system/esp_system.c:121

0x40089e56: abort at C:/Users/e729128/esp/esp-idf/components/newlib/abort.c:46

0x400d0df4: start_other_core at C:/Users/e729128/esp/esp-idf/components/esp_system/port/cpu_start.c:214 (discriminator 3)

[... 637 lines cut ...]
load:0x3fff0030,len:6604
ho 0 tail 12 room 4
load:0x40078000,len:14780
load:0x40080400,len:3792
ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
Waiting for the device to reconnect0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4 2nd stage bootloader
I (29) boot: compile time 17:42:38
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08754h ( 34644) map
I (124) esp_image: segment 1: paddr=0001877c vaddr=3ffb00................

seedub158, 2022-05-25:
I modified the main .c file:
*************************************************************************************************************************
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "led_strip.h"
#include "sdkconfig.h"

static const char *TAG = "example";
static uint8_t s_led_state = 0;
void app_main(void)
{
    while (1) {
        ESP_LOGI(TAG, "Turning the LED %s!", s_led_state == true ? "ON" : "OFF");
        /* Toggle the LED state */
        s_led_state = !s_led_state;
        vTaskDelay(CONFIG_BLINK_PERIOD / portTICK_PERIOD_MS);
    }
}
*************************************************************************************************************************
Monitor:

Windows PowerShell
Copyright (C) Microsoft Corporation. All rights reserved.

Try the new cross-platform PowerShell https://aka.ms/pscore6

PS C:\Users\e729128\Documents\CAT_Metal_Debris_Bluetooth_Sensor_EWR-903507\Espressif\VisualStudioCode\blink\blink> C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\e729128\esp\esp-idf\tools\idf.py -p COM12 monitor
Executing action: monitor
Running idf_monitor in directory c:\users\e729128\documents\cat_metal_debris_bluetooth_sensor_ewr-903507\espressif\visualstudiocode\blink\blink
Executing "C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\e729128\esp\esp-idf\tools/idf_monitor.py -p COM12 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 c:\users\e729128\documents\cat_metal_debris_bluetooth_sensor_ewr-903507\espressif\visualstudiocode\blink\blink\build\blink.elf -m 'C:\Users\e729128\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe' 'C:\Users\e729128\esp\esp-idf\tools\idf.py' '-p' 'COM12'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM12 instead...
--- idf_monitor on \\.\COM12 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6604
ho 0 tail 12 room 4
load:0x40078000,len:14780
load:0x40080400,len:3792
0x40080400: _init at ??:?

entry 0x40080694
I (30) boot: ESP-IDF v4.4 2nd stage bootloader
I (30) boot: compile time 17:42:38
I (30) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=076dch ( 30428) map
I (123) esp_image: segment 1: paddr=00017704 vaddr=3ffb0000 size=02340h (  9024) load
I (127) esp_image: segment 2: paddr=00019a4c vaddr=40080000 size=065cch ( 26060) load
I (139) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=148c4h ( 84164) map
I (170) esp_image: segment 4: paddr=000348ec vaddr=400865cc size=04b98h ( 19352) load
[... 543 lines cut ...]
load:0x40078000,len:14780
load:0x40080400,len:3792
ClearCommError failed (PermissionError(13, 'Access is denied.', None, 5))
Waiting for the device to reconnect0x40080400: _init at ??:?

entry 0x40080694
I (29) boot: ESP-IDF v4.4 2nd stage bootloader
I (29) boot: compile time 17:42:38
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=076dch ( 30428) map
I (123) esp_image: segment 1: paddr=00017704 vaddr=3ffb0000 size=02340h (  9024) load
I (127) esp_image: segment 2: paddr=00019a4c vaddr=40080000 size=065cch ( 26060) load
I (139) esp_image..............

brianignacio5, 2022-05-26:
This doesn't seem to be an issue with the extension but with your ESP-IDF project.

You should review your project and check for the [ESP-IDF repository](http://github.com/espressif/esp-idf/issues) and the [esp32 forum](https://esp32.com) for help

closed 2022-05-26 by brianignacio5

seedub158, 2022-05-26:
it cant be the project because it is an example project created not by me

it also cant be the project because I modified it to remove every single thing except for the message.

seedub158, 2022-05-26:
Looking through the Terminal output I saw:

pu_start: Running on single core variant of a chip, but app is built with multi-core support.
E (206) cpu_start: Check that CONFIG_FREERTOS_UNICORE is enabled in menuconfig

abort() was called at PC 0x400d0df4 on core 0

So I went to VIEW    COMMAND PALLETTE    ESP-IDF: SDK Configuration editor (menuconfig)

on the left/middle there were selections and under Example Configuration, Component config, 
I selected FreeRTOS and checked the box for Run FreeRTOS only on first core

That seemed to fix it

I (69306) example: Turning the LED ON!
I (70306) example: Turning the LED OFF!
I (71306) example: Turning the LED ON!
I (72306) example: Turning the LED OFF!
I (73306) example: Turning the LED ON!
I (74306) example: Turning the LED OFF!
I (75306) example: Turning the LED ON!
I (76306) example: Turning the LED OFF!
I (77306) example: Turning the LED ON!
I (78306) example: Turning the LED OFF!
I (79306) example: Turning the LED ON!
I (80306) example: Turning the LED OFF!
I (81306) example: Turning the LED ON!