espressif/vscode-esp-idf-extension #727: [Bug Report]: cannot start debug session with internal USB/JTAG adapter

ScarsFun, 2022-06-01:
Hello I've a problem starting debug session with ESP32C3 with internal USB/JTAG adapter. Flashing with internal adapter works well. I can start a manual gdb session correctly. problem seem to be the esp debug adapter in vscode extension.
I tryed to start manually the debug adapter extension with this error. I'm not a pyton expert so I cannot find the solution.

thanks in advance.

```
PS C:\Users\MAX\WORK\esp32\blink> python C:\Users\MAX\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\debug_adapter_main.py -e blink.elf -ip localhost -dn esp32c3 -d 5 -ln
2022-06-01 15:16:51,906 - Debug Adapter (main) - DEBUG - START_TIME_2022_06_01_15-16-51
2022-06-01 15:16:51,907 - Debug Adapter (main) - DEBUG - Init errors:None
2022-06-01 15:16:51,907 - Debug Adapter (main) - INFO - Working directory: C:\Users\MAX\WORK\esp32\blink
2022-06-01 15:16:51,908 - Debug Adapter (main) - INFO - Arguments:
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 5,
    'developer_mode': 'none',
    'device_name': 'esp32c3',
    'elfpath': ('blink.elf',),
    'log_file': None,
    'log_mult_files': None,
    'log_no_debug_console': True,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': None}
Traceback (most recent call last):
  File "C:\Users\MAX\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\debug_adapter_main.py", line 30, in <module>
    cli(sys.argv[1:])
  File "C:\Users\MAX\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\MAX\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\Users\MAX\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\MAX\AppData\Local\Programs\Python\Python310\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\MAX\AppData\Local\Programs\Python\Python310\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\MAX\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\debug_adapter\cli.py", line 145, in cli
    dbg_a = DebugAdapter(args=args_main)
  File "C:\Users\MAX\.vscode\extensions\espressif.esp-idf-extension-1.4.0\esp_debug_adapter\debug_adapter\debug_adapter.py", line 76, in __init__
    _, self.target_triple = os.path.split(args.toolchain_prefix)
  File "C:\Users\MAX\AppData\Local\Programs\Python\Python310\lib\ntpath.py", line 185, in split
    p = os.fspath(p)
TypeError: expected str, bytes or os.PathLike object, not NoneType
PS C:\Users\MAX\WORK\esp32\blink>
```
**ESP-IDF doctor**
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\PuTTY\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files (x86)\GNU Arm Embedded Toolchain\10 2021.10\bin;C:\Users\MAX\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\MAX\AppData\Local\Programs\Python\Python310\;C:\Users\MAX\AppData\Local\Microsoft\WindowsApps; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.67.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\MAX\esp\esp-idf
[... 353 lines cut ...]

2022-06-01 15:16:02,962 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2022-06-01 15:16:02,963 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "restart": false
    },
    "command": "disconnect",
    "seq": 2,
    "type": "request"
}
2022-06-01 15:16:02,963 - Debug Adapter (main) - DEBUG - Adapter is Stopping...
2022-06-01 15:16:02,964 - Debug Adapter (main) - DEBUG - Stopping target poller
2022-06-01 15:16:02,964 - Debug Adapter (main) - DEBUG - Closing the GDB process
2022-06-01 15:16:02,965 - Debug Adapter (main) - DEBUG - Closing the Socket
2022-06-01 15:16:02,965 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2022-06-01 15:16:02,972 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {\n        \"restart\": false\n    },\n    \"command\": \"disconnect\",\n    \"seq\": 2,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 10}

2022-06-01 15:16:02,980 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2022-06-01 15:16:02,980 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2022-06-01 15:16:02,980 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2022-06-01 15:16:02,981 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter
```

WangShuoran, 2022-06-22:
I meet same question, but in esp-idf v4.3.2 and this repo v1.2.0 the esp32c3 built-in jtag is ok

WangShuoran, 2022-06-22:
and the esp32s3 built in jtag have no response. the error scene is not same with esp32c3 built-in jtag

WangShuoran, 2022-06-25:
I am use clear Windows system. And use IDF install tools install esp-idf v4.4.1, the v4.3.2 in another PC. This system not have python, but idf default python version is 3.8, `idf4.4_py3.8_env`, I want to give you any detail info. Can you tell me you need info?

brianignacio5, 2022-06-27:
Could you try increasing the timeout in your project's launch.json by setting `tmoScaleFactor: 2` or other value (default is 1). Based on the latest debug adapter log, seems like a timing issue.

as seen in [Debugging doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)

WangShuoran, 2022-07-14:
@brianignacio5

brianignacio5, 2022-07-15:
@WangShuoran  I need the same information given by @ScarsFun 

Also have you tried `tmoScaleFactor: 2` ?

WangShuoran, 2022-07-18:
https://github.com/espressif/openocd-esp32/issues/236#issuecomment-1179686968

brianignacio5, 2022-07-18:
I'm closing this issue from lack of response from original user.