espressif/vscode-esp-idf-extension #732: [Bug Report]: VSCode hangs when hovering local variable named 'data'

brand17, 2022-06-05:
**Describe the bug**
If you have a local variable named `data` - VSCode hangs for a few seconds when you hover over this variable. And then it gives a huge list of all variables with this name from other libraries.
If you try to refactor (rename the variable with `F2`) - it takes a couple of minutes with a huge list of files offered.
I think `C/C++` extension working wrong with `esp-idf` extension when determining the variable scope.
There are no similar errors when `C/C++ extension` works on console c/c++ projects (not `esp-idf`).

**To Reproduce**
1. create a file with `c` extension
2. copy the code below
```
void app_main()
{
    int data;
}

```
3. hover a mouse cursor over `data` variable

**Expected behavior**
When hovering a local variable named `data` - there should be no delay and its result should be a definition of the local variable. Renaming this variable using `F2` should be limited by the current file only.

**Environment (please complete the following information):**

- OS Name & Version: [OS win32 x64 10.0.19044]
- VSCode Version: [1.67.2]
- ESP-IDF Version: [v4.4.1]
- Python Version: [3.8.7]

brianignacio5, 2022-06-08:
What is your [C/C++ Configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md) ? Are you using the `compile_commands.json` ?

brand17, 2022-06-08:
I don't use `compile_commands.json`.

My `c_cpp_properties.json` is:

```
{
    "configurations": [
        {
            "name": "ESP-IDF",
            "compilerPath": "C:\\Espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32-elf\\bin\\xtensa-esp32-elf-gcc.exe",
            "cStandard": "c11",
            "cppStandard": "c++17",
            "includePath": [
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
```

closed 2022-07-01 by github-actions[bot]