espressif/vscode-esp-idf-extension #735: [Bug Report]: Fresh express or advanced install got EBUSY: resource busy or locked, rename... error

justsoft, 2022-06-08:
**Describe the bug**

After unzipped the latest version of esp-idf.zip file, failed to rename the folder to the target folder `esp-idf` on Windows 10 machine.

**To Reproduce**
1)  Install the `Espressif IDF Extension of VSCODE`
2)  The `ESP-IDF Setup` shows up automatically, or press CTRL+SHIFT+P, enter `idf:configure` to launch the `ESP-IDF Setup`
3)  Click either the `EXPRESS` or `ADVANCED`
4) After loaded and extracted the esp-idf-v4.4.1.zip, it failed to rename the unzipped folder.
5) The error message is:
>  ```
{"user":true,"message":"EBUSY: resource busy or locked, rename 'C:\\Users\\john\\esp\\esp-idf-v4.4.1' -> 'C:\\Users\\john\\esp\\esp-idf'","stack":"Error: EBUSY: resource busy or locked, rename 'C:\\Users\\john\\esp\\esp-idf-v4.4.1' -> 'C:\\Users\\john\\esp\\esp-idf'","level":"error","timestamp":"2022-06-07T16:10:51.331Z"}
```

**Expected behavior**
Install goes smoothly without any error.

**Environment:**

- Windows 10 10.0.19044.1706
- VSCode 1.67.2
- ESP-IDF 4.4.1
- Python 3.10.4

brianignacio5, 2022-06-08:
Is the `C:\\Users\\john\\esp` directory doesn't have permission to write or read the folder? Or is it `C:\\Users\\john\\esp\\esp-idf-v4.4.1` being used by other program ?

justsoft, 2022-06-08:
The logon user `john` has sufficient permission to `C:\Users\john\esp` folder, the extension downloaded the esp-idf.zip and extracted the zip file in that folder with no problem.
It seems like some files are still opened while the extension is trying to rename the folder.

xWtr, 2022-06-19:
Hi, Have you found a solution? Same issue.

pavelmaca, 2022-06-20:
Try changing current project directory outside `C:\Users\john\esp\`

closed 2022-07-12 by github-actions[bot]