espressif/vscode-esp-idf-extension #742: [Bug Report]: using any spaces in filepath breaks ESP-IDF usage

toybuilder, 2022-06-11:
tl/dr; Fails horribly if you have any spaces in your filepath.

As someone that is still new with VS Code and with VSC+ESP-IDF, I had a frustrating time getting VSC+ESP-IDF to work.  After spending about three hours, I finally got it to work.   

There is one critical aspect in my experience: I had a space in my file path.  Because I wanted to use ESP-IDF with a project for a client, the working directory included the client name which has a space in it.

This apparently was breaking the device-target selection and menuconfig operation.  The error messages were cryptic and generally unhelpful.  When trying to debug the menuconfig not working, the output session for menuconfig would close before I could read its contents.

I uninstalled and reinstalled the ESP-IDF extension a few times.  I even uninstalled my standard ESP-IDF (the one launched from the desktop) and tried again.

It was only when I tried to open the example in a new directory that it worked.  And it wasn't the path length.

That was three (or more) hours down the drain.

Perhaps it serves me right to use spaces in a path with coding tools.  But I've had Arduino projects under the same client's directory which worked fine with Arduino-ESP32 (via Arduino), so I had no reason to believe that the VSC ESP32 tools wouldn't work in that same folder.

Now that I've gotten past this single detail that was proving to be a critical tripping point, I happily watched the build process run and build successfully.

To reproduce this, go through the steps of setting up the example Blink project, but in a directory with a space somewhere in the filepath.

Setting device-target yields:
[image]
`-- Building ESP-IDF components for target esp32
-- Project sdkconfig file C:/Users/joech/esp/x 1/blink/sdkconfig
Traceback (most recent call last):
  File "C:/Users/joech/esp/esp-idf/tools/kconfig_new/confgen.py", line 640, in <module>
    main()
  File "C:/Users/joech/esp/esp-idf/tools/kconfig_new/confgen.py", line 260, in main
    config = kconfiglib.Kconfig(args.kconfig)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 947, in __init__
    self._init(filename, warn, warn_to_stderr, encoding)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 1085, in _init
    self._parse_block(None, self.top_node, self.top_node).next = None
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 2982, in _parse_block
    prev = self._parse_block(None, parent, prev)
  File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\kconfiglib.py", line 2970, in _parse_block
    raise KconfigError(
kconfiglib.KconfigError: C:/Users/joech/esp/x 1/blink/build/kconfigs_projbuild.in:6: 'C:/Users/joech/esp/x' not found (in 'source "C:/Users/joech/esp/x"'). Check that environment variables are set correctly (e.g. $srctree, which is unset or blank). Also note that unset environment variables expand to the empty string.
CMake Error at C:/Users/joech/esp/esp-idf/tools/cmake/kconfig.cmake:266 (message):
  Failed to run confgen.py`

Attempting to run menuconfig yields a progress-bar that gets stuck:
[image]

I would have expected things to work with spaces in the filepath; or at least some kind of warning/error complaining about the space.

FWIW, this is on Windows 10 with VSC 1.68.0, ESP-IDF 1.4.0

brianignacio5, 2022-06-13:
We are very sorry for the trouble. The thing is, ESP-IDF itself still doesn't fully work with spaces in it so the extension also doesn't support spaces.. Arduino built is different so they do support it.

We will add space validation now, but at some point we will support spaces in environment variables, ESP-IDF and tools.

toybuilder, 2022-06-14:
Even before fixing the tools to add the space validation, it would be very helpful if your documentation is more explicit about spaces in the filepath.  As someone is following along on the "getting started" step-by-step procedure, having a small warning about spaces would go a long way to avoid the pain of trying to figure out why things aren't working.

toybuilder, 2022-06-19:
Oh, I just discovered today that the Git master branch of the IDF documentation includes a notice about not having spaces in the path -- apparently added about two months ago!  However, I didn't see that as I was working off documentation for the stable 4.4.1 release version!

closed 2022-07-25 by brianignacio5

nweaver, 2024-06-12:
Issue still remains.  Are there any intents to actually fix this?  This is especially painful when the user account itself has spaces in the name.

brianignacio5, 2024-06-13:
Did you try the latest master installer [here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ? We added spaces fix in https://github.com/espressif/vscode-esp-idf-extension/pull/1159

nweaver, 2024-06-13:
Used the extension from the VSCode extension collection, did not download directly.

brianignacio5, 2024-06-13:
The next release v1.8.0 in marketplace will have these changes, but we are testing it right now since we added many new things.

Ishan-Sharma-17, 2024-12-12:
[image]

Hi, has the issue been fixed yet? If not, is there any workaround this, since my username itself contains spaces

brianignacio5, 2024-12-12:
It should already be fixed @Ishan-Sharma-17

Ishan-Sharma-17, 2024-12-12:
Do you know why I could still be getting this error? I am using the Visual Studio code extension

brianignacio5, 2024-12-12:
The reason is that white spaces are not supported for the ESP-IDF Tools directory path. This could generate issues while building the project.

Ishan-Sharma-17, 2024-12-12:
Is there any workaround? My laptop username contains a space and I can't really change my laptop username

brianignacio5, 2024-12-12:
You can use another location like `C:\esp\.espressif`

Ishan-Sharma-17, 2024-12-12:
So espressif does not need to be installed in the same directory as visual studio code?

brianignacio5, 2024-12-12:
No it doesn't. There are settings in the vscode extension (idf.toolsPath and idf.espIdfPath) saved in settings.json that are used to find where ESP-IDF and ESP-IDF Tools are located.

PR #761 (2022-07-25): add space validation doctor cmd setup wizard
## Description

Please include a summary of the change and which issue is fixed.

Fix #742
Fix #751 

## Type of change

Add validation of spaces in path in both the setup wizard and the doctor command.

- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing observing result in doctor command and the Setup Wizard with wrong and good paths.

**Test Configuration**:
* ESP-IDF Version: v5.0
* OS (Windows,Linux and macOS): MacOS

## Dependent components impacted by this PR:

No components impacted.

## Checkli [... cut]