espressif/vscode-esp-idf-extension #743: [Bug Report]: Cannot change Path (Cannot read properties of undefined (reading 'uri'))

MarkusvonStaden, 2022-06-14:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [x] Make sure you have searched for existing bugs and features request before you post an issue.
- [x] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [x] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
I installed IDF using the Windows Installer, not the extension.
I used the default installation path from the installer (C:\Espressif\...)
Since the Setup did not find this installation, i wanted to configure the paths manually with the "Configure Paths" command.
After entering the path, I get the error message
`Cannot read properties of undefined (reading 'uri')`

**To Reproduce**
`[CTRL + SHIFT + P] -> ESP_IDF: Configure Paths -> Enter Path`

**Expected behavior**
No Error Message

**Screenshots**

**Environment (please complete the following information):**

- OS Name & Version: Windows 11 Pro 21H2 22000.675
- VSCode Version: 1.68.0
- ESP-IDF Version: 4.4.1
- Python Version: 3.8.7

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**Output**
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\VMware\VMware Player\bin\;C:\Python310\Scripts\;C:\Python310\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\Users\marku\anaconda3\Scripts;C:\Users\marku\anaconda3;C:\Users\marku\anaconda3\Library\bin;C:\Program Files (x86)\PDFtk Server\bin\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;;C:\Program Files\PuTTY\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Users\marku\AppData\Local\Microsoft\WindowsApps;C:\Users\marku\AppData\Roaming\npm;C:\Users\marku\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.68.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Users\marku\AppData\Local\Microsoft\WindowsApps\Microsoft.PowerShell_8wekyb3d8bbwe\pwsh.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\marku\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\marku\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Users\marku\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\marku\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\marku\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\marku\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\marku\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\marku\.espressif\tools\cmake\3.20.3\bin;C:\Users\marku\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Users\marku\.espressif\tools\ninja\1.10.2;C:\Users\marku\.espressif\tools\idf-exe\1.0.3;C:\Users\marku\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\marku\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\marku\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\marku\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM12
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:/Users/marku/.espressif
Git Path (idf.gitPath) C:\Users\marku\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\marku\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: false
Access to C:\Users\marku\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: false
Access to C:\Users\marku\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: false
Access to C:\Users\marku\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: false
Access to C:\Users\marku\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: false
Access to C:\Users\marku\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: false
Access to C:\Users\marku\.espressif\tools\cmake\3.20.3\bin: false
Access to C:\Users\marku\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: false
Access to C:\Users\marku\.espressif\tools\ninja\1.10.2: false
Access to C:\Users\marku\.espressif\tools\idf-exe\1.0.3: false
[... 26 lines cut ...]
> **NOTE:** Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

> **NOTE:** Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process output.

**Logs**
Please share the log file which can be obtained from:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

brianignacio5, 2022-06-15:
This should be already fixed in master with #701 but not yet released in marketplace.

brianignacio5, 2022-06-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/266084219.zip) and see if the issue is fixed ?

MarkusvonStaden, 2022-06-17:
Yeah, this fixes the issue 💯

closed 2022-06-17 by MarkusvonStaden