espressif/vscode-esp-idf-extension #750: [Bug Report]: Debugging on macOS with Microsoft C/C++ extension cannot start because gdb is not signed

davide-b, 2022-06-28:
**Bug description**
When I try to start a debugging session through VSCode, using the Microsoft C/C++ extension, the debug session fails to start and I get a dialog with the message

> Debugger executable '/Users/{user}/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb' is not signed. As a result, debugging may not work properly.

**To Reproduce**
I followed the steps to install the ESP-IDF VSCode extension reported on the Espressif docs page. Then created a launch.json file as follows

```
{
    "version": "0.2.0",
    "configurations": [
      {
        "type": "espidf",
        "name": "ESP-IDF",
        "request": "launch"
      },
      {
        "name": "GDB",
        "type": "cppdbg",
        "request": "launch",
        "MIMode": "gdb",
        "miDebuggerPath": "${command:espIdf.getXtensaGdb}",
        "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
        "windows": {
          "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
        },
        "cwd": "${workspaceFolder}",
        "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
        "setupCommands": [
          { "text": "target remote :3333" },
          { "text": "set remote hardware-watchpoint-limit 2"},
          { "text": "mon reset halt" },
          { "text": "thb app_main" },
          { "text": "flushregs" }
        ],
        "externalConsole": false,
        "logging": {
          "engineLogging": true
        }
      }
    ]
  }
```

When clicking on the launch icon in the debug tab of VSCode I get the following log in the debug console.

```
1: (57) LaunchOptions{"name":"GDB","type":"cppdbg","request":"launch","MIMode":"gdb","miDebuggerPath":"/Users/{user}/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb","program":"/Users/{user}/{folder}/firmware/products/{product}/esp32/build/{product}-esp32.elf","cwd":"/Users/{user}/{folder}/firmware/products/{product}/esp32","environment":[{"name":"PATH","value":"/Users/{user}/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/Users/{user}/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/Users/{user}/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/Users/{user}/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/{user}/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/{user}/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/{user}/.espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/{user}/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/Users/{user}/.espressif/tools/ninja/1.10.2"}],"setupCommands":[{"text":"target remote :3333"},{"text":"set remote hardware-watchpoint-limit 2"},{"text":"mon reset halt"},{"text":"thb app_main"},{"text":"flushregs"}],"externalConsole":false,"logging":{"engineLogging":true},"__configurationTarget":5,"__sessionId":"865af36c-e6b6-4803-9da5-b3a5fea36f94"}
1: (124) codeSign-stderr: /Users/{user}/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb: code object is not signed at all
1: (125) Send Event AD7MessageEvent
1: (127) Starting: "/Users/{user}/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb" --interpreter=mi
1: (169) DebuggerPid=19486
1: (1189) ->=thread-group-added,id="i1"
1: (1189) ->~"GNU gdb (crosstool-NG esp-2021r2-patch3) 9.2.90.20200913-git\n"
1: (1189) ->~"Copyright (C) 2020 Free Software Foundation, Inc.\n"
1: (1189) ->~"License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>\nThis is free software: you are free to change and redistribute it.\nThere is NO WARRANTY, to the extent permitted by law."
1: (1190) ->~"\nType \"show copying\" and \"show warranty\" for details.\n"
1: (1190) ->~"This GDB was configured as \"--host=x86_64-host_apple-darwin12 --target=xtensa-esp32-elf\".\n"
1: (1190) ->~"Type \"show configuration\" for configuration details.\n"
1: (1190) ->~"For bug reporting instructions, please see:\n"
1: (1190) ->~"<http://www.gnu.org/software/gdb/bugs/>.\n"
1: (1190) ->~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
1: (1190) ->~"\n\n"
1: (1190) ->~"For help, type \"help\".\n"
1: (1190) ->~"Type \"apropos word\" to search for commands related to \"word\".\n"
1: (1190) ->(gdb)
1: (1193) <-1001-gdb-set mi-async on
1: (1194) ->1001^done
1: (1194) ->(gdb)
[... 60 lines cut ...]
Access to /Users/{user}/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin: true
Access to /Users/{user}/.espressif/tools/ninja/1.10.2: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: which cmake

--------------------------------------------------------------------------------------------------------------------------------------------
```

Not including any more output because I think it may be irrelevant. Do tell me if you need more information to work on this issue.

brianignacio5, 2022-06-29:
You have both `espidf` and `cppdbg` configurations, you should choose one.

In this repository we maintain `espidf` one but if you have problem with the Microsoft C/C++ debug you should open a issue in their repository.

We provide example configurations in [debugging doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)

Based on the result of your debug session, the xtensa toolchain used in not code signed for MacOS. IF you try to use our debug option, you should configure the extension properly, since python packages are not found and git neither.

davide-b, 2022-06-29:
I removed the espidf configuration from the `launch.json`, but the result is the same.

The debugging doc you provided is exactly the guide that I followed. I copied the Microsoft C/C++ configuration from there. I opened an issue here because the problem seems to arise from the xtensa gdb program that the esp idk extension uses when launching a debug session.

Wouldn't there be a possibility to have the debugger used in the esp idk configuration work in the Microsoft configuration too?

brianignacio5, 2022-06-29:
Well the thing is the debugger is implemented by the C/C++ extension and part of their extension, not ours so we can't maintain it. We have our own debugger, which we maintain. So if you have an issue with their debugger, you should ask in their repository.

Having another look at your logs, it seems that gdb it is executed. The issue is that `target remote :3333` fails. Did. you start openOCD before running the debug session ?

davide-b, 2022-06-29:
Actually the OpenOCD was not started. Starting it manually through the button on the toolbar at the bottom of VSCode solved the issue.

closed 2022-06-29 by davide-b

parksj10, 2023-06-10:
FYI I had to kill anything on port `6666` to enable `openocd` to run... was getting this non-descript error until I did that