espressif/vscode-esp-idf-extension #757: [Bug Report]: Open the OpenOCD and flash with jtag occur error

WangShuoran, 2022-07-06:
- Development Kit:      ESP32-C3,ver3
- Module or chip used:  ESp32-C3
- IDF version:          v4.4.1
- Build System:         idf.py
- Compiler version:     v8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?:        Vscode
- Power Supply:         USB

在我用Vscode插件打开OpenOCD后，选择烧录模式为JTAG时，进行烧录，会报错：
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
❌ Error: Algorithm timed out after 40001 ms.
❌ Error: ra = 0x40383c12
❌ Error: sp = 0x3fc847b0
❌ Error: gp = 0x0
❌ Error: tp = 0x0
❌ Error: t0 = 0x0
❌ Error: t1 = 0x40383bee
❌ Error: t2 = 0x0
❌ Error: s0 = 0x3fc94000
❌ Error: s1 = 0x1702aa49
❌ Error: a0 = 0x0
❌ Error: a1 = 0x0
❌ Error: a2 = 0x4
❌ Error: a3 = 0x0
❌ Error: a4 = 0x0
❌ Error: a5 = 0x0
❌ Error: a6 = 0x0
❌ Error: a7 = 0x403816e4
❌ Error: s2 = 0x66
❌ Error: s3 = 0x1
❌ Error: s4 = 0x3
❌ Error: s5 = 0x0
❌ Error: s6 = 0x0
❌ Error: s7 = 0xa0
❌ Error: s8 = 0x0
❌ Error: s9 = 0xa0
❌ Error: s10 = 0x14
❌ Error: s11 = 0x8
❌ Error: t3 = 0x0
❌ Error: t4 = 0x0
❌ Error: t5 = 0x0
❌ Error: t6 = 0x0
❌ Error: pc = 0x403803da
❌ Error: mstatus = 0x201800
❌ Error: mepc = 0x0
❌ Error: mcaus❌ e = 0x0
Error: Failed to wait algorithm (-302)!
Error: Algorithm run failed (-302)!
Error: Failed to run flasher stub (-302)!
Warn : Failed to get flash mappings (-302)!
❌ Error: Target is already running ❌ an algorithm
Error: Failed to start algorithm (-4)!
Error: Failed to run flasher stub (-4)!
❌ Error: Target is already running an algorithm
Error: Failed to start algor❌ ithm (-4)!
Error: Failed to run flasher stub (-4)!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Failed to find bank 'esp32c3.flash'!
** Flashing Failed **
-1

brianignacio5, 2022-07-06:
What is the value of `idf.openOcdConfigs` ? You should probably ask for help in the Espressif's OpenOCD repository

WangShuoran, 2022-07-07:
I have not this config, but relevant config is:
"idf.espIdfPathWin": "C:/Espressif/frameworks/esp-idf-v4.4.1/",
  "idf.pythonBinPathWin": "C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe",
  "idf.toolsPathWin": "C:\\Espressif",
  "idf.customExtraPaths": "C:\\Espressif\\tools\\xtensa-esp32-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32-elf\\bin;C:\\Espressif\\tools\\xtensa-esp32s2-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32s2-elf\\bin;C:\\Espressif\\tools\\xtensa-esp32s3-elf\\esp-2021r2-patch3-8.4.0\\xtensa-esp32s3-elf\\bin;C:\\Espressif\\tools\\riscv32-esp-elf\\esp-2021r2-patch3-8.4.0\\riscv32-esp-elf\\bin;C:\\Espressif\\tools\\esp32ulp-elf\\2.28.51-esp-20191205\\esp32ulp-elf-binutils\\bin;C:\\Espressif\\tools\\esp32s2ulp-elf\\2.28.51-esp-20191205\\esp32s2ulp-elf-binutils\\bin;C:\\Espressif\\tools\\cmake\\3.20.3\\bin;C:\\Espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220\\openocd-esp32\\bin;C:\\Espressif\\tools\\ninja\\1.10.2;C:\\Espressif\\tools\\idf-exe\\1.0.3;C:\\Espressif\\tools\\ccache\\4.3\\ccache-4.3-windows-64;C:\\Espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
  "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Espressif\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
  "idf.gitPath": "C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe",

brianignacio5, 2022-07-18:
Could you update the `idf.openOcdConfigs` with the `Select OpenOCD Board Configuration` command  and select the proper board you are using?

Also try with the latest openOCD version

brianignacio5, 2022-07-28:
Closing this issue for lack of response. Please comment here if still having issues.

closed 2022-07-28 by brianignacio5