espressif/vscode-esp-idf-extension #759: [Bug Report]: config.env locked, so clean do not work

arhi, 2022-07-14:
Click on the trashcan icon (ESP-IDF Full Clean) and I get

```
EBUSY: resource busy or locked, rmdir '.....\build'
```

When I try to manually delete the build directory or files in it I see that config.env file in build directory is locked (by some python3 process)

- OS Name & Version: Windows 11
- VSCode Version: 1.69.0
- ESP-IDF Version: 4.4.1-dirty
- Python Version: 3.9.7

brianignacio5, 2022-07-14:
This might be related to a running SDK Configuration Editor process running in the background.

Have you tried running the `Dispose current SDK Configuration editor server process` command before deleting this folder ?

arhi, 2022-07-14:
Nope, had no idea it exist :D 
That solved the problem, thanks! Looks like not a BUG but PEBMAC :)

closed 2022-07-14 by arhi