espressif/vscode-esp-idf-extension #760: [Bug Report]: `\` as drive letter in Problems section

xobs, 2022-07-15:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [ ] Make sure you have searched for existing bugs and features request before you post an issue.
- [ ] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [ ] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
The `PROBLEMS` tab shows the drive letter as `\`, which prevents it from being useful as a control-click target.

**To Reproduce**
Build a project with errors

**Expected behavior**
VSCode should show the correct drive letter

**Screenshots**
[image]

[image]

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5]
- VSCode Version: [e.g. 1.35.1]
- ESP-IDF Version: [e.g. v3.2.2]
- Python Version: [e.g. 2.7.10]

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\PowerShell\7;C:\Program Files\ImageMagick-7.1.0-Q16-HDRI;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\ojdkbuild\java-17-openjdk-17.0.2.0.8-1\bin;C:\Program Files\dotnet\;C:\Program Files\nodejs\;C:\Program Files\Git\cmd;C:\Program Files\Go\bin;C:\Program Files (x86)\MATLAB\MATLAB Runtime\v85\runtime\win32;C:\Program Files\PowerShell\7\;C:\Program Files\Microsoft Network Monitor 3\;C:\Program Files (x86)\Windows Kits\10\Windows Performance Toolkit\;C:\Users\Sean\.cargo\bin;C:\Users\Sean\AppData\Local\Microsoft\WindowsApps;C:\Users\Sean\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\Sean\.dotnet\tools;C:\Users\Sean\AppData\Roaming\npm;C:\Users\Sean\go\bin;E:\Software\heroku\bin;C:\Program Files (x86)\Nmap 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.69.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\PowerShell\7\pwsh.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\Sean\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\Sean\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220529\xtensa-esp-elf-gdb\bin;C:\Users\Sean\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220529\riscv32-esp-elf-gdb\bin;C:\Users\Sean\.espressif\tools\xtensa-esp32-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32-elf\bin;C:\Users\Sean\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s2-elf\bin;C:\Users\Sean\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s3-elf\bin;C:\Users\Sean\.espressif\tools\riscv32-esp-elf\esp-2022r1-RC1-11.2.0\riscv32-esp-elf\bin;C:\Users\Sean\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\Sean\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\Sean\.espressif\tools\cmake\3.23.1\bin;C:\Users\Sean\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin;C:\Users\Sean\.espressif\tools\ninja\1.10.2;C:\Users\Sean\.espressif\tools\idf-exe\1.0.3;C:\Users\Sean\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\Sean\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Sean\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\Sean\.espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe
Serial port (idf.port) COM12
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\Sean\.espressif
Git Path (idf.gitPath) C:\Users\Sean\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\Sean\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220529\xtensa-esp-elf-gdb\bin: true
Access to C:\Users\Sean\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220529\riscv32-esp-elf-gdb\bin: true
Access to C:\Users\Sean\.espressif\tools\xtensa-esp32-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:\Users\Sean\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\Sean\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\Sean\.espressif\tools\riscv32-esp-elf\esp-2022r1-RC1-11.2.0\riscv32-esp-elf\bin: true
Access to C:\Users\Sean\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\Sean\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\Sean\.espressif\tools\cmake\3.23.1\bin: true
Access to C:\Users\Sean\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin: true
Access to C:\Users\Sean\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\Sean\.espressif\tools\idf-exe\1.0.3: true
[... 46 lines cut ...]
Check ESP-IDF Python packages Error
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages Error
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Error
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'e:\Code\esp\blackmagic-esp32\.vscode\launch.json'
--------------------------------------------------------------------------------------------------------------------------------------------

In Visual Studio Code select menu `View` -> `Output` -> ESP-IDF, ESP-IDF Debug Adapter, Heap Trace, OpenOCD and SDK Configuration Editor.

> **NOTE:** Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

> **NOTE:** Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process output.

**Logs**
Please share the log file which can be obtained from:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Additional context**
It's unclear where it's getting this value from, or what extension is causing this error. I can use the `ESP-IDF Build` terminal window to control-click in order to jump to errors, but it seems like the "PROBLEMS" tab was designed to do this.

brianignacio5, 2022-07-15:
This is a duplicate of #749 and should be fixed on GitHub master.

Have you tried this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/292629786.zip) to see if the problem is fixed ?

xobs, 2022-07-15:
That does fix the issue. I didn't notice it was a duplicate because I didn't see any bugs with `\` in their name.

Note that as a separate issue, vscode refused to install the vsix file until I quit vscode and manually deleted the esp-idf log directory in the vscode extensions directory. Not a big deal, just a "hey, that's funny" sort of thing.

closed 2022-07-15 by xobs