espressif/vscode-esp-idf-extension #762: [Bug Report]: vscode installs wrong tools for esp-idf on master branch

xobs, 2022-07-19:
**Describe the bug**
When installing esp-idf using vscode, the option is available to select "master" as the version. This clones the `esp-idf` repository instead of getting a tagged release.

Unfortunately, `export.ps1` reports that the tools downloaded by vscode are incorrect. This can be manually corrected by running the scripts.

**To Reproduce**
1. Remove ~/.espressif and ~/esp
2. Restart vscode
3. Install the "master" branch, including downloading tools
4. Open powershell and run "export.ps1"

**Expected behavior**
The setup scripts should not indicate the tools are invalid, and certainly shouldn't give advice to remove all the tools that were just downloaded.

**Screenshots**
```
[8:59:20 am] ~/esp/esp-idf> .\export.ps1
Setting IDF_PATH: C:\Users\Sean\esp\esp-idf
Checking Python compatibility
Adding ESP-IDF tools to PATH...

Name                           Value
----                           -----
OPENOCD_SCRIPTS                C:\Users\Sean\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE              1
ESP_IDF_VERSION                5.0

Added to PATH
-------------
C:\Users\Sean\esp\esp-idf\components\esptool_py\esptool
C:\Users\Sean\esp\esp-idf\components\app_update
C:\Users\Sean\esp\esp-idf\components\espcoredump
C:\Users\Sean\esp\esp-idf\components\partition_table
C:\Users\Sean\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220529\xtensa-esp-elf-gdb\bin
C:\Users\Sean\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220529\riscv32-esp-elf-gdb\bin
C:\Users\Sean\.espressif\tools\xtensa-esp32-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32-elf\bin
C:\Users\Sean\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s2-elf\bin
C:\Users\Sean\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-RC1-11.2.0\xtensa-esp32s3-elf\bin
C:\Users\Sean\.espressif\tools\riscv32-esp-elf\esp-2022r1-RC1-11.2.0\riscv32-esp-elf\bin
C:\Users\Sean\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin
C:\Users\Sean\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin
C:\Users\Sean\.espressif\tools\cmake\3.23.1\bin
C:\Users\Sean\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin
C:\Users\Sean\.espressif\tools\idf-exe\1.0.3\
C:\Users\Sean\.espressif\tools\ccache\4.3\ccache-4.3-windows-64
C:\Users\Sean\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
C:\Users\Sean\esp\esp-idf\tools
Checking if Python packages are up to date...
ERROR: C:\Users\Sean\.espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

Detected installed tools that are not currently used by active ESP-IDF version.
For removing dfu-util, esp32s2ulp-elf, esp32ulp-elf, riscv32-esp-elf, riscv32-esp-elf-gdb, xtensa-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf use command 'python.exe C:\Users\Sean\esp\esp-idf\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Users\Sean\esp\esp-idf\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:
    idf.py build

[8:59:28 am] ~/esp/esp-idf> xtensa-esp-elf-gdb-no-python.exe --version
GNU gdb (esp-gdb) 11.2_20220529
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
[9:00:07 am] ~/esp/esp-idf> python.exe C:\Users\Sean\esp\esp-idf\tools\idf_tools.py uninstall
C:\Users\Sean\.espressif\tools\dfu-util was removed.
C:\Users\Sean\.espressif\tools\esp32s2ulp-elf was removed.
C:\Users\Sean\.espressif\tools\esp32ulp-elf was removed.
C:\Users\Sean\.espressif\tools\riscv32-esp-elf was removed.
C:\Users\Sean\.espressif\tools\riscv32-esp-elf-gdb was removed.
C:\Users\Sean\.espressif\tools\xtensa-esp-elf-gdb was removed.
C:\Users\Sean\.espressif\tools\xtensa-esp32-elf was removed.
C:\Users\Sean\.espressif\tools\xtensa-esp32s2-elf was removed.
C:\Users\Sean\.espressif\tools\xtensa-esp32s3-elf was removed.
[9:00:11 am] ~/esp/esp-idf> xtensa-esp-elf-gdb-no-python.exe --version
xtensa-esp-elf-gdb-no-python.exe: The term 'xtensa-esp-elf-gdb-no-python.exe' is not recognized as a name of a cmdlet, function, script file, or executable program.
Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
[9:00:13 am] ~/esp/esp-idf>
```

**Environment (please complete the following information):**

- OS Name & Version: Windows 11
- VSCode Version: 1.69.2
- ESP-IDF Version: v5.0-dev-4037-g9b8c558e63
- Python Version: 3.10.5 as installed when you type "python" in Windows

**Additional context**
This setup seems to use my system python now, which is very nice because the system python includes venv but the python-3.8 that vscode installs does not contain venv. However, I suspect this is not supposed to happen.

xobs, 2022-07-19:
This initially was part of a bug report on esp-idf: https://github.com/espressif/esp-idf/issues/9379

xobs, 2022-07-26:
Note that this can be fixed by simply running the following inside `~/esp/esp-idf/tools`:

```text
# Mark the dependencies as being installed
python .\idf_tools.py install

# Set up the python environment
python .\idf_tools.py install-python-env
```

brianignacio5, 2022-07-27:
Hi @xobs thanks for your report. 

> ERROR: C:\Users\Sean\.espressif\python_env\idf5.0_py3.10_env\Scripts\python.exe doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

The issue is not that extension is installed the wrong but the the virtual env reference is different from vscode. The virtualenv creation from the extension uses the `idf_tools.py install-python-env` too so venv should be available too.

VSCode extension and ESP-IDF Installer uses the embed python (included in IDF_TOOLS_PATH) that is version 3.8.7. This embed python is used to create the virtual env so the resulting IDF_ENV_PATH is **C:\Users\Sean\.espressif\python_env\idf5.0_py3.8_env** instead of **C:\Users\Sean\.espressif\python_env\idf5.0_py3.10_env** from the command line, which uses your system python executable instead of the embed one.

I've tested this on a Windows 10 machine, and besides the python_env difference as explained before, other tools seems to be the same as described in the $IDF_PATH/tools/tools.json .

xobs, 2022-07-27:
Running `python .\idf_tools.py install-python-env` fixes the issue where it advises running `idf_tools.py uninstall --remove-archives` to uninstall the packages that it just installed.

I'm not sure what the correct approach is here. I have one venv I obtained by runing `export.ps1`, and I have another venv I got by installing vscode. It seems like they both can work side-by-side without interfering with one another.

It's definitely surprising behaviour, though I'm not sure what can be done about it. At least there's an issue created now that people will find if they search for it. Thanks for the response!

closed 2022-07-27 by xobs