espressif/vscode-esp-idf-extension #764: [Bug Report]: Compilation warnings do not appear in the Problems tab

ademenev, 2022-07-21:
**Pre Bug Report Checklist**
Before reporting any bug please make sure of these points.

- [X] Make sure you have searched for existing bugs and features request before you post an issue.
- [X] This is a bug report for the ESP-IDF Visual Studio Code extension and **not** an ESP-IDF bug report.
- [X] I've read the docs and found no information that could have helped solving the issue.

**Describe the bug**
Compilation warnings not displayed in "Problems" tab. The warnings appear in the tab for a very short time, and then disappear. If the compiler generates error(s) **and** warnings,  then the errors and warnings are displayed

**To Reproduce**
Build any project with compiler warnings (for example, unused static functions)

**Expected behavior**
Warnings displayed in "Problems" tab

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

<details>
<summary> Output of Doctor command</summary>

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 19.6.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/TeX/texbin:/opt/X11/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/usr/local/CrossPack-AVR/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.69.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /Users/andrey/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /Users/andrey/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/Users/andrey/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/Users/andrey/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/Users/andrey/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/Users/andrey/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/andrey/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/andrey/.espressif/tools/cmake/3.20.3/CMake.app/Contents/bin:/Users/andrey/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin:/Users/andrey/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/Users/andrey/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /Users/andrey/.espressif/python_env/idf4.4_py3.10_env/bin/python
Serial port (idf.port) /dev/cu.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) /Users/andrey/.espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to /Users/andrey/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to /Users/andrey/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /Users/andrey/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /Users/andrey/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to /Users/andrey/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /Users/andrey/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /Users/andrey/.espressif/tools/cmake/3.20.3/CMake.app/Contents/bin: true
Access to /Users/andrey/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin: true
Access to /Users/andrey/.espressif/tools/ninja/1.10.2: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.27.0
ESP-IDF version 4.4.1
Python version 3.10.5
[... 130 lines cut ...]
        "*.tcc": "c",
        "optional": "c",
        "istream": "c",
        "ostream": "c",
        "ratio": "c",
        "system_error": "c",
        "array": "c",
        "functional": "c",
        "regex": "c",
        "tuple": "c",
        "type_traits": "c",
        "utility": "c",
        "random": "cpp",
        "*.inc": "c",
        "stdio.h": "c",
        "esp_log.h": "c",
        "bitset": "cpp",
        "memory": "cpp",
        "future": "cpp"
    }
}

```

</details>

brianignacio5, 2022-07-26:
This might be related to some issues like #748 and #678 and the IDF Size override the terminal from IDF Build task.

Should be fixed in GitHub master.

brianignacio5, 2022-08-01:
Updates ?

Have you tried the GitHub master? This is the [vsix installer link](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/315473533.zip)

ademenev, 2022-08-01:
I tried it right now. The problems tab is still empty. The build task is highlighted in yellow though:

[image]

brianignacio5, 2022-08-08:
I tried to reproduce locally in macOS Big Sur 11.6.6 and seems to be working properly
[image]
[image]

Weird. Is the ESP-IDF Terminal running when this happens?

brianignacio5, 2022-08-15:
Any updates? Could you provide some steps to reproduce ?

brianignacio5, 2022-08-17:
Latest test got this results. Closing this issue for lack of response. Please comment here if you still have issues.
[image]
[image]

closed 2022-08-17 by brianignacio5

gonczikeve, 2023-05-14:
I still have this issue. 
idf v4.4/release, installed from the vscode store - so the idf path variables are set locally on vscode
While building, the problems window has the blue icon with the number of issues, but at the moment when the "size" task finishes running, the problems window gets cleared.

Size task still loading:

[image]

Everything finished:

[image]

eichhornp, 2023-05-24:
I still have the issue with extension 1.6.2 and esp-idf 4.4.3 .
It behaves like described above

brianignacio5, 2023-05-26:
You can disable the Size task with the `idf.enableSizeTaskAfterBuildTask` in the current master and the next v1.6.3 version.

The problem is that problems are cleared when a new task is executed. As of now tasks don't fail on warning just errors.

PR #779 (2022-08-17): update problem matcher regex
## Description

Fix the problem matcher regex to highlight errors on text editor.

Fixes #764  VSC-939

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual test building blink example on Windows 10 and MacOS

**Test Configuration**:
* ESP-IDF Version: 4.4.1, 4.4.2 and 5.0
* OS (Windows,Linux and macOS): Windows and MacOS

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [x] Verified on all platforms - Windows,Linux and macOS