espressif/vscode-esp-idf-extension #768: Failed to build a simple project

bamontales, 2022-07-29:
Problem: 
Can't build a simple project, (e.g. getting_started/blink).

Logs:
```bash
 *  Executing task: ninja  

[1/4] cmd.exe /C "cd /D C:\Users\admin\Documents\esp32\blink\build\esp-idf\esptool_py && C:\Us...uild/partition_table/partition-table.bin C:/Users/admin/Documents/esp32/blink/build/blink.bin"blink.bin binary size 0x2b9f0 bytes. Smallest app partition is 0x100000 bytes. 0xd4610 bytes (83%) free.
[2/4] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\admin\Documents\esp32\blink\build\bootloader\esp-idf\esptool_py && C:\Users\admin\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/admin/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/admin/Documents/esp32/blink/build/bootloader/bootloader.bin"
Bootloader binary size 0x62c0 bytes. 0xd40 bytes (12%) free.

 *  Executing task: C:\Users\admin\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\admin\esp\esp-idf\tools\idf_size.py c:\Users\admin\Documents\esp32\blink\build\blink.map 

/usr/bin/bash: C:Usersadmin.espressifpython_envidf4.4_py3.8_envScriptspython.exe: command not found

 *  The terminal process "C:\Program Files\Git\bin\bash.exe '--login', '-c', 'C:\Users\admin\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe C:\Users\admin\esp\esp-idf\tools\idf_size.py c:\Users\admin\Documents\esp32\blink\build\blink.map'" terminated with exit code: 127. 
```

I think the culprit here is the part directory naming in windows. See the below part, it doesn't show the proper directory, missing slashes.
`/usr/bin/bash: C:Usersadmin.espressifpython_envidf4.4_py3.8_envScriptspython.exe: command not found`

Can someone help me with this? 

Thanks.

OS: Windows 10
VS Code: v1.69.2
Espressif Extension: v1.4.0

brianignacio5, 2022-08-01:
Did you configure the extension using the `ESP-IDF: Configure ESP-IDF extension` command ?

Could you share the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) as specified in the issue template?

bamontales, 2022-08-01:
> Did you configure the extension using the `ESP-IDF: Configure ESP-IDF extension` command ?

Yes

> Could you share the [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) as specified in the issue template?

`ESP-IDF: Doctor command` output
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Volta\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Users\monta\Downloads\platform-tools;C:\msys64\mingw64\bin;C:\Program Files\PowerShell\7\;C:\Program Files\Go\bin;C:\Program Files\Git\cmd;C:\Users\monta\AppData\Local\Volta\bin;C:\Users\monta\AppData\Local\Microsoft\WindowsApps;;C:\Users\monta\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\monta\go\bin;C:\Users\monta\.deno\bin;C:\your-path\zig-windows-x86_64-your-version 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.69.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\Git\bin\bash.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\monta\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\monta\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Users\monta\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\monta\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\monta\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\monta\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\monta\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\monta\.espressif\tools\cmake\3.20.3\bin;C:\Users\monta\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:\Users\monta\.espressif\tools\ninja\1.10.2;C:\Users\monta\.espressif\tools\idf-exe\1.0.3;C:\Users\monta\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\monta\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\monta\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\monta\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\monta\.espressif
Git Path (idf.gitPath) C:\Users\monta\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\monta\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\monta\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\monta\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\monta\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\monta\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\monta\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\monta\.espressif\tools\cmake\3.20.3\bin: true
Access to C:\Users\monta\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:\Users\monta\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\monta\.espressif\tools\idf-exe\1.0.3: true
Access to C:\Users\monta\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Users\monta\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.1.windows.1
ESP-IDF version 4.4.1
Python version 3.8.7
Python's pip version 20.3.3
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2022.6.15
cffi version: 1.15.1
charset-normalizer version: 2.1.0
click version: 8.1.3
colorama version: 0.4.5
construct version: 2.10.54
contextlib2 version: 21.6.0
cryptography version: 37.0.4
ecdsa version: 0.18.0
esp-windows-curses version: 0.1
[... 188 lines cut ...]
[... cut ...] ion.js:1:33144\n    at w.invoke (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:61:145)\n    at v.deliver (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:61:2266)\n    at p.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:61:1844)\n    at f.$onDidEndTaskProcess (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:96:35605)\n    at processTicksAndRejections (node:internal/process/task_queues:96:5)","level":"error","timestamp":"2022-08-01T12:59:04.554Z"}

```

brianignacio5, 2022-08-02:
Could you change your vscode terminal to use powershell instead of Git bash ?

bamontales, 2022-08-02:
@brianignacio5 it worked, I've tried to change it one time but I guessed it should be changed as the default not just by closing and opening a new instance of PowerShell in vs code terminal. Thanks again.

closed 2022-08-02 by bamontales