espressif/vscode-esp-idf-extension #769: [Bug Report]:

iangregsondev, 2022-07-31:
**Describe the bug**

Opening the IDF Terminal using the following icon

[image]

Gives an error when trying to run idf.py

[image]

**To Reproduce**

Just open the terminal (terminal idf) from the lower toolbar button and type idf.py.

**Expected behavior**

Should recongnise its in the special terminal and not give an error message

**Screenshots**
See above

**Environment (please complete the following information):**

- OS Name & Version: [e.g. MacOS X 10.14.5] Macox 12.4
- VSCode Version: [e.g. 1.35.1] 1.69.2
- ESP-IDF Version: [e.g. v3.2.2] 1.4.0
- Python Version: [e.g. 2.7.10] Python 3.10.5

> **NOTE**: You can use the `ESP-IDF: Doctor command` to generate a report of your configuration in your clipboard which you can copy here.

**Output**

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 21.5.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /Users/ian/.sdkman/candidates/micronaut/current/bin:/Users/ian/.sdkman/candidates/java/current/bin:/Users/ian/.sdkman/candidates/gradle/current/bin:/Users/ian/.pyenv/shims:/Users/ian/.nvm/versions/node/v16.15.1/bin:/Users/ian/.gvm/pkgsets/go1.18.1/global/bin:/Users/ian/.gvm/gos/go1.18.1/bin:/Users/ian/.gvm/pkgsets/go1.18.1/global/overlay/bin:/Users/ian/.gvm/bin:/Users/ian/.gvm/bin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/MacGPG2/bin:/usr/local/share/dotnet:~/.dotnet/tools:/Library/Apple/usr/bin:/Users/ian/.cargo/bin:/Users/ian/.local/bin:/Users/ian/.local/bin:/Users/ian/jetbrains-shell-scripts:/Users/ian/flutter/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.69.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /Users/ian/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /Users/ian/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220715/xtensa-esp-elf-gdb/bin:/Users/ian/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220715/riscv32-esp-elf-gdb/bin:/Users/ian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin:/Users/ian/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s2-elf/bin:/Users/ian/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s3-elf/bin:/Users/ian/.espressif/tools/riscv32-esp-elf/esp-2022r1-RC1-11.2.0/riscv32-esp-elf/bin:/Users/ian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/Users/ian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/Users/ian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/Users/ian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /Users/ian/.espressif/python_env/idf5.0_py3.10_env/bin/python
Serial port (idf.port) /dev/cu.usbmodem1113201
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) /Users/ian/.espressif
Git Path (idf.gitPath) /opt/homebrew/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to /Users/ian/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220715/xtensa-esp-elf-gdb/bin: true
Access to /Users/ian/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220715/riscv32-esp-elf-gdb/bin: true
Access to /Users/ian/.espressif/tools/xtensa-esp32-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32-elf/bin: true
Access to /Users/ian/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s2-elf/bin: true
Access to /Users/ian/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-RC1-11.2.0/xtensa-esp32s3-elf/bin: true
Access to /Users/ian/.espressif/tools/riscv32-esp-elf/esp-2022r1-RC1-11.2.0/riscv32-esp-elf/bin: true
Access to /Users/ian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /Users/ian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /Users/ian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
[... 108 lines cut ...]
> **NOTE:** Use `idf.openOcdDebugLevel` configuration setting to 3 or more to show debug logging in OpenOCD server output.

> **NOTE:** Use `logLevel` in your <project-directory>/.vscode/launch.json to 3 or more to show more debug adapter output.

Please share the output of the channel related to your issue. For example ESP-IDF is used mostly for the extension setup process output.

**Logs**
Please share the log file which can be obtained from:

- Windows: `%USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log`
- Linux & MacOSX: `$HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log`

**Extension Configuration settings**
Open Command Palette (F1 or View Menu -> Command Palette) and type `Preferences: Open Settings (JSON)` or use `{CURRENT_PROJECT}/.vscode/settings.json` to get the following extension configuration settings value:

- `idf.espIdfPath` (MacOS or Linux) or `idf.espIdfPathWin` (Windows) also called `Path to locate ESP-IDF framework (IDF_PATH)` in `Preferences: Open Settings (UI)`.
- `idf.pythonBinPath` (MacOS or Linux) or `idf.pythonBinPathWin` (Windows) also called `Python absolute binary path used to execute ESP-IDF Python Scripts` in `Preferences: Open Settings (UI)`.
- `idf.openOcdConfigs` also called `List of configuration files inside OpenOCD Scripts directory` in `Preferences: Open Settings (UI)`.
- `idf.customExtraPaths` also called `Paths to be appended to PATH` in `Preferences: Open Settings (UI)`.
- `idf.customExtraVars` also called `Variables to be added to system environment variables` in `Preferences: Open Settings (UI)`.
- `idf.adapterTargetName` also called `Target name for ESP-IDF Debug Adapter` in `Preferences: Open Settings (UI)`.
- `idf.customAdapterTargetName` is used when `idf.adapterTargetName` is set to `custom`.

**Additional context**
Add any other context about the problem here.

iangregsondev, 2022-07-31:
I can run the following command manually inside the special terminal and things start to work..

```
. ~/esp/esp-idf/export.sh
```

Although this is a manual process and probably should not need to be done.

brianignacio5, 2022-08-11:
Hi @iangregsondev I've noticed that some tools were not being installed (those tar.xz) and the linked PR should fix the issue.

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/326564623.zip) and see if it works for you ?

closed 2022-08-11 by brianignacio5

arikon, 2023-05-01:
@brianignacio5 The problem remains in the last version of extension (v1.6.1)

arikon, 2023-05-02:
@brianignacio5 Created new issue https://github.com/espressif/vscode-esp-idf-extension/issues/959

crack-dawn, 2024-04-02:
Yes, indeed! For those using Visual Studio Code with the ESP-IDF extension, you can create a new ESP-IDF component without having to manually activate the ESP-IDF shell environment through the command line. Here's how:

1. Open your ESP-IDF project in Visual Studio Code.
2. Press `Ctrl + Shift + P` (or `Cmd + Shift + P` on macOS) to open the Command Palette.
3. Type "ESP-IDF: Create Component" in the search field of the Command Palette.
[image]

5. Select this command when it appears.
6. Input  your component name.
[image]
7. Here, two user-defined components have been created, one called LED and the other called nowMy.
[image]

This action will prompt you for the name of the component and where to create it within your ESP-IDF project directory structure. The extension takes care of activating the correct environment and running the necessary `idf.py` commands under the hood to create the component. 

By leveraging the ESP-IDF extension in VSCode, you benefit from integrated development environment features that simplify many common ESP-IDF development tasks.

PR #776 (2022-08-11): add tar xz to fix master idf tools install
## Description

Add extraction of `tar.xz` since before was only using `tar.gz` before to fix ESP-IDF v5.0 packages using tar.xz

Fixes #769 VSC-943

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

- Test A: Test by manually running `ESP-IDF: Configure ESP-IDF extension` wizard command.

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS Big Sur

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [x] Added Documentation
- [x] Added Unit Test
- [x] [... cut]