espressif/vscode-esp-idf-extension #770: [Bug Report]: Cannot select programming COM port

janzwiegers, 2022-08-01:
**Installation on absolute fresh Windows 11 copy**
Installation was done on a fresh Windows 11 copy.
Restarted OS, restarted VS Code.

**Describe the bug**
Cannot select programming port nor program the device
Build works fine

**To Reproduce**
Generate Blinky project
Select port

**VS Code Extension Log**
{"user":true,"message":"Something went wrong while getting the serial port list","stack":"TypeError: Cannot read properties of undefined (reading 'uri')\n\tat Object.<anonymous> (\\dist\\extension.js:1:19824)\n\tat Generator.next (<anonymous>)\n\tat s (\\dist\\extension.js:1:17926)\n\tat processTicksAndRejections (node:internal/process/task_queues:96:5)","level":"error","timestamp":"2022-08-01T08:26:09.743Z"}
{"user":true,"message":"Something went wrong while getting the serial port list","stack":"TypeError: Cannot read properties of undefined (reading 'uri')\n\tat Object.<anonymous> (\\dist\\extension.js:1:19824)\n\tat Generator.next (<anonymous>)\n\tat s (\\dist\\extension.js:1:17926)\n\tat processTicksAndRejections (node:internal/process/task_queues:96:5)","level":"error","timestamp":"2022-08-01T08:31:28.265Z"}
{"user":true,"message":"Something went wrong while getting the serial port list","stack":"TypeError: Cannot read properties of undefined (reading 'uri')\n\tat Object.<anonymous> (\\dist\\extension.js:1:19824)\n\tat Generator.next (<anonymous>)\n\tat s (\\dist\\extension.js:1:17926)\n\tat processTicksAndRejections (node:internal/process/task_queues:96:5)","level":"error","timestamp":"2022-08-01T08:31:57.720Z"}

**Expected behavior**
Device should program

**Environment**
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python310\Scripts\;C:\Python310\;C:\Program Files\Microsoft\jdk-17.0.3.7-hotspot\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files\Git\cmd;C:\Users\jan\AppData\Local\Microsoft\WindowsApps;C:\Users\jan\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files\JetBrains\IntelliJ IDEA 2022.1.4\bin;;C:\Users\jan\AppData\Local\GitHubDesktop\bin;C:\Users\jan\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.69.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) d:\Development\espressif\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) d:\Development\espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;d:\Development\espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;d:\Development\espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;d:\Development\espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;d:\Development\espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;d:\Development\espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;d:\Development\espressif\tools\cmake\3.20.3\bin;d:\Development\espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;d:\Development\espressif\tools\ninja\1.10.2;d:\Development\espressif\tools\idf-exe\1.0.3;d:\Development\espressif\tools\ccache\4.3\ccache-4.3-windows-64;d:\Development\espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"d:\\Development\\espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) d:\Development\espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s2.cfg
ESP-IDF Tools Path (idf.toolsPath) d:\Development\espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to d:\Development\espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to d:\Development\espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to d:\Development\espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to d:\Development\espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to d:\Development\espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to d:\Development\espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to d:\Development\espressif\tools\cmake\3.20.3\bin: true
Access to d:\Development\espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to d:\Development\espressif\tools\ninja\1.10.2: true
Access to d:\Development\espressif\tools\idf-exe\1.0.3: true
Access to d:\Development\espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to d:\Development\espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: "/usr/bin/git" --version
The system cannot find the path specified.

--------------------------------------------------------------------------------------------------------------------------------------------

**Resolve**
I need instructions to resolve this issue.
Out of the box installation did not work

brianignacio5, 2022-08-01:
There is an issue with the git path. Could you update `idf.gitPath` in your extension configuration setting ? or try to run the setup wizard again.

Also please share the extension log as specified in the issue template.

janzwiegers, 2022-08-01:
I ran setup wizard again an shows no issues at all.
The log is under **VS Code Extension Log**
Where is the settings stored for VC Code to update odf.gitPath?
I can see what is wrong but don't know where to fix it.
Under command palete -> ESP-IDF->Configure paths I only have IDF_PATH and IDF_TOOLS_PATH, both set correctly

brianignacio5, 2022-08-01:
User are located [here](https://code.visualstudio.com/docs/getstarted/settings#_settings-file-locations)
Settings can also exist in <project-directory>/.vscode/settings.json or a `.code-workspace` file

There could be some place overriding the `idf.gitPath`

janzwiegers, 2022-08-01:
CP->Preferences->Open Settings (UI)->Extensions->ESP-IDF help to find all the related settings.
VS Code not complaining about git any more, but still have same issue.
Will dig some more into the settings, just wish I knew where it failed.
Python maybe? The virtual environment path is also correct.

This is still the same log error:
{"user":true,"message":"Something went wrong while getting the serial port list","stack":"TypeError: Cannot read properties of undefined (reading 'uri')\n\tat Object.<anonymous> (\\dist\\extension.js:1:19824)\n\tat Generator.next (<anonymous>)\n\tat s (\\dist\\extension.js:1:17926)\n\tat processTicksAndRejections (node:internal/process/task_queues:96:5)","level":"error","timestamp":"2022-08-01T10:09:51.818Z"}

brianignacio5, 2022-08-01:
Could you also try the GitHub master version ? This is the [vsix installer link](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/315473533.zip)

janzwiegers, 2022-08-01:
OK great that fixed 99% of the issues!
I can configure PORT + UART via CP->Preferences->Open Settings (UI)->Extensions->ESP-IDF.
But the quick selection at the bottom, PORT & UART does not do anything.

But great help thanks!
I can program now.

closed 2022-08-01 by janzwiegers