espressif/vscode-esp-idf-extension #772: [Bug Report]: cannot use install latest version of esp-idf

v-3n0m, 2022-08-04:
**Describe the bug**
Not able to install the latest version of esp-idf  using vscode extension getting 404 error when downloading the version 4.4.2 .

**To Reproduce**
Install the release version 4.4. or latest 4.4 .

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**

- OS Name & Version: [Windows 11]
- VSCode Version: [e.g.1.4.0 ]
- ESP-IDF Version: [e.g. v4.4]
- Python Version: [e.g. 3.10]

brianignacio5, 2022-08-05:
Does this happens with GitHub or Espressif or both download mirrors?

v-3n0m, 2022-08-05:
@brianignacio5 yes it is happening with both GitHub and Espressif. Here is the screen shot of using github 
[image]

and this is using Espressif:
[image]

brianignacio5, 2022-08-05:
Thanks for reporting this. This is an issue in ESP-IDF itself, seems the release zip is not available.

Will ask for them to fix it asap.

v-3n0m, 2022-08-08:
Now it works fine thank you.

closed 2022-08-08 by v-3n0m