espressif/vscode-esp-idf-extension #775: "Configure the ESP-IDF extension"  on Ubuntu Jammy, 22.04 LTS

jonsmirl, 2022-08-06:
VSCode command "Configure the ESP-IDF extension" does not work on Ubuntu Jammy. It tells you to install the Linux prerequisites and sends you to a web page. I have all of those installed. I then installed the IDF from the command line and the IDF work fine.

jonsmirl, 2022-08-07:
Output from ESP Doctor
```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.0-43-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 /home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env 
System environment variable PATH 
 /home/jonsmirl/.local/bin:/home/jonsmirl/esp/esp-idf/components/esptool_py/esptool:/home/jonsmirl/esp/esp-idf/components/espcoredump:/home/jonsmirl/esp/esp-idf/components/partition_table:/home/jonsmirl/esp/esp-idf/components/app_update:/home/jonsmirl/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/jonsmirl/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/jonsmirl/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/jonsmirl/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/jonsmirl/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/jonsmirl/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin:/home/jonsmirl/esp/esp-idf/tools:/opt/android-studio/jre/bin:/home/jonsmirl/.nvm/versions/node/v18.3.0/bin:/home/jonsmirl/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/.environment/cipd/packages/pigweed/:/home/jonsmirl/aosp/esp-matter/connectedhomeip/connectedhomeip/examples/chip-tool/out/ 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.0 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.4.0 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/jonsmirl/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
Custom extra paths (idf.customExtraPaths) /home/jonsmirl/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/jonsmirl/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/jonsmirl/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/jonsmirl/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/jonsmirl/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/jonsmirl/.espressif/tools/cmake/3.20.3/bin:/home/jonsmirl/.espressif/tools/ninja/1.10.2:/home/jonsmirl/openocd-esp32/src
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/jonsmirl/openocd-esp32/tcl"}
Virtual env Python Path (idf.pythonBinPath) /home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/jonsmirl/esp/esp-idf
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-IDF Custom extra paths
Access to /home/jonsmirl/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/jonsmirl/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/jonsmirl/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to /home/jonsmirl/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/jonsmirl/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/jonsmirl/.espressif/tools/cmake/3.20.3/bin: false
Access to /home/jonsmirl/.espressif/tools/ninja/1.10.2: false
Access to /home/jonsmirl/openocd-esp32/src: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.34.1
ESP-IDF version 4.4.2
Python version 3.10.4
Python's pip version 22.2.2
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2022.6.15
cffi version: 1.15.1
charset-normalizer version: 2.1.0
click version: 8.1.3
colorama version: 0.4.5
construct version: 2.10.54
contextlib2 version: 21.6.0
cryptography version: 37.0.4
ecdsa version: 0.18.0
Flask version: 0.12.5
Flask-Compress version: 1.12
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
[... cut ...]
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: /home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/jonsmirl/esp/esp-idf/tools/check_python_dependencies.py -r "/home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt"

--------------------------------------------------------------------------------------------------------------------------------------------

```

jonsmirl, 2022-08-07:
I tried do this, as the requirements command suggests.
```
jonsmirl@ares:~$ /home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/bin/python -m pip install -r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt
/home/jonsmirl/.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages/_distutils_hack/__init__.py:30: UserWarning: Setuptools is replacing distutils.
  warnings.warn("Setuptools is replacing distutils.")
Ignoring pywin32: markers 'sys_platform == "win32"' don't match your environment
Ignoring typing: markers 'python_version < "3.5"' don't match your environment
Requirement already satisfied: setuptools>=21 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from -r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 1)) (62.3.2)
Requirement already satisfied: click in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from -r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 3)) (8.1.3)
Collecting psutil>=5.5.1
  Using cached psutil-5.9.1-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.manylinux_2_17_x86_64.manylinux2014_x86_64.whl (282 kB)
Requirement already satisfied: pygdbmi<=0.9.0.2 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from -r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 5)) (0.9.0.2)
Requirement already satisfied: requests>=2.21.0 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from -r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 7)) (2.28.1)
Collecting xmlrunner>=1.7.7
  Using cached xmlrunner-1.7.7-py3-none-any.whl
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 7)) (1.26.11)
Requirement already satisfied: idna<4,>=2.5 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 7)) (3.3)
Requirement already satisfied: charset-normalizer<3,>=2 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 7)) (2.1.0)
Requirement already satisfied: certifi>=2017.4.17 in ./.espressif/python_env/idf4.4_py3.10_env/lib/python3.10/site-packages (from requests>=2.21.0->-r /home/jonsmirl/.vscode/extensions/espressif.esp-idf-extension-1.4.0/esp_debug_adapter/requirements.txt (line 7)) (2022.6.15)
Installing collected packages: xmlrunner, psutil
Successfully installed psutil-5.9.1 xmlrunner-1.7.7
jonsmirl@ares:~$ 
```

jonsmirl, 2022-08-07:
Still fails using github master:
code --install-extension ./esp-idf-extension.vsix

jonsmirl, 2022-08-07:
I activated the espressif environment '. ./export.sh'

Then used pip to manually install:
gcovr
idf-component-manager
websocket_client

Now I am able to configure the vscode plugin.

brianignacio5, 2022-08-11:
Hi @jonsmirl I think the issue was that the previous default value for `idf.gitPath` was `git` instead of git's full path `/usr/bin/git`. This should have been fixed in GitHub master vsix.

Could you test this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/320305346.zip) ?

Related link: [how to Install extension from vsix](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

jonsmirl, 2022-08-11:
I removed my extension, restarted vcode, installed this one -- and everything works. But does removing the extension clear out the settings I had set? Because my system was working before I uninstalled the previous version due to me manually installing the packages.

The git command seems like a reasonable problem, the plugin would not be able to find it and say the prerequisites were not met. You probably need to test this on a fresh Jammy installation.

brianignacio5, 2022-08-11:
Removing the extension doesn't clear the settings. You could try to temporary remove settings from User Settings.json and try to open the Setup wizard and see if the error still there.

jonsmirl, 2022-08-11:
I renamed settings and the wizard is still there... but I made a lot of changes to my system trying to track down why I could not configure. Here's what was in my user settings...
```
{
    "cmake.configureOnOpen": true,
    "idf.espIdfPath": "/home/jonsmirl/esp/esp-idf",
    "idf.toolsPath": "/home/jonsmirl/esp/esp-idf",
    "idf.customExtraPaths": "/home/jonsmirl/esp/esp-idf/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/jonsmirl/esp/esp-idf/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/jonsmirl/esp/esp-idf/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/jonsmirl/esp/esp-idf/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/jonsmirl/esp/esp-idf/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/jonsmirl/esp/esp-idf/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/jonsmirl/esp/esp-idf/tools/cmake/3.23.1/bin:/home/jonsmirl/esp/esp-idf/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/jonsmirl/esp/esp-idf/tools/ninja/1.10.2",
    "idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"/home/jonsmirl/esp/esp-idf/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts\"}",
    "idf.openOcdConfigs": [
        "board/esp32c3-builtin.cfg"
    ],
    "idf.adapterTargetName": "esp32c3",
    "idf.enableCCache": true,
    "idf.enableIdfComponentManager": true,
    "idf.pythonBinPath": "/home/jonsmirl/esp/esp-idf/python_env/idf4.4_py3.10_env/bin/python"
}
```

brianignacio5, 2022-08-15:
I think the issue is fixed because the default value of `idf.gitPath` changed from `git` to `/usr/bin/git` which fixes the issue I think.

closed 2022-08-15 by brianignacio5