espressif/vscode-esp-idf-extension #782: Projects that used to build with previous extension version now report compile errors with 1.5.0

jmporter34, 2022-08-17:
### OS

Windows

### Operating System version

10

### Visual Studio Code version

1.70.1

### ESP-IDF version

4.4

### Python version

3.10.0

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python310\Scripts\;C:\Python310\;C:\ProgramData\LLVM for Renesas RL78 10.0.0.202104\bin;C:\Program Files (x86)\GCC for Renesas RL78 4.9.2.202002-GNURL78-ELF\rl78-elf\rl78-elf\bin;C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\TortoiseGit\bin;C:\Program Files\WinMerge;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files\dotnet\;C:\ProgramData\LLVM for Renesas RL78 10.0.0.202104\bin;D:\Git\cmd;D:\Python\WPy64-3940\python-3.9.4.amd64;C:\Users\porterj02\AppData\Local\Microsoft\WindowsApps;C:\Users\porterj02\AppData\Local\Programs\Git\cmd;C:\Users\porterj02\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\porterj02\AppData\Local\particle\bin;C:\OpenSSL-Win32\bin;C:\Users\porterj02\AppData\Roaming\npm; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.0 
Workspace folder c:\Code\Projects\ESP32_Common\apps\MFRC522_ReadWrite_Util 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/esp-idf/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\tools\cmake\3.16.4\bin;C:\Espressif\tools\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin;C:\Espressif\tools\tools\ninja\1.10.0;C:\Espressif\tools\tools\idf-exe\1.0.1;C:\Espressif\tools\tools\ccache\3.7;C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Espressif\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20210721/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/tools/python_env/idf4.3_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM26
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif\tools
Git Path (idf.gitPath) ${env:programfiles}\Git\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\tools\cmake\3.16.4\bin: true
Access to C:\Espressif\tools\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin: true
Access to C:\Espressif\tools\tools\ninja\1.10.0: true
Access to C:\Espressif\tools\tools\idf-exe\1.0.1: true
Access to C:\Espressif\tools\tools\ccache\3.7: true
Access to C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: false
Spaces in C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: false
Spaces in C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: false
Spaces in C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: false
Spaces in C:\Espressif\tools\tools\cmake\3.16.4\bin: false
Spaces in C:\Espressif\tools\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin: false
Spaces in C:\Espressif\tools\tools\ninja\1.10.0: false
Spaces in C:\Espressif\tools\tools\idf-exe\1.0.1: false
Spaces in C:\Espressif\tools\tools\ccache\3.7: false
Spaces in C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64: false
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: "${env:programfiles}\Git\cmd\git.exe" --version
The filename, directory name, or volume label syntax is incorrect.

--------------------------------------------------------------------------------------------------------------------------------------------

### Description

I have a bunch of ESP32 projects.  They all built in the ESP-IDF 4.4 CMD environment, and they all built with the previous VS Code ESP-IDF extension (I just used some of them last week - I could build, flash and debug all successfully).  This morning when I launched VS Code I got a message that ESP-IDF extension had been updated to version 1.5.0.  I've tried building several of my projects in VS Code afterward, and they all report compile errors (specifically, some include files can't be found - but again, they were being found with the previous ESP-IDF version).  All the projects still build successfully in the ESP-IDF 4.4 CMD environment.

### Debug Message

```plain
cannot open source file "delay.hpp"
cannot open source file "esp_log.h"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-08-18:
Could you modify `idf.gitPath` in your settings.json to `idf.gitPathWin` ? There were many issues in WSL due to having a single git path setting.

jmporter34, 2022-08-18:
Hi, here is what I see when I click the Settings icon in the lower left, then search for "idf.gitPath":

Idf: Git Path
Git executable path
C:/Espressif/tools/tools/idf-git/2.30.1/cmd/git.exe

Idf: Git Path Win
Git executable path
${env:programfiles}\Git\cmd\git.exe

Are you suggesting that I modify idf.gitPath so that its value is equivalent to the value of idf.gitPathWin ?
Or something else ?  Thanks

brianignacio5, 2022-08-18:
I mean the other way around.

Set ` idf.gitPathWin` to `C:/Espressif/tools/tools/idf-git/2.30.1/cmd/git.exe`

jmporter34, 2022-08-18:
I did as you suggested, and it sort of works.  If I open the Settings window in the editor, then close VS Code, then relaunch it, the PROBLEMS tab will be empty.  Then I click the Build icon, the little "Building Project" window pops up, and a few seconds later it disappears.  It doesn't leave the "Build Completed" message on screen like it used to though, instead I just see what appears to be a map file summary in the TERMINAL tab.  Do I infer from that (and from the empty PROBLEMS tab) that the build was successful ?

Then if I close the Settings window, or leave it open but bring a different file to the foreground in the editor, the PROBLEMS tab immediately shows a non-zero number, and a list of include files that it can't find if I bring it to the foreground.  If I now click the Build icon again, the same little "Building Project" window pops up and disappears again after a few seconds.  No "Build Completed" or "Build Failed" message remains, but I see the same map file summary in the TERMINAL tab.  Did it succeed or fail ?  No idea.

brianignacio5, 2022-08-19:
Hi @jmporter34 with regards to the `Build Successfully` message, the thing is that the `idf.notificationSilentMode` now is `true` by default (which hides those Build successfully and build fails messages). If you want to see these messages just set `"idf.notificationSilentMode": false` in your settings.json

jmporter34, 2022-08-25:
Hi @brianignacio5   So I turned off silent mode as you suggested, and the "Build successfully" message now appears for a few seconds before disappearing (as it used to with the previous version).  But I'm still seeing all these "cannot open source file" messages in the PROBLEMS tab, which I wasn't seeing with the previous version.  I've attached a screenshot that shows the contradictory "Build successfully" message and the "cannot open source file messages".  Any idea what's causing all the so-called problems that don't actually prevent the project from building ?  Thanks
[image]

brianignacio5, 2022-08-25:
These problems are generated by the Microsoft C/C++ extension. What is the content of `.vscode/c_cpp_properties.json` ?

This is a bit tricky thing to properly configure (using compile_commands,json as in C_CPP configuration should work well for most cases) with a bit of tinkering.

jmporter34, 2022-08-25:
I don't have a c_cpp_properties.json for this project.  Only launch.json,
settings.json and tasks.json

brianignacio5, 2022-08-29:
Seems the original issue is fixed. For the errors in the Problems tab seems to be related to the extension you are using for C/C++ resolution which is not done but our extension as explained in the previous documentation link.

closed 2022-08-29 by brianignacio5

jmporter34, 2022-09-02:
@brianignacio5 
I created a c_cpp_properties.json for this project, using the Default configuration at
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/C_CPP_CONFIGURATION.md
and then adding my specific include paths and compiler path.  All of the "can't find include file" errors in the PROBLEMS window are gone now. Great! Thanks!

But... every time I open the project folder, I see the messages that I've pasted below in the OUTPUT window.
Do you know where those are coming from, and how I can get rid of them ?
(again, they don't stop the project from building - they're just an eyesore)

'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking riscv32-esp-elf version
Error: Command failed: riscv32-esp-elf-gcc --version
'riscv32-esp-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.
Error checking xtensa-esp32s3-elf version
Error: Command failed: xtensa-esp32s3-elf-gcc --version
'xtensa-esp32s3-elf-gcc' is not recognized as an internal or external command,
operable program or batch file.

Error checking idf-exe version
TypeError: Cannot read properties of null (reading 'length')

brianignacio5, 2022-09-06:
This comes from the extension activation that checks if the tools are available in several paths. I'll try to see why it keeps showing if the extension is already configured.