espressif/vscode-esp-idf-extension #784: Heap Tracing with Example is not working

mbrunnen, 2022-08-19:
### OS

Linux

### Operating System version

Debian GNU/Linux 11 (bullseye)

### Visual Studio Code version

1.70.2

### ESP-IDF version

4.4.2

### Python version

3.9.2

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.10.0-16-amd64 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/mcuxpressoide/ide:/home/brmo/.local/opt/silabs/slc_cli/bin/slc-cli:/home/brmo/.local/opt/STM32CubeProgrammer/bin:/home/brmo/.local/opt/STM32CubeMX/STM32CubeMX:/home/brmo/.config/emacs/bin:/opt/jabref/bin:/home/brmo/.local/opt/perl5/lib:/home/brmo/.local/opt/perl5/bin:/home/brmo/.local/opt/go/bin:/home/brmo/.local/opt/cargo/bin:/home/brmo/.local/opt/npm/bin:/home/brmo/.local/opt/gem/bin:/home/brmo/.local/bin:/home/brmo/.local/opt/cabal/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.2 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/zsh 
ESP-IDF Extension version 1.5.0 
Workspace folder /home/brmo/projects/boxine/src/private/sysview_tracing_heap_log 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/brmo/.local/opt/espressif/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) /home/brmo/.local/opt/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/brmo/.local/opt/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/brmo/.local/opt/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/brmo/.local/opt/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/brmo/.local/opt/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/brmo/.local/opt/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/brmo/.local/opt/espressif/tools/cmake/3.23.1/bin:/home/brmo/.local/opt/espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/brmo/.local/opt/espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/brmo/.local/opt/espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/brmo/.local/opt/espressif/python_env/idf4.4_py3.9_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) .vscode/esp32s3-jlink.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/brmo/.local/opt/espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to /home/brmo/.local/opt/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/brmo/.local/opt/espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/brmo/.local/opt/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/brmo/.local/opt/espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to /home/brmo/.local/opt/espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/brmo/.local/opt/espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/brmo/.local/opt/espressif/tools/cmake/3.23.1/bin: false
Access to /home/brmo/.local/opt/espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin: true
Access to /home/brmo/.local/opt/espressif/tools/ninja/1.10.2: false
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
[... 213 lines cut ...]
Info : esp32s3.cpu0: Core was reset.
[OpenOCD]
Info : esp32s3.cpu0: Target halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : esp32s3.cpu1: Debug controller was reset.
Info : esp32s3.cpu1: Core was reset.
[OpenOCD]
Info : esp32s3.cpu1: Target halted, PC=0x40000400, debug_reason=00000000
[OpenOCD]
Info : esp32s3.cpu0: Target halted, PC=0x42005098, debug_reason=00000001
[OpenOCD]
Info : Set GDB target to 'esp32s3.cpu0'
[OpenOCD]
Info : esp32s3.cpu1: Target halted, PC=0x40043A3E, debug_reason=00000000
[OpenOCD]
Info : Total trace memory: 16384 bytes
[OpenOCD]
❌ Error: Not enough args! Need 2 trace data destinations!
Error: Failed to init cmd ctx (-4)!
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

mbrunnen, 2022-08-19:
The error disappeared, when I changed `gdbinit` from
`mon esp sysview start file:///tmp/heap_log.svdat`
to
`mon esp sysview start file:///tmp/heap_log.cpu0.svdat file:///tmp/heap_log.cpu1.svdat`

radurentea, 2022-09-09:
Hi @mbrunnen,

Thank you for the report! 

I've looked into it and fixed the issue on my local, however, there is another issue regarding heap tracing coming from the ESP-IDF, which will be fixed with the ESP-IDF 5.1 release. Until the release there is no workaround I'm afraid for the ESP32-S3 chip.

mbrunnen, 2022-09-11:
Hi @radurentea, these are good news, thank you! Do you have a reference for the issue in IDF?

mbrunnen, 2022-09-14:
Hi @radurentea, thank you for that information! Unfortunately this commit seems to be released in v5.1: 
[image]

closed 2022-10-06 by github-actions[bot]

mickeyl, 2024-03-22:
May I ask about the current status here? I tried with ESP-IDF 5.2.1 and the latest version of the extension (1.7.1 as of today) and it doesn't work here. My example hangs in `esp_heap_trace_start()`.

OpenOCD debugging in general works though.

0xjmux, 2024-04-07:
I'm also interested in the status of this. 

I was able to get the program to execute (nominally) by modifying `gdbinit` like mbrunned mentioned in [this reply](https://github.com/espressif/vscode-esp-idf-extension/issues/784#issuecomment-1220883858). 

GDB Log:
```
xtensa-esp32-elf-gdb -x gdbinit build/sysview_tracing_heap_log.elf
GNU gdb (esp-gdb) 12.1_20231023
...

Reading symbols from build/sysview_tracing_heap_log.elf...
0x4037a416 in esp_cpu_wait_for_intr () at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/esp_hw_support/cpu.c:145
145     }
JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[esp32s3.cpu0] requesting target halt and executing a soft reset
[esp32s3.cpu0] Debug controller was reset.
[esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[esp32s3.cpu0] Reset cause (3) - (Software core reset)
[esp32s3.cpu1] requesting target halt and executing a soft reset
[esp32s3.cpu0] Core was reset.
[esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Debug controller was reset.
[esp32s3.cpu1] Core was reset.
[esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
[esp32s3.cpu1] Reset cause (3) - (Software core reset)
[esp32s3.cpu0] Reset cause (3) - (Software core reset)
Temporary breakpoint 1 at 0x42006040: file /home/jmux/Documents/esp/v5.2.1/esp-idf/components/app_trace/heap_trace_tohost.c, line 36.
Note: automatically using hardware breakpoints for read-only addresses.
Temporary breakpoint 2 at 0x42006058: file /home/jmux/Documents/esp/v5.2.1/esp-idf/components/app_trace/heap_trace_tohost.c, line 48.
[esp32s3.cpu0] Target halted, PC=0x42006040, debug_reason=00000001
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x40043A40, debug_reason=00000000
[New Thread 1070178928]
[New Thread 1070182088]
[New Thread 1070174276]
[Switching to Thread 1070178928]

Thread 2 "main" hit Temporary breakpoint 1, heap_trace_start (mode_param=(HEAP_TRACE_LEAKS | unknown: 0x2)) at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/app_trace/heap_trace_tohost.c:36
36      {
Total trace memory: 16384 bytes
Open file /tmp/heap_log.cpu0.svdat
Open file /tmp/heap_log.cpu1.svdat
App trace params: from 2 cores, size 4294967295 bytes, stop_tmo -1 s, poll period 0 ms, wait_rst 0, skip 0 bytes
Connect targets...
Targets connected.
[esp32s3.cpu0] Target halted, PC=0x42006058, debug_reason=00000001
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x40043A40, debug_reason=00000000
[New Thread 1070197012]
[New Thread 1070191164]
[New Thread 1070185316]
[New Thread 1070199984]
[New Thread 1070194136]
[New Thread 1070188288]

Thread 2 "main" hit Temporary breakpoint 2, heap_trace_stop () at /home/jmux/Documents/esp/v5.2.1/esp-idf/components/app_trace/heap_trace_tohost.c:48
48      {
Resume targets
16500 
16753 
16996 
17072 
Disconnect targets...
[esp32s3.cpu0] Target halted, PC=0x4038046F, debug_reason=00000000
Set GDB target to 'esp32s3.cpu0'
[esp32s3.cpu1] Target halted, PC=0x40043A40, debug_reason=00000000
Targets disconnected.
Tracing is STOPPED. Size is 17072 of -1 @ 76.483856 (73.379166) KiB/s
Data: blocks incomplete 0, lost bytes: 0
Block read time [126.265007..126.265007] ms
Block proc time [0.082000..0.082000] ms
(gdb)
```
Following along with the steps in the example, I then open it in Systemview - but when opening either heap log in Systemview the program hangs and refuses to read it. 
[image]

I have two questions:
1. Is the reason for this incompatibility known, and is there a fix for it? I've tried copying `SYSVIEW_FreeRTOS.txt` to `/opt/SEGGER/SystemView_vX`, but it didn't help; the heap log being produced seems to be incompatible with Systemview, but all of Segger's examples load fine on my machine. 
2.  Are there plans to make the GUI buttons for these tools in the VSCode dev environment actually work? So far, basically every debugging tool in ESP-IDF I've tried only works when running it from the CLI. The VSCode extension looks very nice, but at this point I'm spending more time banging my head against my desk trying to get it to work than actually writing code. 

Thanks.