espressif/vscode-esp-idf-extension #786: [Feature Request]: Allow setting a custom build path by conf setting

gfurtadoalmeida, 2022-08-24:
**Is your feature request related to a problem? Please describe.**
As the build is heavy on disk usage, it is, depending on the project, very slow.
The #668 work allows setting a custom build folder, but relative to the workspace.

**Describe the solution you'd like**
Being able to set a custom build path, that for speed, would be on a [RAM Drive](https://en.wikipedia.org/wiki/RAM_drive).

**Describe alternatives you've considered**
Moving the entire project to a RAM Drive.

**Additional context**
For testing on Windows, one could use the [ImDisk Toolkit](https://sourceforge.net/projects/imdisk-toolkit/)

brianignacio5, 2022-08-30:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/345066569.zip) ?

[how to Install extension from vsix](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix)

gfurtadoalmeida, 2022-08-31:
Sure! I'll do it first thing in the morning and post the result.

brianignacio5, 2022-09-22:
Sorry for late reply. Could you take a look at this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/372458811.zip) ?

gfurtadoalmeida, 2022-09-23:
No problems :thumbsup:
The build is working now, both in-memory and on-disk!
I just need to verify if it will flash from memory too.

gfurtadoalmeida, 2022-09-23:
Everything is working, build and flash!
We might close this issue.
Thank you @brianignacio5 !

gfurtadoalmeida, 2022-10-11:
The [last vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/372458811.zip) is working as expected.
The work to fix the issue is being done on #794.

PR #794 (2022-10-24): use absolute build path over workspace relative
## Description

Use `idf.buildPath` as absolute build path instead of relative path.

Fixes #786 

## Type of change

- New feature (non-breaking change which adds functionality)

## How has this been tested?

Run existing extension commands and test the behavior has not changed.

- Manual Test

**Test Configuration**:
* ESP-IDF Version: v4.4.2
* OS (Windows,Linux and macOS): macOS

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [x] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Windows,Linux and  [... cut]
closed 2022-10-11 by gfurtadoalmeida