espressif/vscode-esp-idf-extension #789: Project builds for ESP32 target, fails to build for ESP32S3 target

jmporter34, 2022-08-25:
### OS

Windows

### Operating System version

10

### Visual Studio Code version

1.70.2

### ESP-IDF version

4.4.3

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python310\Scripts\;C:\Python310\;C:\ProgramData\LLVM for Renesas RL78 10.0.0.202104\bin;C:\Program Files (x86)\GCC for Renesas RL78 4.9.2.202002-GNURL78-ELF\rl78-elf\rl78-elf\bin;C:\Program Files\Python39\Scripts\;C:\Program Files\Python39\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\TortoiseGit\bin;C:\Program Files\WinMerge;C:\Program Files\nodejs\;C:\ProgramData\chocolatey\bin;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files\dotnet\;C:\ProgramData\LLVM for Renesas RL78 10.0.0.202104\bin;D:\Git\cmd;D:\Python\WPy64-3940\python-3.9.4.amd64;C:\Users\porterj02\AppData\Local\Microsoft\WindowsApps;C:\Users\porterj02\AppData\Local\Programs\Git\cmd;C:\Users\porterj02\AppData\Local\Programs\Microsoft VS Code\bin;C:\Users\porterj02\AppData\Local\particle\bin;C:\OpenSSL-Win32\bin;C:\Users\porterj02\AppData\Roaming\npm; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.2 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.0 
Workspace folder c:\Code\Projects\ESP32_Common\apps\Ttpventus_Pump_Util 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/esp-idf/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin;C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Espressif\tools\tools\cmake\3.16.4\bin;C:\Espressif\tools\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin;C:\Espressif\tools\tools\ninja\1.10.0;C:\Espressif\tools\tools\idf-exe\1.0.1;C:\Espressif\tools\tools\ccache\3.7;C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Espressif\\tools\\tools\\openocd-esp32\\v0.10.0-esp32-20210721/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/tools/python_env/idf4.3_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM7
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif\tools
Git Path (idf.gitPath) C:/Espressif/tools/tools/idf-git/2.30.1/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\tools\xtensa-esp32s2-elf\esp-2020r3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Espressif\tools\tools\cmake\3.16.4\bin: true
Access to C:\Espressif\tools\tools\openocd-esp32\v0.10.0-esp32-20210721\openocd-esp32\bin: true
Access to C:\Espressif\tools\tools\ninja\1.10.0: true
Access to C:\Espressif\tools\tools\idf-exe\1.0.1: true
Access to C:\Espressif\tools\tools\ccache\3.7: true
Access to C:\Espressif\tools\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
[... 95 lines cut ...]
} 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open 'c:\Code\Projects\ESP32_Common\apps\Ttpventus_Pump_Util\.vscode\c_cpp_properties.json'
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

I have a project that builds successfully with both the ESP-IDF CMD and the VS Code ESP-IDF extension, when targeting ESP32.  If I change the target to ESP32S3, make appropriate changes to Kconfig.projbuild and sdkconfig, and do a rebuild all in CMD, it builds successfully.  But then it fails to build in VS Code, as follows in Debug Message.

### Debug Message

```plain
Traceback (most recent call last):
  File "C:\Espressif\esp-idf\tools\idf_size.py", line 923, in <module>
    main()
  File "C:\Espressif\esp-idf\tools\idf_size.py", line 383, in main
    mem_regions = MemRegions(detected_target)
  File "C:\Espressif\esp-idf\tools\idf_size.py", line 133, in __init__
    raise RuntimeError('Target {} is not implemented in idf_size'.format(target))
RuntimeError: Target esp32s3 is not implemented in idf_size
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dobairoland, 2022-08-30:
Please reconfigure your Vscode plugin and link it to a right version of ESP-IDF.

In the output you can see that it uses:
> ESP-IDF version 4.2.1

ESP32-S3 is supported starting from 4.4.

brianignacio5, 2022-08-30:
As you can you see in the vscode doctor command output the ESP-IDF version is `4.2.1` which is different from your command line configuration. Please update `idf.espIdfPathWin` in your vscode settings with the 4.4.3 version either manually update settings.json or with the setup wizard.

jmporter34, 2022-08-31:
Oh I see now.  I was using ESP-IDF command line tools 4.2 when I installed the VS Code extension.  Then later I upgraded the command line tools to 4.3, then later to 4.4.  But I never pointed the extension at them.  Everything works now.  Thank you for the help @brianignacio5  and @dobairoland

jmporter34, 2022-08-31:
Sorry, I closed the issue after the project built and I flashed the target over UART.  When I changed the flash method to JTAG and attempted to flash, it failed with the following:

❌ Error: JTAG sc[OpenOCD]
❌ an chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: esp32.cpu0: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
[OpenOCD]
❌ Error: Couldn't halt tar[OpenOCD]
get before SoC reset
** Flashing Failed **
-1

I was previously able to flash (and debug) successfully over JTAG when the target was ESP32.  I've double and triple checked the wiring between the ESP-PROG and the ESP32S3.  Is there some other setting that you're aware of that I need to change in order to be able to flash and debug the ESP32S3 over JTAG ?  Thank you.

brianignacio5, 2022-09-01:
Did you also update the openOCD config files ? Use the `ESP-IDF: Select OpenOCD Board Configuration` and select ESP32-S3 with ESP-PROG from the list.

jmporter34, 2022-09-01:
I had not done that.  I did it a little while ago, but it didn't seem to make any difference, so apparently there's some other issue as well.  Still working on it.
I'm curious, is changing the OCD setting something that I'm going to have to do manually every time I change target ?  (i.e. is it a global setting ?  I have some projects that target ESP32, and a couple of new ones that target ESP32S3).  Or is there a way to save that in the project folder so that it changes automatically when I load a different project ?  Thank you.

brianignacio5, 2022-09-06:
Unfortunately yes because even for a single target like esp32 you can have several board options for openOCD (each board might have different configuration if not using esp-prog). Trying to integrate them into a single command though.

The corresponding setting (`idf.openOcdConfigs`) could be defined per project. Right now this setting uses the `idf.saveScope` setting to define where to save a new value when using `ESP-IDF: Select OpenOCD Board Configuration` but will update to be saved in the workspace folder by default.

brianignacio5, 2022-09-06:
Will close this issue since the original issue is fixed. Let us know if you need any additional help.

closed 2022-09-06 by brianignacio5

jmporter34, 2022-09-06:
@brianignacio5 Sounds good, thanks again for all the help!