espressif/vscode-esp-idf-extension #791: `AttributeError: module 'esptool' has no attribute '_main'` at idf5.+

Coloryr, 2022-08-26:
### OS

Windows

### Operating System version

win11

### Visual Studio Code version

1.70

### ESP-IDF version

5.0

### Python version

3.8.7

### Doctor command output

[image]

### Description

[https://github.com/espressif/esp-idf/issues/9646#issue-1352065510](in here)

### Debug Message

```plain
not
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-08-29:
This seems to be an issue with the virtual environment you are using. Try deleting the virtual environment and install it again.

brianignacio5, 2022-09-13:
Closing this issue due to lack of response from user. Please comment here if you need further assistance.

closed 2022-09-13 by brianignacio5

Jeff-Conley, 2022-09-13:
I am deleted the entire esp-tools folder and reinstalled it fresh which should have installed a new python environment. I am still getting the error message after reinstalling.

brianignacio5, 2022-09-14:
Could you share the ESP-IDF: Doctor Command Output @Jeff-Conley ?

Jeff-Conley, 2022-09-14:
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22000 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\TortoiseSVN\bin;C:\Program Files\Microchip\xc8\v2.31\bin;C:\Program Files\Microchip\xc32\v4.00\bin;C:\Program Files\PuTTY\;C:\Program Files\Microchip\xc8\v2.35\bin;C:\Program Files\Microchip\xc16\v2.00\bin;C:\Program Files (x86)\Atmel\Studio\7.0\atbackend;C:\Program Files\dotnet\;C:\Program Files\Git\cmd;C:\Users\Jeffrey Conley\AppData\Local\Microsoft\WindowsApps;C:\Users\Jeffrey Conley\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.71.1 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.0 
Workspace folder c:\test_code\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) c:\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) c:\esp-idf-tools\esp-mdf
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:\esp-tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;c:\esp-tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;c:\esp-tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;c:\esp-tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;c:\esp-tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;c:\esp-tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;c:\esp-tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;c:\esp-tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;c:\esp-tools\tools\cmake\3.23.1\bin;c:\esp-tools\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;c:\esp-tools\tools\ninja\1.10.2;c:\esp-tools\tools\idf-exe\1.0.3;c:\esp-tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;c:\esp-tools\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"c:\\esp-tools\\tools\\openocd-esp32\\v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\esp-tools\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM4
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32s3.cfg
ESP-IDF Tools Path (idf.toolsPath) c:\esp-tools
Git Path (idf.gitPath) c:\esp-tools\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:\esp-tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to c:\esp-tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to c:\esp-tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to c:\esp-tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to c:\esp-tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to c:\esp-tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to c:\esp-tools\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to c:\esp-tools\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to c:\esp-tools\tools\cmake\3.23.1\bin: true
Access to c:\esp-tools\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin: true
Access to c:\esp-tools\tools\ninja\1.10.2: true
Access to c:\esp-tools\tools\idf-exe\1.0.3: true
Access to c:\esp-tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to c:\esp-tools\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path true
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in c:\esp-tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: false
Spaces in c:\esp-tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: false
Spaces in c:\esp-tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: false
Spaces in c:\esp-tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: false
[... cut ...]
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
--------------------------------------------------------------------------------------------------------------------------------------------

kurau1, 2022-09-26:
Hello, @Jeff-Conley 
Do you solve the problem?
i am facing the same issue. i reinstall windows, reinstall VScode, it didn't work.
i wanna use idf5! it can bulid project successfully but can't flash device. and give this information in terminal.
```
Traceback (most recent call last):
  File "f:\ESP32\idf\esp-idf\components\esptool_py\esptool\esptool.py", line 22, in <module>
    esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'
```
i have reinstall windows and vscode so many times. i install esp-idf v4.4 in VScode, it worked, can build and flash. but when i install esp-idf v5. i can't flash! it's a torture.

brianignacio5, 2022-09-26:
This seems to be happening in ESP-IDF itself as shown in [here](https://github.com/espressif/esp-idf/issues/8979). Could you delete the `esptool_py` directory  pulling the  sub component again and your espressif's `python_env` virtual env to reinstall faulty python packages.

xiaolongba, 2022-10-05:
Hello @brianignacio5 

I also ran into this issue and followed your tips above, but the issue is still present 

> Could you delete the `esptool_py` directory pulling the sub component again and your espressif's `python_env` virtual env to reinstall faulty python packages

it seems that the workaround mentioned is not valid. could you give me a hand?
[report.txt](https://github.com/espressif/vscode-esp-idf-extension/files/9716923/report.txt)

xiaolongba, 2022-10-05:
> So it isn't a great workaround but I got it working by creating a new windows user on my computer named "build" and installed it fresh using that. Seems to be working again on that account but still not working on my main account.

it is magic😂

xiaolongba, 2022-11-03:
@brianignacio5 

Hello Bro, any updates about it? the issue is still present. the guy from ESP-IDF told me that the root is caused by the vscode-extension, while as you mentioned above, it seems to be related with ESP-IDF😂

any other workarounds can share with me? thanks in advance.

brianignacio5, 2022-11-03:
Hey @xiaolongba I'm trying to find the reason why is it failing in the extension xD. Not sure what is making it break from extension.

xiaolongba, 2022-11-03:
> Hey @xiaolongba I was busy with other tasks but today I'll try to find the reason why is it failing in the extension xD. Not sure what is making it break from extension.

perfect. many thanks for your prompt replies

xiaolongba, 2022-11-07:
@brianignacio5 
Hello Bro,

it seems that there is one guy who gave us one workaround I have a try just now and it works. the details as this link:https://github.com/espressif/esp-idf/issues/9861#issuecomment-1304592452