espressif/vscode-esp-idf-extension #795: "Term 'dfu-util' is not recognized as the name of a cmdlet, ..." error when flashing ESP32S2 through USB

OniasC, 2022-08-31:
### OS

Windows

### Operating System version

win 10

### Visual Studio Code version

1.70.2

### ESP-IDF version

4.4.2

### Python version

3.8

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files\PuTTY\;C:\Program Files (x86)\Graphviz 2.28\bin;C:\Program Files\dotnet\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files (x86)\Graphviz 2.28\bin;C:\Users\onias\AppData\Local\Programs\MiKTeX\miktex\bin\x64\;C:\Users\onias\AppData\Local\Programs\Python\Python36-32;C:\Program Files (x86)\Nmap;C:\Program Files\CMake\bin;C:\Users\onias\AppData\Local\Programs\Microsoft VS Code\bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.2 
Visual Studio Code language en 
Visual Studio Code shell C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.0 
Workspace folder c:\Users\onias\OneDrive\Documents\esp32_study\hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:\Users\onias\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\onias\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Users\onias\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\onias\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\onias\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\onias\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\onias\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\onias\.espressif\tools\cmake\3.23.1\bin;C:\Users\onias\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin;C:\Users\onias\.espressif\tools\ninja\1.10.2;C:\Users\onias\.espressif\tools\idf-exe\1.0.3;C:\Users\onias\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\onias\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\onias\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:\Users\onias\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM7
OpenOCD Configs (idf.openOcdConfigs) board/esp32s2-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Users\onias\.espressif
Git Path (idf.gitPath) C:\Users\onias\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Users\onias\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin: true
Access to C:\Users\onias\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:\Users\onias\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:\Users\onias\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin: true
Access to C:\Users\onias\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:\Users\onias\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:\Users\onias\.espressif\tools\cmake\3.23.1\bin: true
Access to C:\Users\onias\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin: true
Access to C:\Users\onias\.espressif\tools\ninja\1.10.2: true
Access to C:\Users\onias\.espressif\tools\idf-exe\1.0.3: true
Access to C:\Users\onias\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:\Users\onias\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
[... 183 lines cut ...]

I followed (or tried to follow) the steps mentioned in:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/dfu.html#api-guide-dfu-flash-win
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/dfu.html#api-guide-dfu-flash

I am missing something. Is this not supported through vs code at the moment?

### Debug Message

```plain
dfu-util : The term 'dfu-util' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
At line:1 char:1
+ dfu-util -d 303a:2 -D c:\Users\onias\OneDrive\Documents\esp32_study\h ...
+ ~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (dfu-util:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException
```

### Other Steps to Reproduce

I tried programming another board through UART so I believe my configurations at at least somewhat correct.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2022-09-06:
Hi @OniasC!

Flashing using DFU is supported.
I've noticed some configurations issues with your setup from the Doctor Command.

- your environment path has spaces in it, which can lead to a lot of issues, try to remove them.
- I also see that you are missing paths for NINJA and CMake which makes me think you haven't properly installed all the prerequisites, I suggest you to use the [ESP-IDF Tools Installer](https://docs.espressif.com/projects/esp-idf/en/v4.2.1/esp32/get-started/windows-setup.html#get-started-windows-tools-installer) as explained in our documentation [here](https://docs.espressif.com/projects/esp-idf/en/v4.2.1/esp32/get-started/index.html)

radurentea, 2022-09-15:
Closing the issue due to inactivity. Feel free to re-open it if the problem persists

closed 2022-09-15 by radurentea