espressif/vscode-esp-idf-extension #796: error when trying to run menuconfig

T0biasCZe, 2022-09-04:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

1.71

### ESP-IDF version

4.3

### Python version

3.8.7

### Doctor command output

doesnt output anything

### Description

i am folowing the blink example tutorial, and when clicking ESP IDF SDK Configuration editor, it throws 
`[SDK Configuration Editor]
---------------------------ERROR--------------------------

Error: spawn C:\Users\user\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
-----------------------END OF ERROR-----------------------
`
when running tghe menuconfig from terminal, it says missing module click
[image]

### Debug Message

```plain
cant find the file
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-09-05:
Seems your python virtual environment is missing some packages. Could you share the `ESP-IDF Doctor Command` output ? When you run this command the result is copied in the clipboard and you can paste it anywhere.

brianignacio5, 2022-09-13:
Closing this issue for lack of response. Comment here if you still have any issues.

closed 2022-09-13 by brianignacio5