espressif/vscode-esp-idf-extension #797: Debugging an stm32s3 currently not working from VSCode

sbithinx, 2022-09-06:
### OS

Linux

### Operating System version

Manjaro Linux 21.3.7

### Visual Studio Code version

1.71.0

### ESP-IDF version

4.4.2

### Python version

3.10.5

### Doctor command output

Doctor command in VSCode is displaying not output.

### Description

Debugging using the Expressif VSCode Extension is currently not working for esp32s3. I'm using Manjaro Linux and a JLink Plus for debugging.
Using the 'openocd' and 'gdb' via command-line it is possible to debug the esp32s3. Starting 'openocd" via VSCode and debugging via 'gdb' from command-line works as well. But starting 'openocd' and 'gdb' via VSCode or starting 'openocd' via command-line and trying to connect to it via VSCode, does not work.
This seams to be related to the ESP-IDF Debugger of VSCode and its port settings. Unfortunately the documentation increases the confusion about solving this issue by myself.

### Debug Message

```plain
NA
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sbithinx, 2022-09-06:
Unfortunately it is not possible to create an issue, if all the requested log are put into the issue report, because it is to long.

sbithinx, 2022-09-06:
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/9495381/esp_idf_vsc_ext.log)
[debug.log](https://github.com/espressif/vscode-esp-idf-extension/files/9495389/debug.log)

brianignacio5, 2022-09-07:
The doctor command output is copied to the clipboard. Seems that the openOCD in the extension is not configured properly.

sbithinx, 2022-09-07:
Output of the doctor command:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.60-1-MANJARO 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /home/sebastian/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/home/sebastian/.dotnet/tools:/var/lib/flatpak/exports/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/var/lib/snapd/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.71.0 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/zsh 
ESP-IDF Extension version 1.5.0 
Workspace folder /home/sebastian/workspace/eu-pd 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/sebastian/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) /home/sebastian/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/sebastian/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/sebastian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/sebastian/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/sebastian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/sebastian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/sebastian/.espressif/tools/cmake/3.23.1/bin:/home/sebastian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/sebastian/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/sebastian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/sebastian/.espressif/python_env/idf4.4_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) config/esp32s3-jlink.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/sebastian/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to /home/sebastian/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/sebastian/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/sebastian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/sebastian/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to /home/sebastian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/sebastian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/sebastian/.espressif/tools/cmake/3.23.1/bin: true
Access to /home/sebastian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin: true
Access to /home/sebastian/.espressif/tools/ninja/1.10.2: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in /home/sebastian/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: false
Spaces in /home/sebastian/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: false
Spaces in /home/sebastian/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: false
Spaces in /home/sebastian/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: false
Spaces in /home/sebastian/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: false
Spaces in /home/sebastian/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: false
Spaces in /home/sebastian/.espressif/tools/cmake/3.23.1/bin: false
Spaces in /home/sebastian/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin: false
Spaces in /home/sebastian/.espressif/tools/ninja/1.10.2: false
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.37.2
ESP-IDF version 4.4.2
[... cut ...]
//       "externalConsole": false,
//       "logging": {
//         "engineLogging": true
//       }
//     }
//   ]
// } 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at ENOENT: no such file or directory, open '/home/sebastian/workspace/eu-pd/.vscode/c_cpp_properties.json'
--------------------------------------------------------------------------------------------------------------------------------------------

```

sbithinx, 2022-09-07:
Another error message of the Debug Adapter:

```
Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
Starting. Cmd: /home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter_main.py -d 3 -e /home/sebastian/workspace/eu-pd/build/toniebox-esp32-pd-dev-debug.elf -l /home/sebastian/workspace/eu-pd/debug.log -p 43474 -dn esp32s3 -a 0x180000 -t xtensa-esp32s3-elf-

Listening on port 43474
Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
Debug Adapter initialized
2
Failed to connect to "remote 127.0.0.1:3333"!
```

brianignacio5, 2022-09-07:
Based on the doctor command output launch.json, the `mode": "manual",` will not launch openOCD, that's why when you start the debug adapter it is failing to connect.

sbithinx, 2022-09-07:
This are out-commented configs for testing relevant is only:

```
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "logLevel": 3,
    }
  ]
}
```

and by default the `mode` should be `auto`.

brianignacio5, 2022-09-09:
The debugPort is the port used for the debug adapter as specified in [debugging documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md)

sbithinx, 2022-09-15:
Sorry, but this explanation of `debugPort` is just its name in a longer sentence. It just describes the same as the name itself. I mean witch port is this for openOCD? Could you please give some more technical detail about it in relationship to openOCD?

This is the output of starting the openOCD Server via VSCode:
```
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220411 (2022-04-11-08:47)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
Info : tcl server disabled
[OpenOCD]
Info : Listening on port 4444 for telnet connections
[OpenOCD]
Info : J-Link V11 compiled Jul 22 2022 10:21:23
Info : Hardware version: 11.00
[OpenOCD]
Info : VTarget = 3.322 V
[OpenOCD]
Info : clock speed 5000 kHz
[OpenOCD]
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : starting gdb server for esp32s3.cpu0 on 3333
[OpenOCD]
Info : Listening on port 3333 for gdb connections
```

```
[Debug Adapter]
2022-09-15 10:15:08,362 - Debug Adapter (main) - INFO - Working directory: /home/sebastian
[Debug Adapter]
2022-09-15 10:15:08,363 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 1572864,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 3,
    'developer_mode': 'none',
    'device_name': 'esp32',
    'elfpath': (   '/home/sebastian/workspace/******/build/******-esp32-pd-debug.elf',),
    'log_file': '/home/sebastian/workspace/******/debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32-elf-'}
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2022-09-15 10:15:08,363 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2022-09-15 10:15:08,363 - Debug Adapter (main) - INFO - Starting. Cmd: /home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter_main.py -d 3 -e /home/sebastian/workspace/******/build/******-esp32-pd-debug.elf -l /home/sebastian/workspace/******/debug.log -p 43474 -dn esp32 -a 0x180000 -t xtensa-esp32-elf-

[Debug Adapter]
2022-09-15 10:15:08,366 - Debug Adapter (main) - INFO - Listening on port 43474
DEBUG_ADAPTER_READY2CONNECT
2022-09-15 10:15:08,366 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[Debug Adapter]
2022-09-15 10:15:08,436 - Gdb - INFO - NOTIFY: thread-group-added {'id': 'i1'}
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333
[... 116 lines cut ...]
[Debug Adapter]
2022-09-15 10:15:23,747 - Debug Adapter (main) - ERROR - Failed to connect to "remote 127.0.0.1:3333"!
Traceback (most recent call last):
  File "/home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/command_processor.py", line 144, in on_launch_request
    self.da.run(start=(not self.da.state.no_debug))
  File "/home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 822, in run
    self._gdb.exec_run()
  File "/home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/hw_specific/esp.py", line 189, in exec_run
    self._update_memory_map()
  File "/home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/hw_specific/esp.py", line 167, in _update_memory_map
    self.connect()
  File "/home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/gdb.py", line 578, in connect
    self.target_select(remote_mode, self._remote_target, tmo=tmo)
  File "/home/sebastian/.vscode/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_backend/debug_backend/gdb.py", line 258, in target_select
    raise DebuggerError('Failed to connect to "%s %s"!' % (tgt_type, tgt_params))
debug_adapter.debug_backend.debug_backend.defs.DebuggerError: Failed to connect to "remote 127.0.0.1:3333"!
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2022-09-15 10:15:24,264 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
```

> Is config/esp32s3-jlink.cfg the same cfg you use in the terminal openOCD launch ?

Yes

brianignacio5, 2022-09-15:
The [esp-idf debug adapter](https://github.com/espressif/esp-debug-adapter) is sub module repository for this extension that uses the debug adapter protocol and serves as a middleman between visual studio code and OpenOCD <-> GDB. It is implemented as a server in python hence the need to specify its port. Ports for openOCD are defined with other extension settings.

I can see that for some reason the debug adapter is launching with target `esp32` instead of `esp32s3`. What is the value of `idf.adapterTargetName` in your settings.json ?

sbithinx, 2022-09-15:
Thx, for the explanation. I guess the [esp-idf debug adapter](https://github.com/espressif/esp-debug-adapter) is getting something wrong with it port configuration, as if I directly use OpenOCD and GDB on the command-line debugging works.

It is set to `esp32s3` in my `settings.json`:

`"idf.adapterTargetName": "esp32s3",`

brianignacio5, 2022-09-15:
Maybe is being overwritten in `/home/sebastian/workspace/eu-pd/.vscode/settings.json` ?

sbithinx, 2022-09-15:
It is only configured in the Workspace Settings `settings.json`. The User Settings `settings.json` does not have `idf.adapterTargetName` set at all.

sbithinx, 2022-09-19:
Here may workspace settings.json:

```
{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.adapterTargetName": "esp32s3",
  "idf.openOcdConfigs": [
    "config/esp32s3-jlink.cfg"
  ],
  "idf.port": "/dev/ttyUSB0",
  "files.associations": {
    "*.inc": "c",
    "*.ipp": "c",
    "array": "c",
    "deque": "c",
    "list": "c",
    "string": "c",
    "unordered_map": "c",
    "unordered_set": "c",
    "vector": "c",
    "string_view": "c",
    "initializer_list": "c",
    "regex": "c"
  }
}
```

And the `esp32s3.cfg`:
```
# Source the JTAG interface configuration file
source [find interface/jlink.cfg]
# Source the ESP32-S3 configuration file
source [find target/esp32s3.cfg]
adapter speed 5000
tcl_port disabled
```

brianignacio5, 2022-10-13:
I think the debug adapter seems to be working but one of your user interaction (maybe changing threads or something in your code) caused a gdb connection dropped. You could also return debug adapter log level to 2 to make things faster and avoid timeouts. Closing this issue since original issue seems to be fixed.

closed 2022-10-13 by brianignacio5