espressif/vscode-esp-idf-extension #800: ModuleNotFoundError: No module named 'setuptools.command.build'

xiaguangbo, 2022-09-12:
### OS

Windows

### Operating System version

win11

### Visual Studio Code version

1.72.0-insider

### ESP-IDF version

master

### Python version

3.8.7

### Doctor command output

no

### Extension

ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Command failed: "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe" -m pip install --upgrade --constraint C:\xiaguangbo\program\esp\.espressif\espidf.constraints.v5.1.txt --no-warn-script-location -r "c:\Users\16175\.vscode-insiders\extensions\espressif.esp-idf-extension-1.5.0\esp_debug_adapter\requirements.txt"
    ERROR: Command errored out with exit status 1:
     command: 'C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\Scripts\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\16175\\AppData\\Local\\Temp\\pip-install-6hhxp7jj\\xmlrunner_4cc69db8693b4fe5a522aabecf5e600f\\setup.py'"'"'; __file__='"'"'C:\\Users\\16175\\AppData\\Local\\Temp\\pip-install-6hhxp7jj\\xmlrunner_4cc69db8693b4fe5a522aabecf5e600f\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\16175\AppData\Local\Temp\pip-pip-egg-info-pf2v78oc'
         cwd: C:\Users\16175\AppData\Local\Temp\pip-install-6hhxp7jj\xmlrunner_4cc69db8693b4fe5a522aabecf5e600f\
    Complete output (40 lines):
    running egg_info
    creating C:\Users\16175\AppData\Local\Temp\pip-pip-egg-info-pf2v78oc\xmlrunner.egg-info
    writing C:\Users\16175\AppData\Local\Temp\pip-pip-egg-info-pf2v78oc\xmlrunner.egg-info\PKG-INFO
    writing dependency_links to C:\Users\16175\AppData\Local\Temp\pip-pip-egg-info-pf2v78oc\xmlrunner.egg-info\dependency_links.txt
    writing top-level names to C:\Users\16175\AppData\Local\Temp\pip-pip-egg-info-pf2v78oc\xmlrunner.egg-info\top_level.txt
    writing manifest file 'C:\Users\16175\AppData\Local\Temp\pip-pip-egg-info-pf2v78oc\xmlrunner.egg-info\SOURCES.txt'
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\16175\AppData\Local\Temp\pip-install-6hhxp7jj\xmlrunner_4cc69db8693b4fe5a522aabecf5e600f\setup.py", line 8, in <module>
        setup(
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\__init__.py", line 153, in setup
        return distutils.core.setup(**attrs)
      File "distutils\core.py", line 148, in setup
      File "distutils\dist.py", line 966, in run_commands
      File "distutils\dist.py", line 985, in run_command
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\command\egg_info.py", line 299, in run
        self.find_sources()
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\command\egg_info.py", line 306, in find_sources
        mm.run()
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\command\egg_info.py", line 541, in run
        self.add_defaults()
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\command\egg_info.py", line 577, in add_defaults
        sdist.add_defaults(self)
      File "distutils\command\sdist.py", line 226, in add_defaults
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\command\sdist.py", line 109, in _add_defaults_python
        build_py = self.get_finalized_command('build_py')
      File "distutils\cmd.py", line 299, in get_finalized_command
      File "distutils\cmd.py", line 107, in ensure_finalized
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\command\build_py.py", line 36, in finalize_options
        orig.build_py.finalize_options(self)
      File "distutils\command\build_py.py", line 43, in finalize_options
      File "distutils\cmd.py", line 286, in set_undefined_options
      File "distutils\dist.py", line 857, in get_command_obj
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\setuptools\dist.py", line 755, in get_command_class
        self.cmdclass[command] = cmdclass = ep.load()
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\pkg_resources\__init__.py", line 2450, in load
        return self.resolve()
      File "C:\xiaguangbo\program\esp\.espressif\python_env\idf5.1_py3.8_env\lib\site-packages\pkg_resources\__init__.py", line 2456, in resolve
        module = __import__(self.module_name, fromlist=['__name__'], level=0)
    ModuleNotFoundError: No module named 'setuptools.command.build'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

### Description

Use the plugin to install the development environment and that's it

### Debug Message

```plain
no
```

### Other Steps to Reproduce

Espressif IDF
v1.5.0

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-09-13:
Seems like an issue with your virtual environment setup and packages. Maybe try reinstalling the python_env ?

xiaguangbo, 2022-09-13:
@brianignacio5 Just delete ".espressif/python_env" and use the plugin again to install it, thank you

closed 2022-09-14 by brianignacio5