espressif/vscode-esp-idf-extension #802: Environment Variables don't work in "Custom Extra Paths"

Delzye, 2022-09-13:
### OS

Windows

### Operating System version

win10

### Visual Studio Code version

1.71.0

### ESP-IDF version

4.3.0

### Python version

3.8.7

### Doctor command output

should not be relevant?

### Extension

```
{"user":true,"message":"Invalid custom environment variables format","stack":"SyntaxError: Unexpected token \\ in JSON at position 1\n\tat JSON.parse (<anonymous>)\n\tat le (\\dist\\extension.js:1:14322)\n\tat m.<anonymous> (\\dist\\extension.js:1:1025185)\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:1:1021225\n\tat new Promise (<anonymous>)\n\tat f (\\dist\\extension.js:1:1020970)\n\tat m.configureWithDefaultValues (\\dist\\extension.js:1:1024728)\n\tat new m (\\dist\\extension.js:1:1021296)\n\tat m.init (\\dist\\extension.js:1:1021423)\n\tat \\dist\\extension.js:1:1205228\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:1:1204461\n\tat new Promise (<anonymous>)\n\tat De (\\dist\\extension.js:1:1204206)\n\tat ot (\\dist\\extension.js:1:1205095)\n\tat Tn._callActivateOptional (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:17649)\n\tat Tn._callActivate (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:17307)\n\tat \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:97:15118\n\tat process.processTicksAndRejections (node:internal/process/task_queues:96:5)\n\tat async E._activate (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:85:8224)\n\tat async E._waitForDepsThenActivate (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:85:8166)\n\tat async E._initialize (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:85:7530)","level":"error","timestamp":"2022-09-13T06:05:52.802Z"}
{"message":"Command failed: \"C:\\esp32\\toolchain\\esp-idf-tools\\tools\\idf-git\\2.30.1\\cmd\\git.exe\" describe --tags\nfatal: No names found, cannot describe anything.\n","stack":"Error: Command failed: \"C:\\esp32\\toolchain\\esp-idf-tools\\tools\\idf-git\\2.30.1\\cmd\\git.exe\" describe --tags\nfatal: No names found, cannot describe anything.\n\n\tat ChildProcess.exithandler (node:child_process:408:12)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat Socket.<anonymous> (node:internal/child_process:451:11)\n\tat Socket.emit (node:events:526:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2022-09-13T06:05:54.784Z"}
{"message":"Command failed: \"C:\\esp32\\toolchain\\esp-idf-tools\\tools\\idf-git\\2.30.1\\cmd\\git.exe\" describe --tags\nfatal: No names found, cannot describe anything.\n","stack":"Error: Command failed: \"C:\\esp32\\toolchain\\esp-idf-tools\\tools\\idf-git\\2.30.1\\cmd\\git.exe\" describe --tags\nfatal: No names found, cannot describe anything.\n\n\tat ChildProcess.exithandler (node:child_process:408:12)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat ChildProcess._handle.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2022-09-13T06:05:55.437Z"}
{"message":"Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n","stack":"Error: Open On-Chip Debugger  v0.10.0-esp32-20210401 (2021-04-01-15:46)\r\nLicensed under GNU GPL v2\r\nFor bug reports, read\r\n\thttp://openocd.org/doc/doxygen/bugs.html\r\n\n\tat \\dist\\extension.js:1:9233\n\tat ChildProcess.exithandler (node:child_process:399:7)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat Socket.<anonymous> (node:internal/child_process:451:11)\n\tat Socket.emit (node:events:526:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2022-09-13T06:05:55.770Z"}

```

### Description

**Problem**:
I want to create a "portable" installation of VSCode with the ESP-IDF extension. To do that, I need to use environment variables, so I can set the paths in the "Custom Extra Paths" to wherever the installation is currently mounted at.

**Expected Behavior**:
Adding a path (which contains an environment variable) should work just like I added it to PATH via Windows.

**Actual Behavior**:
Any executable, whose path contains an environment variable is not found.

**What have I tried?**:
I tried to set the environment variable in any way I know, even a row of stuff that should not work (using ${config:...}).
The normal way (%ENV%) does not work either. It does not make a difference, whether ENV was created by VSCode or manually by me in Windows. However, if I copy the paths from this setting (e.g. %ENV%/dfu-utils/...) into the PATH variable of Windows, it works. I'm not sure how paths from this setting are handled, but clearly they are not just appended to PATH, otherwise it should not make a difference, whether I append it manually or VSCode does it.

**Workarounds**:
Change the paths in this setting each time. Not really practicable.
Create a script, which sets an environment variable containing what I want to have in "Custom Extra Paths" and append that to PATH. This works, but since that setting already exists, it still seems unnecessarily complicated.

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2022-09-23:
Hi @Delzye,

In order to use env variables in the Custom Extra Paths you should use the following syntax: `${env:VARNAME}` such as `${env:HOME}` as explained in the  [settings documentation](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETTINGS.md)

Let me know if it helped

Delzye, 2022-09-23:
Hi @radurentea,
I'm pretty sure I tried that when I also tried ${config:...} but I will try again on Monday.

Delzye, 2022-10-09:
Sorry for the late answer, so using ${env:..} works, if I created the environment variable myself.
But if it is a VSCode-generated one, it does not work, even though both appear in the ENV directory.

Is there some difference between those that I do not understand?

radurentea, 2022-10-12:
The VSCode generated one should work as well, maybe it got do something with where you get your setting from (User or Workspace)?

Delzye, 2022-10-12:
I have all of these as user settings.

brianignacio5, 2022-10-14:
How are you defining the env variables in vscode @Delzye ? The ${env:VAR_NAME} uses env variables in your system but not vscode. There is also an extension setting called `idf.customExtraVars` for additional custom variables for our extension.

Delzye, 2022-10-19:
Yeah, I mixed up which variables are available at which points.

One last question, is it possible to use the IDF_PATH variable in CustomExtraVars?
Usually I would do that with {config:IDF_PATH}, but that does not work.

brianignacio5, 2022-10-21:
Well is kind of anti pattern no? Because we have the `idf.espIdfPath` configuration setting (which can also use {env: ENVVARNAME} or {config: SETTINGNAME} which is used for IDF_PATH in the extension tasks, seems better to define it in that setting instead of `idf.customExtraVars`

By the way, {config:SETTINGNAME} is to refer to settings name such as `idf.espIdfPath` not env variables, which is why we use {env:ENVVARNAME}.

closed 2022-10-21 by brianignacio5