espressif/vscode-esp-idf-extension #803: Cannot see Qemu icon, cannot run Qemu

JohnFromHere, 2022-09-14:
### OS

Windows

### Operating System version

10

### Visual Studio Code version

1.70.2

### ESP-IDF version

1.5.0

### Python version

3.8

### Doctor command output

launched this command via the command palette, but nothing seems to be happening. Where is the output supposed to display?...

### Extension

[image]

### Description

Everything is ok for creating the Qemu project following these guidelines : https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/QEMU.md (steps 1 to 3)
Step 5 (building) is ok as seen on the screenshot.
On the screenshot, you may also notice there is no icon for Qemu in the activity bar!

However, steps 4 and 6 don't seem to be available: 

4. The extension should be self configured, otherwise run the setup wizard.

=> how to run this wizard?

6. Use the Launch QEMU server command or the [QEMU Server] link in the activity bar to launch QEMU with the binaries from the build directory.

As I have nothing appearing there, I was wondering if there is a turnaround for this?

Also, do I have to install and run on windows a Qemu server on its own or is it integrated with Espressif extension esp-idf?

The same problem and display occur when I use the Espressif Qemu sample project.

Additional info: Docker server is run on windows before building, the remote connection in VS Code is open.

Any help would be greatly appreciated!

Thanks in advance,
John
[image]
[image]
[image]

### Debug Message

```plain
No error message, I simply can't launch Qemu functionalities.
When I try to call the command Qemu Manager from the command Palette, nothing happens, and then when trying the Qemu Monitor command, it displays in the notification zone : Qemu is not running. Run first.

But how to do it?...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-09-15:
HI @JohnFromHere the reason you can't see the QEMU Server in the blue bar at the bottom is because you have many extensions that contribute icons there and icons are hidden based on position. You can still run the commands `ESP-IDF: Launch QEMU server` and `Launch QEMU debug session` from the command palette (pressing F1 or menu View -> Command Palette).

JohnFromHere, 2022-09-15:
Hi Brian,

thank you for your quick answer.
However, I have already tried to acces Qemu options from the command Palette but nothing as `SP-IDF: Launch QEMU server` appears as seen on the attached screenshot here.
When I try the other commands, they display in the notification zone : Qemu is not running. Run first
[image]

brianignacio5, 2022-09-15:
Are you running the project in the container from the Dockerfile with the extension .devcontainer configuration ? 

Could you share the extension log and doctor command output ? Also check the ESP-IDF Output channel. It seems you are building the project with another extension.

JohnFromHere, 2022-09-15:
//Are you running the project in the container from the Dockerfile with the extension .devcontainer configuration ?
I think so, at least is what I think from the Explorer screenshot above. Is there another way to be sure of that?

Could you share the extension log and doctor command output ?
There is nothing in the extension log from the date I built this project.
When I run the doctor command, it seems nothing is happening, at least In the Terminal, Debug Console or Output consoles. Is there another place to find the doctor ouput?

//Also check the ESP-IDF Output channel. It seems you are building the project with another extension.

In the Output console, I enclose here all the info from the build process:
main] Building folder: workspaces 
[main] Configuring folder: workspaces 
[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_C_COMPILER:FILEPATH=/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -S/workspaces -B/workspaces/build -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
[cmake] -- Building ESP-IDF components for target esp32
[cmake] -- Project sdkconfig file /workspaces/sdkconfig
[cmake] -- Looking for sys/types.h
[cmake] -- Looking for sys/types.h - found
[cmake] -- Looking for stdint.h
[cmake] -- Looking for stdint.h - found
[cmake] -- Looking for stddef.h
[cmake] -- Looking for stddef.h - found
[cmake] -- Check size of time_t
[cmake] -- Check size of time_t - done
[cmake] -- Found Python3: /opt/esp/python_env/idf5.1_py3.8_env/bin/python3 (found version "3.8.10") found components: Interpreter 
[cmake] -- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
[cmake] -- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
[cmake] -- App "Qemu_HelloWorld" version: 1
[cmake] -- Adding linker script /workspaces/build/esp-idf/esp_system/ld/memory.ld
[cmake] -- Adding linker script /opt/esp/idf/components/esp_system/ld/esp32/sections.ld.in
[cmake] -- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.ld
[cmake] -- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
[cmake] -- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
[cmake] -- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
[cmake] -- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
[cmake] -- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
[cmake] -- Adding linker script /opt/esp/idf/components/soc/esp32/ld/esp32.peripherals.ld
[cmake] -- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
[cmake] -- Component paths: /opt/esp/idf/components/app_trace /opt/esp/idf/components/app_update /opt/esp/idf/components/bootloader /opt/esp/idf/components/bootloader_support /opt/esp/idf/components/bt /opt/esp/idf/components/cmock /opt/esp/idf/components/console /opt/esp/idf/components/cxx /opt/esp/idf/components/driver /opt/esp/idf/components/efuse /opt/esp/idf/components/esp-tls /opt/esp/idf/components/esp_adc /opt/esp/idf/components/esp_app_format /opt/esp/idf/components/esp_common /opt/esp/idf/components/esp_eth /opt/esp/idf/components/esp_event /opt/esp/idf/components/esp_gdbstub /opt/esp/idf/components/esp_hid /opt/esp/idf/components/esp_http_client /opt/esp/idf/components/esp_http_server /opt/esp/idf/components/esp_https_ota /opt/esp/idf/components/esp_https_server /opt/esp/idf/components/esp_hw_support /opt/esp/idf/components/esp_lcd /opt/esp/idf/components/esp_local_ctrl /opt/esp/idf/components/esp_netif /opt/esp/idf/components/esp_phy /opt/esp/idf/components/esp_pm /opt/esp/idf/components/esp_psram /opt/esp/idf/components/esp_ringbuf /opt/esp/idf/components/esp_rom /opt/esp/idf/components/esp_serial_slave_link /opt/esp/idf/components/esp_system /opt/esp/idf/components/esp_timer /opt/esp/idf/components/esp_wifi /opt/esp/idf/components/espcoredump /opt/esp/idf/components/esptool_py /opt/esp/idf/components/fatfs /opt/esp/idf/components/freertos /opt/esp/idf/components/hal /opt/esp/idf/components/heap /opt/esp/idf/components/http_parser /opt/esp/idf/components/idf_test /opt/esp/idf/components/ieee802154 /opt/esp/idf/components/json /opt/esp/idf/components/log /opt/esp/idf/components/lwip /workspaces/main /opt/esp/idf/components/mbedtls /opt/esp/idf/components/mqtt /opt/esp/idf/components/newlib /opt/esp/idf/components/nvs_flash /opt/esp/idf/components/openthread /opt/esp/idf/component [... cut ...]
[build] Successfully created esp32 image.
[build] Generated /workspaces/build/Qemu_HelloWorld.bin
[build] [839/839 100% :: 127.634] cd /workspaces/build/esp-idf/esptool_py && python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /workspaces/build/partition_table/partition-table.bin /workspaces/build/Qemu_HelloWorld.bin
[build] Qemu_HelloWorld.bin binary size 0x2adc0 bytes. Smallest app partition is 0x100000 bytes. 0xd5240 bytes (83%) free.
[build] Build finished with exit code 0

JohnFromHere, 2022-09-15:
When I try to manually perform the flashing, I run into this problem (Qemu server still not running):

`root@b66b1ba15500:/workspaces/build# qemu-system-xtensa -nographic -machine esp32 -drive file=Qemu_HelloWorld.bin,if=mtd,format=raw
Adding SPI flash device
qemu-system-xtensa: failed to read the initial flash content`

brianignacio5, 2022-09-15:
The doctor command output is in the clipboard. If you press Paste in any editor it will print there.

The output tab has a dropdown with different options, you are showing the `CMake/Build` which is given by another extension. Choose `ESP-IDF` from this dropdown. Also the build task you are using is not from our extension.

Take a look at [troubleshooting](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) to see how to get the information we need to find out the issue.

JohnFromHere, 2022-09-15:
Output: Correct, I didn't know that, I am new to VS Code!
Here is the Output display:
Error: spawn /opt/esp/python_env/idf5.0_py3.8_env/bin/python ENOENT
	at Process.ChildProcess._handle.onexit (node:internal/child_process:282:19)
	at onErrorNT (node:internal/child_process:477:16)
	at processTicksAndRejections (node:internal/process/task_queues:83:21)

I checked this folder and the actual Python there is idf5.1 => how and where can I change this parameter to the correct folder name?

_______________________________________________________________________________________

Here is the output of the doctor command:

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.10.16.3-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.1_py3.8_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/e4503b30fc78200f846c62cf8091b76ff5547662/bin/remote-cli:/opt/esp/idf/components/esptool_py/esptool:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin:/opt/esp/python_env/idf5.1_py3.8_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.70.2 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.0 
Workspace folder /workspaces 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\johns\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C:\Users\johns\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\johns\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\johns\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C:\Users\johns\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:\Users\johns\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:\Users\johns\.espressif\tools\cmake\3.23.1\bin;C:\Users\johns\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin;C:\Users\johns\.espressif\tools\ninja\1.10.2;C:\Users\johns\.espressif\tools\idf-exe\1.0.3;C:\Users\johns\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\johns\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\johns\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) board/esp32-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C: false
Access to \Users\johns\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;C: false
Access to \Users\johns\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;C: false
Access to \Users\johns\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;C: false
Access to \Users\johns\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;C: false
Access to \Users\johns\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C: false
Access to \Users\johns\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C: false
Access to \Users\johns\.espressif\tools\cmake\3.23.1\bin;C: false
Access to \Users\johns\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220411\openocd-esp32\bin;C: false
Access to \Users\johns\.espressif\tools\ninja\1.10.2;C: false
Access to \Users\johns\.espressif\tools\idf-exe\1.0.3;C: false
Access to \Users\johns\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C: false
Access to \Users\johns\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: false
[... cut ...]
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Command failed: /opt/esp/python_env/idf5.0_py3.8_env/bin/python -m pip list --format json
/bin/sh: 1: /opt/esp/python_env/idf5.0_py3.8_env/bin/python: not found

--------------------------------------------------------------------------------------------------------------------------------------------

Thanks again for your patience and help!

brianignacio5, 2022-09-16:
It is defined by default in the `<projectDirectory>/.devcontainer/devcontainer.json`. Just update the `idf.pythonBinPath` to `/opt/esp/python_env/idf5.1_py3.8_env/bin/python`. We Will also update it in the extension templates. Tried using ${env: IDF_PYTHON_ENV_PATH} there but I guess is not supported in dev containers yet.

brianignacio5, 2022-09-19:
If you rebuilt the container with the `Remote-Containers: Rebuild and reopen in Container` with this devcontainer.json file updated then there must be `idf.pythonBinPath` in your container `settings.json` that is overriding this value.

If you run the `Preferences: Open User settings (JSON)` or `Preferences: Open Workspace settings (JSON)` or in your current project ".vscode/settings.json" it might have some value that overrides the devcontainer.json values.

brianignacio5, 2022-09-26:
Closing this issue for lack of response. Comment here if you still have any issue.

closed 2022-09-26 by brianignacio5

JohnFromHere, 2022-09-26:
Sorry for my lack of response, I was away from my office for more than a week. I am not sure to have fully understood how I have to do the last changes you mentioned, however I will try my best to do that! My problem is still unsolved for the time being but thanks anyway!

brianignacio5, 2022-09-27:
I've meant to run the `Remote-Containers: Rebuild and reopen in Container` to rebuild the container so it gets the new idf.pythonBinPath or set the new path in your project `.vscode/settings.json`