espressif/vscode-esp-idf-extension #807: I can't install vscode esp idf extension because of python error.

soulk87, 2022-09-22:
### OS

Windows

### Operating System version

Window 10

### Visual Studio Code version

1.71.2(user setup)

### ESP-IDF version

4.4.3

### Python version

3.8.10

### Doctor command output

[image]

### Description

I can't install esp-idf extension. Please check My log below...

### Debug Message

```plain
Warning File C:\esp\espressif\tools\idf-python\3.8.7\vcruntime140_1.dll
                                        already exists and was not updated.
Warning File C:\esp\espressif\tools\idf-python\3.8.7\winsound.pyd
                                        already exists and was not updated.
Extracted C:\esp\espressif\tools\idf-python\3.8.7 ...
C:\esp\espressif\tools\idf-python\3.8.7\python.exe is not valid. (ERROR_INVALID_PYTHON)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

radurentea, 2022-09-23:
Hi @soulk87,

If you have followed the [Installation tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) your IDF_TOOLS_PATH should be `C:\Users\<YOUR WINDOWS USER>\.espressif`.

If you didn't follow the tutorial installation I recommend you do it, because you might not have installed the Pre-Requisites and this might be another reason for your error. Since you are using Windows, I recommend you installing them by using the [ESP-IDF Tools Installers](https://dl.espressif.com/dl/esp-idf/?idf=4.4)

soulk87, 2022-09-23:
Hi @radurentea ,
Thank you for your answer~!
Actually I already installed ESP_IDF before installing the ESP_IDF extension.
ESP_IDF could not be started properly due to an error at the time. (Please check the attached image)
[image]
I couldn't start with ESP_IDF, so I was looking for another way to build an esp32 project and it was the idf extension.
But still I can't install the ESP_IDF...

radurentea, 2022-09-23:
I think the easiest way to fix the installation is to uninstall ESP-IDF and install it via the [ESP-IDF Tools Installer](https://dl.espressif.com/dl/esp-idf/?idf=4.4)

brianignacio5, 2022-09-27:
Hi @soulk87 

Could you share log file which can be obtained from:

* **Windows**: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
* **Linux & MacOSX**: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

soulk87, 2022-10-05:
Hi @brianignacio5 
Thank you for your response.
I think my pc(window) has some problem with ESP-IDF.
Still I can't install ESP-IDF(not with extension).
I will ask about this situation at the another repository.
After check about ESP-IDF will come here again.

closed 2022-10-05 by soulk87