espressif/vscode-esp-idf-extension #811: ESP32 JTAG programming fails without power cycle

littleboot, 2022-09-27:
### OS

Windows

### Operating System version

10

### Visual Studio Code version

1.71.2

### ESP-IDF version

4.4.2

### Python version

3.8

### Doctor command output

none

### Description

After a clean install of VScode and the IDF plugin I have set-up the plugin top use the latest stable IDF version as of writing 4.4.2.
I open the blink example and connect the esp-prog to my device.
I needed to replace the USB drivers with zadig to make openOcd work at all, the latest instruction from espressif are incorrect and point to the FTDI drivers (this took me a few hours to find out, before I remembered the driver had to be replaced):
https://docs.espressif.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html#step-by-step-instruction

I used the following instruction to replace the drivers and make the ESP-prog work at all:
https://community.platformio.org/t/esp32-pio-unified-debugger/4541/20

Now I program the device, all is oke.
Now I program the device again, this time programming fails
now I power cycle the board
Programming will work again, and I program the blink example again
but after programming and monitoring the device it is stuck in a bootloop
[image]
If I perform a power cycle the application is started correctly and the serial output is correct.

This is not the first time I have had problems programming using the JTAG interface, this is really I mean really frustrating that after following the instruction by the letter and using a stable release I'm not able to get basic IDE functionality to work.
The failure behavior is almost the same as my last "fixed" bug report: https://github.com/espressif/openocd-esp32/issues/223

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-09-28:
I have a few questions @littleboot 

Are you using this esp-idf extension ?

Could you share the troubleshooting information described in the issue template or the readme ?

If the issue is fixed in openOCD, are you using the same version in vscode ?

littleboot, 2022-09-28:
Hi Brian, Thank you for your quick response.
I was really frustrated when I posted the issue and did not put in the effort to get all the required details (I was fighting with VScode + IDF extension for the whole day checking my hardware ect, today I'm refreshed and will have a go at it again and gather and provide more details).
I'm using the latest IDF VScode extension available within VScode marktplace, used this to download the 4.4.2 framework and tools. openOCD version 0.11.0 I think. replaced FTDI driversusing zadig (please add this information to the docs). setup example project "blink" using the extension. set project to use JTAG for programming and next I experienced the problem as described in the issue report.

I just did a clean windows 11 install on a second machine and will try to get it to work or reproduce the problem and gather the required details and update the issue report above.

littleboot, 2022-09-28:
Oke I tested the VScode + IDF extension on a new machine with a clean windows 11 install. And this will fail.
I experience exactly the same problem.
The hardware is oke, I do not have any issues in eclipse + IDF plugin.

These are the steps I took:

1. Installed Vscode
2. Installed IDF extension and 4.4.2 framework and tools
3. Installed USB drivers for USB2Serial chip CH340 (custom board)
4. Installed WinUSB drivers for ESP-prog Interface 0 and interface 1
5. Reboot PC (just in case)
6. Open Vscode, and configure example project Blink (save project on desktop)
7. Set JTAG as flash method
8. Build project
9. Flash using JTAG, Succes see output file 01
10. Start idf_monitor and observe "watchdog timeout flash read error" 
11. Power cycle board, still same monitor output "watchdog timeout flash read error" see output file 02
12. Try programming again, this failes again and OpenOCD gets stuck at ""programming started** and later fails see output file 03
[01 - first time jtag prog output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/9664235/01.-.first.time.jtag.prog.output.txt)
[02 - Monitor output after first programming action.txt](https://github.com/espressif/vscode-esp-idf-extension/files/9664236/02.-.Monitor.output.after.first.programming.action.txt)
[03 - JTAG program fail second try.txt](https://github.com/espressif/vscode-esp-idf-extension/files/9664237/03.-.JTAG.program.fail.second.try.txt)
[screenshots of steps i took.zip](https://github.com/espressif/vscode-esp-idf-extension/files/9664244/screenshots.of.steps.i.took.zip)

brianignacio5, 2022-10-14:
I don't think this issue has much to do with the extension but with the serial read itself. Issues arise on idf_monitor and then OpenOCD fails to program after. Are you using the latest openOCD version in the extension with the fixes you requested before ? You can check with the `ESP-IDF: Doctor Command`

If the extension is configured properly and the idf_monitor and openOCD start working then the extension side should be ok. Seems the fail happens in the program itself not the extension.

littleboot, 2022-10-15:
> I don't think this issue has much to do with the extension but with the serial read itself. Issues arise on idf_monitor and then OpenOCD fails to program after. Are you using the latest openOCD [...]

I can't reproduce the problem anymore, please remove this issue thanks for the help. Don't know what fixed the problem (I think the default JTAG speed of 20KHz which I lowered to 10KHz was the problem, but I'm not sure though), currently using the Eclipse IDE for my projects in combination with VSCode, because the editor is so much better sadly I need the debugging functionality and SVD register viewer of the Eclipse IDE.

closed 2022-10-18 by brianignacio5