espressif/vscode-esp-idf-extension #812: Can't use ESP-IDF Debug Adapter debuging for esp32-s3

wishinlife, 2022-10-01:
### OS

Linux

### Operating System version

windows11+wsl2+ubuntu20.04+vscode

### Visual Studio Code version

1.71.2

### ESP-IDF version

4.4.2

### Python version

3.8.10

### Doctor command output

None.

### Description

1、Can't use ESP-IDF Debug Adapter debuging for esp32-s3, microsoft c/c++ extension debug is no problem.
2、Set device target , select esp32s3, only ESP-PROG and ESP USB Bridge config, select any one, openocd start failed,, if manual modify esp32s3-bridge.cfg to esp32s3-builtin.cfg in file settings.json , openocd can start normally.

**launch.json:**
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "ESP_Debug",
      "request": "launch",
      "debugPort": 43474,
      "logLevel": 2,
      "mode": "auto", // 模板的manual,不能自动启动 Debug Adapter
      "verifyAppBinBeforeDebug": false,
      "tmoScaleFactor": 1,
      "initGdbCommands": [
        "target extended-remote :3333",
        "symbol-file ${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
        "mon reset halt",
        "flushregs",
        "flushregs",
        "thb app_main",
        "c"
      ],
      "env": {
        "PATH": "${env:PATH}:${config:idf.customExtraPaths}"
      }
    }
  ]
}

**settings.json:**
{
  "C_Cpp.intelliSenseEngine": "Tag Parser",
  "idf.port": "/dev/ttyACM0",
  "idf.adapterTargetName": "esp32s3",
  "idf.openOcdConfigs": [
    "board/esp32s3-builtin.cfg"
  ],
  "idf.flashType": "UART"
}

### Debug Message

```plain
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2022-10-01 22:16:11,083 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2022-10-01 22:16:11,083 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333
[OpenOCD]
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
[OpenOCD]
Info : rejected 'gdb' connection, no more connections allowed
[OpenOCD]
Info : dropped 'gdb' connection
[Debug Adapter]
2022-10-01 22:16:17,062 - Debug Adapter (main) - ERROR - int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Traceback (most recent call last):
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/command_processor.py", line 144, in on_launch_request
    self.da.run(start=(not self.da.state.no_debug))
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 825, in run
    self.on_target_stopped(rsn)
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 672, in on_target_stopped
    self.update_threads()
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 733, in update_threads
    self.thread_selected = int(gdb_resp[0])
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2022-10-01 22:16:17,122 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

meltdown03, 2022-10-02:
I only have the choices: ESP-PROG and USB Bridge mode for the esp32-s3. I've even built the extension from source code. Does it still use the `/espIdf/openOcd/boardConfiguration.ts' file to set the OpenOCD config files manually? I edited this file to set it to the `esp32s3-builtin.cfg` file but it still doesn't give me the option, the the ESP-PROG and Bridge modes.

meltdown03, 2022-10-02:
I see what it's doing now, it's using the `esp-config.json` file in the the openOCD's scripts folder. The file was actually updated to include the builtin.cfg file 12 days ago. https://github.com/espressif/openocd-esp32/commit/610883f6520edf1d67478342f29d75531d93372c

wishinlife, 2022-10-02:
> I see what it's doing now, it's using the file in the the openOCD's scripts folder. The file was actually updated to include the builtin.cfg file 12 days ago. [espressif/openocd-esp32@610883f](https://github.com/espressif/openocd-esp32/commit/610883f6520edf1d67478342f29d75531d93372c) EDIT: It seems it used that file because I had the options set. When I removed these from my settings, I got the choices from the file instead.`esp-config.json``idf.customExtraPaths/Vars``boardConfiguration.ts`

Thanks, i manual modify the file, add _esp32s3-builtin.cfg_ , openocd start success.

radurentea, 2022-10-04:
Hi @wishinlife,

I'm glad you managed to find a fix for it. As you've mentioned, the fix was added to the openOCD but in order to make it work you would have to use the master version of ESP-IDF.

wishinlife, 2022-10-05:
Hi @radurentea,

ESP-IDF Debug Adapter debuging for esp32-s3 is not working properly. Is there a solution?

Thanks.

> [Debug Adapter]
DEBUG_ADAPTER_STARTED
2022-10-01 22:16:11,083 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2022-10-01 22:16:11,083 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333
[OpenOCD]
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
[OpenOCD]
Info : rejected 'gdb' connection, no more connections allowed
[OpenOCD]
Info : dropped 'gdb' connection
[Debug Adapter]
2022-10-01 22:16:17,062 - Debug Adapter (main) - ERROR - int() argument must be a string, a bytes-like object or a number, not 'NoneType'
Traceback (most recent call last):
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/command_processor.py", line 144, in on_launch_request
    self.da.run(start=(not self.da.state.no_debug))
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 825, in run
    self.on_target_stopped(rsn)
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 672, in on_target_stopped
    self.update_threads()
  File "/home/zhuxy/.vscode-server/extensions/espressif.esp-idf-extension-1.5.0/esp_debug_adapter/debug_adapter/debug_adapter.py", line 733, in update_threads
    self.thread_selected = int(gdb_resp[0])
TypeError: int() argument must be a string, a bytes-like object or a number, not 'NoneType'
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2022-10-01 22:16:17,122 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter

brianignacio5, 2022-10-11:
Based on your error @wishinlife seems the board was still connected to GDB before. Make sure to close GDB OpenOCD and restart the board before trying to debug again.

brianignacio5, 2022-10-25:
Closing this issue since the original issue is fixed.

closed 2022-10-25 by brianignacio5