espressif/vscode-esp-idf-extension #813: Extension isn't working for esp-idf commit 02605f1 (latest master)

moefear85, 2022-10-04:
### OS

Linux

### Operating System version

Ubuntu 22.10

### Visual Studio Code version

1.71.2

### ESP-IDF version

5.0b1

### Python version

3.10.6

### Doctor command output

[doctor.txt](https://github.com/espressif/vscode-esp-idf-extension/files/9706467/doctor.txt)

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/9706470/esp_idf_vsc_ext.log)

### Description

The commit c51d4ab1428119e751f43c5d956b6426199187bc was working. It broke at d08898aabc8a62a9e77b06956a1a749d216d521e (in esp-idf), and still is in 02605f1a31be4233b97e83a0460bb458f277dfea.

Every time I launch esp-idf, I see this in the output:
> Error checking esp-rom-elfs version
> TypeError [ERR_INVALID_ARG_VALUE]: The argument 'file' cannot be empty. Received ''

The extension won't configure. Every time I click on "configure esp-idf extension", I get:
> First, install the [ESP-IDF Prerequisites for Linux](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html),

> restart Visual Studio Code and run this wizard again.

### Debug Message

```plain
not relevant
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

meltdown03, 2022-10-08:
I get the same thing on my system. I have all the prerequisites installed, actually I am already using the extension without issue but just wanted to reconfigure it. I also see the error about the `esp-rom-elfs`

radurentea, 2022-10-12:
Hi @moefear85 and @meltdown03,

I have created a [PR](https://github.com/espressif/vscode-esp-idf-extension/pull/821) for fixing the esp-rom-elfs download error, which will fix the configuration windows as well. However, you will have to make sure to uninstall the extension and delete the esp-idf folder, just to make sure all previous settings are deleted.

moefear85, 2022-10-24:
@radurentea thx for helping. I'll wait until it gets merged.

moefear85, 2023-02-12:
Hi... it's been over 4 months since I've opened this issue. It is still not possible to build a project in VSCode using esp-idf latest master. Is there a timeframe for the release of an official update to the extension so that this is possible? I'm stuck on commit c51d4ab because of this.

I'd gladly try latest master of the extension from github, but I can't build from source for various reasons.