espressif/vscode-esp-idf-extension #822: Unable to select /dev/ttyCH343USB0

lituo666, 2022-10-15:
### OS

Linux

### Operating System version

Ubuntu 22.04 server

### Visual Studio Code version

1.72.2

### ESP-IDF version

4.4.2

### Python version

3.8.7

### Doctor command output

none

### Description

The onboard USB to serial port chip is WCH CH343. After adding the driver of https://github.com/WCHSoftGroup/ch343ser_linux , the recognized device name is /dev/ttyCH343USB0, but this serial port cannot be selected in the vscode idf extension.

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lituo666, 2022-10-26:
The device "/dev/ttyCH343USB0" is not displayed after using the "ESP-IDF Select port to use" command.

I'm sure I have this device

```
# ls /dev/ttyCH343USB0 -al
crw-rw---- 1 root dialout 170, 0 Oct 26 19:03 /dev/ttyCH343USB0
```

"Doctor Command" command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.0-50-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /home/lituo/.vscode-server/bin/d045a5eda657f4d7b676dedbfa7aab8207f8a075/bin/remote-cli:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.72.2 
Visual Studio Code language zh-cn 
Visual Studio Code shell /usr/bin/zsh 
ESP-IDF Extension version 1.5.0 
Workspace folder /home/lituo/esp32c3_lvgl/esp32c3_lvgl_GC9A01 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /home/lituo/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) /home/lituo/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/lituo/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/lituo/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/lituo/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/lituo/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/lituo/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/lituo/.espressif/tools/cmake/3.23.1/bin:/home/lituo/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/lituo/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/home/lituo/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /home/lituo/.espressif/python_env/idf4.4_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-ftdi.cfg
ESP-IDF Tools Path (idf.toolsPath) /home/lituo/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to /home/lituo/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to /home/lituo/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to /home/lituo/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to /home/lituo/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to /home/lituo/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to /home/lituo/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to /home/lituo/.espressif/tools/cmake/3.23.1/bin: true
Access to /home/lituo/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin: true
Access to /home/lituo/.espressif/tools/ninja/1.10.2: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in /home/lituo/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: false
Spaces in /home/lituo/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: false
Spaces in /home/lituo/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: false
Spaces in /home/lituo/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: false
Spaces in /home/lituo/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: false
Spaces in /home/lituo/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: false
Spaces in /home/lituo/.espressif/tools/cmake/3.23.1/bin: false
Spaces in /home/lituo/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin: false
Spaces in /home/lituo/.espressif/tools/ninja/1.10.2: false
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unexpected token o in JSON at position 2
--------------------------------------------------------------------------------------------------------------------------------------------
```

brianignacio5, 2022-11-01:
It seems the drivers you are using are not supported in standard Python serial package. We recommend you to try `cdc-acm` or make sure your Linux Kernel is up to date (5.10+)

closed 2022-11-02 by lituo666