espressif/vscode-esp-idf-extension #823: Cannot setup VSCode on Ubuntu with latest ESP-IDF

dizcza, 2022-10-16:
### OS

Linux

### Operating System version

Ubuntu 22.04.1 LTS

### Visual Studio Code version

  latest/stable:    d045a5ed 2022-10-13 (111) 238MB classic (Snap)

### ESP-IDF version

ESP-IDF v5.1-dev-1146-g1308ff3d21

### Python version

3.9.12

### Doctor command output

It shows nothing in the output. That's the problem.

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/9794016/esp_idf_vsc_ext.log)

### Description

I'm new to VSCode and I'm trying to switch from Eclipse.
I've been using Eclipse ESP-IDF plugin for a year, the reason I'm switching to VSCode is that I hope it's supported better and in particular it should resolve code navigation which is broken in Eclipse plugin (I've fired several issues there - seems no way out to fix).
But after I followed the instructions on how to install ESP-IDF plugin in VSCode I feel that the story is much the same here.
The latest error in the error log I'm sharing is
```
{"message":"esp-rom-elfs doesn't have a compatible version for linux-amd64","stack":"Error: esp-rom-elfs doesn't have a compatible version for linux-amd64\n\tat h.getVersionToUse (/dist/extension.js:1:122712)\n\tat /dist/extension.js:1:125003\n\tat Array.map (<anonymous>)\n\tat h.<anonymous> (/dist/extension.js:1:124843)\n\tat Generator.next (<anonymous>)\n\tat s (/dist/extension.js:1:120567)","level":"error","timestamp":"2022-10-16T09:13:27.880Z"}
{"user":true,"level":"warn","message":"Extension configuration report has been copied to clipboard with errors","timestamp":"2022-10-16T09:13:27.880Z"}
```
and I have no idea how to fix it.

### Debug Message

```plain
No example has been launched yet.
```

### Other Steps to Reproduce

Here is how I set up VSCode.

A0. Installed ESP-IDF manually in `~/tools/esp-idf/` in a separate conda env `esp`.
1. Installed the ESP-IDF plugin.
2. Clicked Advanced settings.
3. Set up the path to the local folder `~/tools/esp-idf/` which contains a week-old master branch.
4. Installed.
5. I don't remember if there was an option to choose the python path while installing. I'm using a miniconda env `/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python -> python3` which leads to `~/miniconda3/envs/esp/bin/python` when `esp` env is activated. Now I cannot check that! (See next)
6. A screen popped out with "Everything has been successfully installed and you can close this window". Doubt that since no output was shown in the Output window. Perhaps, an error occurred which is shown in the log file only. Anyway, I closed it.
7. Opened Esspressif examples and clicked "hello_world".
8. Chose a common root directory for all projects - it's `~/Projects/Embedded/vscode-workspace/`
9. Clicked Build.
10. Nothing happened.
11. Clicked menuconfig
12. An error was shown. Cannot paste it here (see next).
13. Uninstalled ESP-IDF plugin.
14. Installed ESP-IDF plugin again.
15. Now I cannot even configure it. It shows me
> ESP-IDF Extension for Visual Studio Code
Welcome.
First, install the [ESP-IDF Prerequisites for Linux](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-setup.html),
restart Visual Studio Code and run this wizard again.

which is a totally wrong message since (1) I've already installed all prerequisites and (2) had I not installed them, I wouldn't be able to install the plugin in the first place!
What a mess!

It's so easily done in Eclipse and I had no problems whatsoever installing ESP-IDF there using local path and conda env.

----

I have checked existing issues, online documentation and the Troubleshooting Guide. I did check. But I mean how you search for "VSCode ESP-IDF plugin installation completed with errors it didn't show and then it refused to install it for the second time"?

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-10-18:
This is a duplicate of #813

dizcza, 2022-10-18:
Ah great. I became a lucky guy who run into trouble just from a start.

Looks like a duplicate indeed. Feel free to close the issue then.

dizcza, 2022-11-03:
9 days passed since the issue is fixed and users are getting the same issues on some of their systems, and you still didn't publish a hotfix release, and you don't describe how to build VScode extension manually on official documentation (probably for a good reason).

Please make a (minor) release.

kolipakakondal, 2022-11-03:
Hi @dizcza 

You can pick the latest build commit from GitHub and pick the VSIX to get the latest changes of the extension.

1. Download .vsix https://github.com/espressif/vscode-esp-idf-extension/suites/9063183188/artifacts/419243791
2. Go to VSCode > Extensions View > Click on `Views and More Actions` > Select `Install from VSIX`
3.  Select the downloaded .VSIX file and you're done

[image]

dizcza, 2022-11-03:
@kolipakakondal I'll give it a try.

Thanks for the info.

brianignacio5, 2022-11-04:
Also @dizcza there is documentation for manually build the vsix here:

https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/INSTALL.md

The reason we haven't made a minor release yet it is because we are testing a few features and bug fixes and making sure it works before we introduce more bugs :(

dizcza, 2022-11-04:
Although VScode installed  the Espressif extension from the file you linked above,
I cannot build the hello world project from examples. Upon its build configuration I get

```
[variant] Loaded new set of variants
[kit] Successfully loaded 1 kits from /home/dizcza/.local/share/CMakeTools/cmake-tools-kits.json
[main] Configuring folder: hello_world 
[proc] Executing command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -S/home/dizcza/Projects/Embedded/vscode-workspace/hello_world -B/home/dizcza/Projects/Embedded/vscode-workspace/hello_world/build -G Ninja
[cmake] CMake Error at CMakeLists.txt:5 (include):
[cmake]   include could not find requested file:
[cmake] 
[cmake]     /tools/cmake/project.cmake
[cmake] 
[cmake] 
[cmake] Not searching for unused variables given on the command line.
[cmake] -- The C compiler identification is GNU 11.3.0
[cmake] -- The CXX compiler identification is GNU 11.3.0
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: /usr/bin/cc - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: /usr/bin/c++ - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] -- Configuring incomplete, errors occurred!
[cmake] See also "/home/dizcza/Projects/Embedded/vscode-workspace/hello_world/build/CMakeFiles/CMakeOutput.log".
[proc] The command: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_BUILD_TYPE:STRING=Debug -S/home/dizcza/Projects/Embedded/vscode-workspace/hello_world -B/home/dizcza/Projects/Embedded/vscode-workspace/hello_world/build -G Ninja exited with code: 1 and signal: null
```

<details>

<summary>CmakeOutput.log</summary>

```
The system is: Linux - 5.15.0-52-generic - x86_64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/dizcza/Projects/Embedded/vscode-workspace/hello_world/build/CMakeFiles/3.22.1/CompilerIdC/a.out"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/dizcza/Projects/Embedded/vscode-workspace/hello_world/build/CMakeFiles/3.22.1/CompilerIdCXX/a.out"

Detecting C compiler ABI info compiled with the following output:
Change Dir: /home/dizcza/Projects/Embedded/vscode-workspace/hello_world/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/ninja cmTC_1f575 && [1/2] Building C object CMakeFiles/cmTC_1f575.dir/CMakeCCompilerABI.c.o
Using built-in specs.
COLLECT_GCC=/usr/bin/cc
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
[... 361 lines cut ...]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lc] ==> lib [c]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [/usr/lib/gcc/x86_64-linux-gnu/11/crtendS.o] ==> obj [/usr/lib/gcc/x86_64-linux-gnu/11/crtendS.o]
    arg [/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crtn.o] ==> obj [/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crtn.o]
  collapse obj [/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/Scrt1.o] ==> [/usr/lib/x86_64-linux-gnu/Scrt1.o]
  collapse obj [/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crti.o] ==> [/usr/lib/x86_64-linux-gnu/crti.o]
  collapse obj [/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu/crtn.o] ==> [/usr/lib/x86_64-linux-gnu/crtn.o]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/11] ==> [/usr/lib/gcc/x86_64-linux-gnu/11]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/11/../../../x86_64-linux-gnu] ==> [/usr/lib/x86_64-linux-gnu]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/11/../../../../lib] ==> [/usr/lib]
  collapse library dir [/lib/x86_64-linux-gnu] ==> [/lib/x86_64-linux-gnu]
  collapse library dir [/lib/../lib] ==> [/lib]
  collapse library dir [/usr/lib/x86_64-linux-gnu] ==> [/usr/lib/x86_64-linux-gnu]
  collapse library dir [/usr/lib/../lib] ==> [/usr/lib]
  collapse library dir [/usr/lib/gcc/x86_64-linux-gnu/11/../../..] ==> [/usr/lib]
  implicit libs: [stdc++;m;gcc_s;gcc;c;gcc_s;gcc]
  implicit objs: [/usr/lib/x86_64-linux-gnu/Scrt1.o;/usr/lib/x86_64-linux-gnu/crti.o;/usr/lib/gcc/x86_64-linux-gnu/11/crtbeginS.o;/usr/lib/gcc/x86_64-linux-gnu/11/crtendS.o;/usr/lib/x86_64-linux-gnu/crtn.o]
  implicit dirs: [/usr/lib/gcc/x86_64-linux-gnu/11;/usr/lib/x86_64-linux-gnu;/usr/lib;/lib/x86_64-linux-gnu;/lib]
  implicit fwks: []
```

</details>

dizcza, 2022-11-04:
Yeah, I might have clicked to configure as some Cmake Tools extension (don't recall if it was Microsoft) because VSCode popped up with a window suggesting so and I might have given in to it. I didn't notice a catch because I'm partially confused with the overwhelming use of "C/C++ VSCode Extension" in your docs here and there, and so I thought this was something I needed to do in order to set everything up.

Other things I'm confused in:
* First the user should select an Espressif target (esp32, esp32s2, etc.) with the ESP-IDF: Set Espressif device target command. Default is esp32 and the one used in this tutorial. --> I don't have any 'esp32' in the list. When I enter this command, I get a list with only one choice being the name of the current project
[image]
I chose that.
* Clicking `ESP-IDF: SDK Configuration editor (menuconfig)` yields
```
[SDK Configuration Editor]
---------------------------ERROR--------------------------

Please use idf.py only in an ESP-IDF shell environment.

-----------------------END OF ERROR-----------------------
[SDK Configuration Editor]
---------------------------ERROR--------------------------

SDK Configuration editor confserver process exited with code: 1
-----------------------END OF ERROR-----------------------
```
* I configured `ESP-IDF: SDK Configuration editor` to point to my local `esp-idf` directory `~/tools/esp-idf` and conda env python path `/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python` that I'm using specifically to build all ESP-IDF projects I have and not to clash with other Python environments. I'm using the latest master branch of ESP-IDF there and have had no problems building a project in a terminal.
* But now I cannot open `ESP-IDF: SDK Configuration editor` window again. The console shows me
```
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
```
and the config window does not pop up. I expect to have the ESP-IDF SDK config window pop up each time I click this option.

As you see, I'm using:
* not fully tested Espressif extension installed from a commit artifact
* local paths to ESP-IDF with conda env
* latest (officially not supported by you) version of ESP-IDF
* and I'm new to VSCode (though I have a great deal of coding experience in other IDEs).

And the problem could happen in any (or more than one!) of these deviations from a basic how-to you have on your documentation page.

Perhaps, the best way for me to proceed is to wait for the tested release of the VSCode extension.

brianignacio5, 2022-11-04:
So it seems you haven't configured the extension yet. Before the basic use tutorial, there is a [install](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) tutorial to install the ESP-IDF (or use existing one) and that's why SDK Configuration Editor is not working.

If you look at the doctor command output as requested for issue template and troubleshooting information (output is copied in the clipboard you can paste anywhere)

dizcza, 2022-11-12:
Sorry for the late response.

So I managed to build and flash an ESP-IDF hello world project in VSCode. I did two important things, and none of them taken apart would solve the issue I had.

1. As you remember, I install ESP-IDF in a conda env `~/miniconda3/envs/esp/`. When activated, `which python` shows

```
(esp) dizcza@ubu:~$ which python
/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python
(esp) dizcza@ubu:~$ ll /home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python
lrwxrwxrwx 1 dizcza dizcza 7 жов 11 09:46 /home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python -> python3*
```

So it points to `python3` not `/home/dizcza/miniconda3/envs/esp/bin/python3.9`. I've installed ESP-IDF via terminal. This is the first issue. To fix that, I needed to provide `/home/dizcza/miniconda3/envs/esp/bin/python3.9` (and not `/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python` which I had been doing before in VSCode and had the issue I'm describing, had been doing in Eclipse plugin and had no issues) to "Enter absolute python executable path to use" menu in the VSCode ESP-IDF configuration. I clicked Install and it *did* run a few steps downloading and installing some binary files ESP-IDF needs (before, in previous attempts installing the VSCode extension, it just ended up with an "Everything is set up!" notification - it's very confusing since before that the extension was NOT configured and installed at all).

Still, it did not solve the problem completely.

2. When I hit Build the project, I got

```
CMake Error at /home/dizcza/tools/esp-idf/tools/cmake/project.cmake:342 (__project):
  The CMAKE_C_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.
```

Which made me think that `. export.sh` is not called in your VSCode extension. Why?

Anyway, I fixed that by manually comparing the $PATH env activated (inside the `esp` conda env) and deactivated (outside of `esp` conda env). I added the following to IDF custom extra paths in the VSCode extension settings.

```
/home/dizcza/tools/esp-idf/components/esptool_py/esptool:/home/dizcza/tools/esp-idf/components/espcoredump:/home/dizcza/tools/esp-idf/components/partition_table:/home/dizcza/tools/esp-idf/components/app_update:/home/dizcza/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/home/dizcza/.espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:/home/dizcza/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/dizcza/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/dizcza/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/dizcza/.espressif/tools/xtensa-clang/14.0.0-38679f0333/xtensa-esp32-elf-clang/bin:/home/dizcza/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/dizcza/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/dizcza/.espressif/tools/cmake/3.24.0/bin:/home/dizcza/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/dizcza/.espressif/tools/ninja/1.10.2/:/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin:/home/dizcza/tools/esp-idf/tools
```

With these two modifications, finally, it worked. Interestingly, installing ESP-IDF extension in VSCode changed the link to
```
$ ll /home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python
lrwxrwxrwx 1 dizcza dizcza 46 лис 12 21:35 /home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python -> /home/dizcza/miniconda3/envs/esp/bin/python3.9*
```

While I do understand why I had no issues compiling an ESP project in a terminal inside the conda env, I don't quite understand how is possible that Eclipse plugin works fine with `/home/dizcza/.espressif/python_env/idf5.1_py3.9_env/bin/python` pointing to just `python3` but it does not work for VSCode... and why `export.sh` is not getting activated in your extension.

dizcza, 2022-11-12:
> I added the following to IDF custom extra paths in the VSCode extension settings.
>
> ``` [...]

This is obviously a hotfix and will be broken as soon as I update the ESP-IDF.

brianignacio5, 2022-11-25:
The thing is that settings are saved at the end of the setup workflow. The vscode extension doesn't use `export.sh`. The extension will try to create a virtual env with the given python executable so it would fail if you are using an existing python env.

You could always use the manual configuration, The setup process is described in [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md)

What I want to understand is, do you want to use an existing setup ? or install a new one with the extension ? As far as I can see you are trying to use an existing one. If the extension is failing to complete the setup at any step, please share the troubleshooting output for that scenario. 

If you trying to use an existing setup, the question is what you already have installed and want to reuse. The vscode extension needs to install some python packages too (like gcovr and websocket) so an additional step is required. We probably need to make sure that the given python is one for an existing setup instead of create a new one.

dizcza, 2022-11-25:
I'm trying to reuse an existing setup:
* ESP-IDF master branch `/home/dizcza/tools/esp-idf`
* conda env `/home/dizcza/miniconda3/envs/esp`
This conda env is used to set up ESP-IDF by running `./install.sh`. I update the master branch periodically and then run `./install.sh esp32` again each time. Then I open Eclipse, choose Download and configure ESP-IDF -> from existing folder -> install all packages and select the conda's python path. I want to perform the same steps for VSCode without manually hacking the PATH envs (in VSCode extension) as I described in the previous message.

> You could always use the manual configuration, The setup process is described in [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md)

This is an option. I won't need to reupdate the settings.json each time I pull source code from ESP-IDF github and install it in the same conda env from a terminal, will I?