espressif/vscode-esp-idf-extension #824: Fails to install tools

wreyford, 2022-10-23:
Type: <b>Bug</b>

Uninstall ESP-IDF and tools. Then install the plugin, then use advanced tab, and use default directories, and select master branch. It itnsalls ESP-IDF, then requests to install tools, and immediately fails. I tried > 3 X and no go.

Extension version: 1.5.0
VS Code version: Code - Insiders 1.73.0-insider (504c5a768a001b2099dd2b44e9dc39e10ccdfb56, 2022-10-21T05:33:59.974Z)
OS version: Windows_NT x64 10.0.19044
Modes:
Sandboxed: Yes

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-7700HQ CPU @ 2.80GHz (8 x 2808)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: disabled_off<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_renderer: enabled_on<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled<br>webgpu: disabled_off|
|Load (avg)|undefined|
|Memory (System)|15.96GB (9.43GB free)|
|Process Argv|--crash-reporter-id 3a3ec103-e30b-4a15-9fd1-4559b5bdbb5c|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv695:30137379
vsins829:30139715
vsliv368:30146709
vsreu685:30147344
python383:30185418
vspor879:30202332
vspor708:30202333
vspor363:30204092
vslsvsres303:30308271
pythonvspyl392:30422396
pythontb:30258533
pythonptprofiler:30281269
vshan820:30294714
pythondataviewer:30285072
vscod805cf:30301675
bridge0708:30335490
bridge0723:30353136
cmake_vspar411:30581797
vsaa593cf:30376535
pythonvs932:30404738
cppdebug:30492333
vscaac:30438845
vsclangdf:30492506
c4g48928:30535728
dsvsc012cf:30540253
pylantcb52:30590116
pyind367:30587847
vscccc:30564267
pyindex848:30577858
nodejswelcome1:30587009

```

</details>

brianignacio5, 2022-10-24:
Please complete the required troubleshooting information in the issue template so we can understand what is happening in your setup.

wreyford, 2022-10-24:
@brianignacio5 ,
I'm sorry, I filled out all the info required in the VSCode Insiders, when submitting. Perhaps I missed something
I'll try here:
On my system the following steps reproduced the problem:
Setup:
Extension version: 1.5.0
VS Code version: Code - Insiders 1.73.0-insider (504c5a768a001b2099dd2b44e9dc39e10ccdfb56, 2022-10-21T05:33:59.974Z)
OS version: Windows_NT x64 10.0.19044
Modes:
Sandboxed: Yes

Process to replicate.
Delete ESP-IDF ,and the folders where they are installed. Uninstall the Espressif Extension.

Then reinstall Extension.: Espressif IDF v1.5.0 within VSCode Insiders.
Then Press F1 -> ESP-IDF: Configure ESP-IDF Extension

If you are lucky, it will come up with the ability to select from the Advanced install, and configure.
Most times it just says to restart VSCode.
[image]

It is exactly this issue that prompted me to delete and start all over.

When and If I ever get it to display the proper Install screen, and I select advanced, the ESP-IDF installs, then it prompts you to install the tools, then starts and immediately fails. Sorry , but I don't have a screenshot of the terminal messages, as that would have helped a lot.

**You can close this issue**, as I after so many tries decide to try out the ESPRESSIF IDE, and after installing it I'm very happy working with it. So many features just work, and although it was quite a job to get up and running I can flash and compile without issues. Updating master also a breeze.

closed 2022-10-24 by wreyford

brianignacio5, 2022-10-24:
Oh I see. Apologies, was wondering why it could be failing. Could you share the extension log ?

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

wreyford, 2022-10-24:
Sure, but has no entries for yesterday, as I deleted the extension, but still failing today ...
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/9850053/esp_idf_vsc_ext.log)

brianignacio5, 2022-10-24:
It seems to be related to #813 Which it has a PR to fix it soon!

brianignacio5, 2022-10-26:
Should be fixed in master with #821