espressif/vscode-esp-idf-extension #826: [Feature Request]:  Add ESP32S3 Builtin JTAG to known boards list

Panometric, 2022-10-26:
**Is your feature request related to a problem? Please describe.**
As a new user of the ESP32-box, I found it very difficult to get the debugger running. 

**Describe the solution you'd like**
When using the extension to configure openocd, the esp32s3 built-in variant should be on the list. 

**Describe alternatives you've considered**
Selected setting for ESP32C3 which is very similar, and then edited the settings manually. 

**Additional context**
Mainstream product should be in this list:
[image]

brianignacio5, 2022-10-27:
The boards listed in the extension are obtained from `esp-config.json` from OpenOCD. The issue is fixed in OpenOCD master as discussed in #812. When you are using the extension you can manually define the values with custom board or use the openOCD version that includes the built-in esp32 S3 config files value.

Panometric, 2022-10-27:
I have the capability to use openocd form the command line using `openocd -f board/esp32s3-builtin.cfg` . The problem is just with that the extension does showing not show it, per the screenshot above from 
extension v1.5.0. 

I understand I can get around it, but it was difficult to figure out.

brianignacio5, 2022-11-01:
The issue should be fixed with the [OpenOCD v0.11.0-esp32-20221026](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026). 

As described in the documentation, you can also manually define the `idf.openOcdConfigs` configuration setting to `board/esp32s3-builtin.cfg`.

I'll make the option for manual input with the `Custom board`

PR #830 (2022-11-07): add input for select custom openocd board
## Description

In the `ESP-IDF: Select OpenOCD Board Configuration` allow the user to manually input the configuration files for the `Custom Board` option. 

Also add the `ESP32-S3 debugging via builtin USB-JTAG` option if not available in `esp-config.json` temporary since it is only included in the latest OpenOCD release.

Fix #826

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

- Manual Testing using the UI command 

**Test Configuration**:
* ESP-IDF Version: 4.4.2
* OS (Windows,Linux and macOS [... cut]
closed 2022-11-07 by brianignacio5