espressif/vscode-esp-idf-extension #827: [Feature Request]:  Create automatic Launch.json settings for

Panometric, 2022-10-26:
**Is your feature request related to a problem? Please describe.**
Getting the idf and vscode platform to run a debugger on a fresh install is one of the most frustrating I have ever used. 
Among other things, having to paste some text into launch.json that is only shown in [Debugging.md](https://github.com/espressif/vscode-esp-idf-extension/blob/81cc28e0f90dc3aadeeb0060f88e06583e380622/docs/DEBUGGING.md) was difficult to find. 
Most other platforms will suggest working values that work without additional interaction.  It's great that the espidf type is supported and the parameters are known, but a new user can't easily find that. 

**Describe the solution you'd like**
Ideally, then the extension configures for a project, or the device is selected, the launch.json should be populated if empty.  And the Add configuration button on launch,json should show ESP IDF, and add the minimal values when selected.  

**Describe alternatives you've considered**
There don't seem to be any easy button.  you must find it  in the docs and paste it in manually. 

**Additional context**
This is the initial dialog on a freshly imported project:
[image]
Manually pressing add gets this far, but no esp-idf settings are in this list. 
[image]

Selecting espidf should insert this (renamed for clarity):
```
{
    "version": "0.2.0",
    "configurations": [
        {
            "type": "espidf",
            "name": "ESP Debug",
            "request": "launch",
        }
    ]
}
```

brianignacio5, 2022-10-27:
Isn't the `ESP-IDF: Add vscode configuration folder` doing what you are requesting ?

These files are added when you create a project with the `ESP-IDF: Show Examples projects` and `ESP-IDF: New Project`

When you click create a debug launch.json file:
[image]

I'll add the fix for the IntelliSense autocomplete in #825

Panometric, 2022-10-27:
Yes,  `ESP-IDF: Add vscode configuration folder` would work. I did not know it was there. 

I think my problem was that they aren't if you are opening an exiting project, like examples from https://github.com/espressif/esp-box/tree/master/examples 

If you try one of these, when the esp-idf configures when opening a project folder for the first time, it does create settings.json,  but it does not create launch.json. 

If it did `ESP-IDF: Add vscode configuration folder` so that the configuration was complete, it would work as expected.

PR #825 (2022-11-01): fix empty workspace folder issues
## Description

Please include a summary of the change and which issue is fixed.

Fixes using the doctor command when no workspace folder is opened. 
Fix Python packages json match in doctor command
Open the doctor command result text file after finished.

Fix #827

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing running the `ESP-IDF Doctor Command`. Delete `launch.json` and create from debug activity bar. Also tested VSCode autocomplete.

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS  [... cut]
closed 2022-11-01 by brianignacio5