espressif/vscode-esp-idf-extension #832: [Feature Request]: make is easier to use multiple ESP-IDF versions side by side

chipweinberger, 2022-11-03:
[image]

When setting up ESP-IDF extension, It *overwrites* the existing settings.

I have 2 different apps, one on v4.4.2 and one on v5.0.

It would be great if there were 2 choice during setup
- Setup ESP-IDF defaults for all apps
- Setup ESP-IDF only for this app

chipweinberger, 2022-11-03:
On top of that, you should probably also have the ESP-IDF version in the bottom VSC toolbar.

Just a simple "[espressif logo] v5.0-0x213409" in the toolbar.

brianignacio5, 2022-11-04:
While the current UI is not optimize for this (thanks for your feature request we will try to work on this). You can indeed use multiple versions of ESP-IDF (one for each project) by definition of settings in each workspace folder as described in [working with multiple projects](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md)

On your current open project, set `idf.saveScope` in your `<project-dir>/.vscode/settings.json` to `WorkspaceFolder` with the `ESP-IDF: Select where to save configuration settings`. After that use the setup wizard you shared before. Settings will be set in the current workspace folder.

Close first project, open second project, repeat as previously shown.

chipweinberger, 2022-11-20:
Thanks for the reply! That would be great.

Just to clarify, I have 2 separate ESP-IDF folders / repos on my system.

I suppose I could just have one, and use git checkout, but I think that would probably be more annoying when both projects are open at the same time so I use two.

Hopefully that use case can be covered.

brianignacio5, 2022-11-25:
Well, as I described earlier, you can indeed use a specific ESP-IDF per project right now.

The missing part would be to have an UI to reflect that.

chipweinberger, 2022-12-11:
_Going Stale:_ I suppose I will bump this, since it is still a feature request I'd like to see

chipweinberger, 2023-01-23:
bump. The UI proposed for this would be a great feature. I just came back to this thread to read your instructions again! =)

chipweinberger, 2023-01-23:
Couple Ideas:
- "per project" settings i.e. `workspaceFolder` should be the default
- "Use Existing" should show all available options

Possible implementation of Use Existing:
- there should be a new `idf.availableIdfConfigs` 
- `Use Existing Setup` should show`idf.availableIdfConfigs`
- When Express & Advanced finish, they should append to `idf.availableIdfConfigs`

Screens
---
### Welcome
**Choose ESP-IDF For This Project**
- ☑️ v5.1 (**IDF**: `/desktop/esp-idf-v5` & **Tools**: `/documents/.esspressif/`) (Default)
- 🔲 v4.3 (**IDF**: `/desktop/esp-idf-v4.3`& **Tools**: `/documents/.esspressif/`)
- 🔲 v4.4 (**IDF**: `/desktop/esp-idf-v4.4`& **Tools**: `/documents/.esspressif/`)

🆕 Add New Setup (on the next screen show Express & Advanced)

[Submit Button]  ☑️ Update default settings

---

closed 2023-02-28 by brianignacio5

chipweinberger, 2023-02-28:
Wow thanks for implementing Brian!! I'm very excited to try it out!

PR #898 (2023-02-28): Select multiple IDF versions in setup wizard
## Description

Expand use existing setup to load existing ESP-IDF setups in the user systems.

Every time the user saves a new ESP-IDF path it will saved in the extension global state and appear later in the `Existing Setup` and also uses `IDF_TOOLS_PATH` to load `esp-idf.json` for existing ESP-IDF setups.

Fixes #832

## Type of change
- New feature (non-breaking change which adds functionality)

## How has this been tested?

Test changing multiple ESP-IDF versions using manual testing of wizard.

- Test A
- Test B

**Test Configuration**:
* ESP-I [... cut]