espressif/vscode-esp-idf-extension #836: IDF Installer fails with build tools requirements - IDF 5.0 release branch

sukesh-ak, 2022-11-08:
### OS

Windows

### Operating System version

11

### Visual Studio Code version

1.73

### ESP-IDF version

5.0

### Python version

3.8

### Doctor command output

Board is not relevant since its an installation issue.

### Extension

Failed to build psutil
  error: subprocess-exited-with-error
  
  Building wheel for psutil (pyproject.toml) did not run successfully.
  exit code: 1
  
  [38 lines of output]
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build\lib.win-amd64-cpython-38
  creating build\lib.win-amd64-cpython-38\psutil
  copying psutil\_common.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_compat.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_psaix.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_psbsd.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_pslinux.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_psosx.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_psposix.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_pssunos.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\_pswindows.py -> build\lib.win-amd64-cpython-38\psutil
  copying psutil\__init__.py -> build\lib.win-amd64-cpython-38\psutil
  creating build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\runner.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_aix.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_bsd.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_connections.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_contracts.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_linux.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_memleaks.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_misc.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_osx.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_posix.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_process.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_sunos.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_system.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_testutils.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_unicode.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\test_windows.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\__init__.py -> build\lib.win-amd64-cpython-38\psutil\tests
  copying psutil\tests\__main__.py -> build\lib.win-amd64-cpython-38\psutil\tests
  running build_ext
  building 'psutil._psutil_windows' extension
[... 113 lines cut ...]

If you open Built Tools installer there is a whole lot of things selected with some GB+ content. 
But for my test requirements I need only the bare minimum to run VSCODE+IDF+IDF Extension.

```
building 'psutil._psutil_windows' extension
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
```

### Debug Message

```plain
building 'psutil._psutil_windows' extension
  error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
```
```

### Other Steps to Reproduce

Windows 11 + Latest VS Code
I am using Windows Sandbox for installer testing right now. So the machine instance is clean without any other software.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

sukesh-ak, 2022-11-08:
Earlier the installation outside of the Extension was very straight forward.
https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/get-started/index.html#get-started-get-esp-idf

Was this requirement added recently?
Get it with `Microsoft C++ Build Tools`: https://visualstudio.microsoft.com/visual-cpp-build-tools/

brianignacio5, 2022-11-08:
It is specified in the [extension prerequisites](https://github.com/espressif/vscode-esp-idf-extension#prerequisites) but I think is a recent change within the psutil python package built for Windows.

sukesh-ak, 2022-11-08:
> It is specified in the [extension prerequisites](https://github.com/espressif/vscode-esp-idf-extension#prerequisites) but I think is a recent change within the psutil python package built for Windows.

How did it work before without explicitly installing the Build Tools?
Doc says For Windows the [C++ Build Tools](https://visualstudio.microsoft.com/visual-cpp-build-tools) `might be required`.

Most people have been reluctant to jump on ESP-IDF due to installer issues (People use PIO due to that). But with IDF 5.x & Component registry, these installation issues become a roadblock.

I made a video of IDF installer and compiling helloworld project few days ago to share (not uploaded yet). 
Fast forward today, everything is broken now and complicated.

noelcun, 2022-11-08:
I experienced this issue today on a fresh machine, even previous esp-idf versions such as 4.X are failing setup due to psutil python package failure exiting the installation

brianignacio5, 2022-11-08:
This is an issue with psutil directly, check here:

https://github.com/giampaolo/psutil/issues/2165#issuecomment-1306915687

sukesh-ak, 2022-11-08:
> It was working without it until recently @sukesh-ak This is a recent issue with `psutil` directly, check here:
> 
> [giampaolo/psutil#2165 (comment)](https://github.com/giampaolo/psutil/issues/2165#issuecomment-1306915687)

Thank you.
Does this need an update for the IDF VS Extension?

sukesh-ak, 2022-11-08:
I was able to install. Probably that fix worked.

closed 2022-11-08 by sukesh-ak

brianignacio5, 2022-11-09:
> > It was working without it until recently @sukesh-ak This is a recent issue with `psutil` directly, check here:
> > [giampaolo/psutil#2165 (comment)](https://github.com/giampaolo/psutil/issues/2165#issuecomment-1306915687)
> 
> Thank you. Does this need an update for the IDF VS Extension?

It shouldn't need any updates from our side. Glad to hear is working ok.