espressif/vscode-esp-idf-extension #837: CMake Error: No source directory specified for -S

xiaolongba, 2022-11-09:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.73.0

### ESP-IDF version

latest, master branch

### Python version

3.9

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19043 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 D:\Python39\Scripts\;D:\Python39\;%ALLEGRO_BRD2ODB%\nv\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;c:\Program Files (x86)\HP\HP Performance Advisor;D:\PuTTY\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;D:\Git\cmd;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;D:\CMake\bin;D:\doxygen\bin;D:\Mscgen;D:\stlink-1.3.0-win64\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\ProgramData\chocolatey\bin;D:\nodejs\;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Program Files\usbipd-win\;C:\Program Files (x86)\Nordic Semiconductor\nrf-command-line-tools\bin\;D:\Keil_v5\ARM\ARMCC\bin;D:\serialplot\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;D:\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Roaming\npm 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.73.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.1 
Workspace folder f:\BLE_WIFI\Espressif\SDK\advertising\non-connectable-scannale-undirected-adv 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) f:\BLE_WIFI\Espressif\SDK\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) F:\BLE_WIFI\Espressif\SDK\esp-idf\components\esptool_py\esptool;F:\BLE_WIFI\Espressif\SDK\esp-idf\components\app_update;F:\BLE_WIFI\Espressif\SDK\esp-idf\components\espcoredump;F:\BLE_WIFI\Espressif\SDK\esp-idf\components\partition_table;F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\cmake\3.24.0\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;F:\Tools\Espressif\IDF_Tools\.espressif\tools\ninja\1.10.2\;F:\Tools\Espressif\IDF_Tools\.espressif\tools\idf-exe\1.0.3\;F:\Tools\Espressif\IDF_Tools\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;F:\Tools\Espressif\IDF_Tools\.espressif\python_env\idf5.1_py3.9_env\Scripts;F:\BLE_WIFI\Espressif\SDK\esp-idf\tools;
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"F:/Tools/Espressif/IDF_Tools/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) F:\Tools\Espressif\IDF_Tools\.espressif\python_env\idf5.1_py3.9_env\Scripts\python.exe
Serial port (idf.port) COM35
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) f:\Tools\Espressif\IDF_Tools\.espressif
Git Path (idf.gitPath) ${env:programfiles}\Git\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to F:\BLE_WIFI\Espressif\SDK\esp-idf\components\esptool_py\esptool: true
Access to F:\BLE_WIFI\Espressif\SDK\esp-idf\components\app_update: true
Access to F:\BLE_WIFI\Espressif\SDK\esp-idf\components\espcoredump: true
Access to F:\BLE_WIFI\Espressif\SDK\esp-idf\components\partition_table: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\cmake\3.24.0\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\ninja\1.10.2\: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\idf-exe\1.0.3\: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to F:\Tools\Espressif\IDF_Tools\.espressif\python_env\idf5.1_py3.9_env\Scripts: true
Access to F:\BLE_WIFI\Espressif\SDK\esp-idf\tools: true
Access to : false
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
[... 41 lines cut ...]

NULL

### Description

I upgraded the extension from v1.5.0 to v1.5.1 yesterday, however it cannot build now and the error info as below:
```
CMake Error: No source directory specified for -S
CMake Error: Run 'cmake --help' for all supported options.
```
However, it works well after rolling back to v1.5.0. could you help me dobule-check what happened?

### Debug Message

```plain
NULL
```

### Other Steps to Reproduce

NULL

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-11-09:
One of the features we added was using the `idf.buildPath` to specify the build folder and the current workspace folder for the source directory (CMake -S parameter) in the build task, something like:

`-B idf.buildPath -S this.curWorkspace.fsPath`. Is ` f:\BLE_WIFI\Espressif\SDK\advertising\non-connectable-scannale-undirected-adv` the current workspace folder with ESP-IDF source code ?

Could you share the extension log for more information ?

xiaolongba, 2022-11-09:
> Is ` f:\BLE_WIFI\Espressif\SDK\advertising\non-connectable-scannale-undirected-adv` the current workspace folder with ESP-IDF source code ?

yes, this is my example project.

> Could you share the extension log for more information ?
[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/9968331/esp_idf_vsc_ext.log)

> Also what is the value in `idf.cmakeCompilerArgs`

[image]

brianignacio5, 2022-11-09:
What about the build task output? Can you see the arguments being called ? Should be an additional terminal `ESP-IDF Build Task`

xiaolongba, 2022-11-09:
> Can you see the arguments being called ?

yes, the details as below:
[image]

brianignacio5, 2022-11-09:
What is this? Why is it duplicating the entries of -B and -S here ? 😮

xiaolongba, 2022-11-09:
> What is this? Why is it duplicating the entries of -B and -S here ? 😮

I don't know, bro😂 I only upgraded the extension and then click the build button.

xiaolongba, 2022-11-10:
@brianignacio5 

Hello Bro,

I found the root and it is caused by the `-DCCACHE_ENABLE=1`

```
 *  The terminal process "C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B f:\BLE_WIFI\Espressif\SDK\advertising\non-scannable-non-connectable-directed-adv\build -S -DCCACHE_ENABLE=1 f:\BLE_WIFI\Espressif\SDK\advertising\non-scannable-non-connectable-directed-adv" terminated with exit code: 1. 
```

I unchecked the following option, as below:
[image]

it works well after that, so I think you need to modify the order of Cmake options parameter.

JunkCoding, 2022-11-10:
> 

Thanks. Solved for me.

IH303, 2022-11-19:
I have the same issue:
```
Task wird im Ordner node_module_project ausgeführt: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /home/ih303/Projekte/esp-module/node_module_project/build -S -DCCACHE_ENABLE=1 /home/ih303/Projekte/esp-module/node_module_project 

CMake Error: No source directory specified for -S
CMake Error: Run 'cmake --help' for all supported options.
```

**-DCCACHE_ENABLE=1** is inserted between **-S** and the source directory

I think @xiaolongba is right:
> it works well after that, so I think you need to modify the order of Cmake options parameter.

brianignacio5, 2022-11-25:
Could you PTAL at this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/450641412.zip) to see if the issue is fixed ?

Diegorro98, 2022-11-27:
@brianignacio5 I can confirm that the issue is fixed, I got the same error and no longer happens with that vsix and CCache enabled.

IH303, 2022-11-28:
@brianignacio5 I can also confirm that now everything is working fine. Thank you.

xiaolongba, 2022-11-28:
> Could you PTAL at this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/450641412.zip) to see if the issue is fixed ?

thanks, work well for me as well, pls close this ticket now

PR #848 (2022-11-29): fix args order build task
## Description

Check that build compiler arguments doesn't already exists before adding them to the list. Make sure to add the CCache argument at the end.

Fix #837 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual build test on Windows

**Test Configuration**:
* ESP-IDF Version: 5.1
* OS (Windows,Linux and macOS): Windows 

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all pla [... cut]
closed 2022-11-28 by xiaolongba