espressif/vscode-esp-idf-extension #838: v1.5.1

JeonghunLee, 2022-11-10:
### OS

Windows

### Operating System version

Window 11

### ESP-IDF version

1.73.0

### Python version

3.8.7

### Doctor command output

Sorry , 

### Description

ESP-IDF Monitor is not working properly in VS Code 
suddenly, it's not working after updated v1.5.1 

e.g.
set IDF_PATH=C:\Users\jhlee\esp\esp-idf
C:\Users\jhlee\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\jhlee\esp\esp-idf\tools\idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32-elf-  **--target**  esp32 d:\Works\Project\mod610_sw\build\mxx610-app.elf

so I removed --target in dist/extension.js 

"--toolchain-prefix",this.config.toolchainPrefix];this.config.wsPort&&n.push("--ws","ws://localhost:"+this.config.wsPort),n.push(this.config.elfFilePath);

### Debug Message

```plain
Sorry, now not used OpenOCD based on VSCode
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-11-10:
Which command are you running ? Which device are you using ?

Could you provide a bit more information  about what is the error being shown ?

JeonghunLee, 2022-11-10:
1. my device is esp32 
2. I used ESP-IDF monitor Button in VS Code 

it's not working properly because of --target argument 
```
PS D:\Works\Project\mod610_sw> set IDF_PATH=C:\Users\jhlee\esp\esp-idf
PS D:\Works\Project\mod610_sw> C:\Users\jhlee\.espressif\python_env\idf4.2_py3.8_env\Scripts\python.exe C:\Users\jhlee\esp\esp-idf\tools\idf_monitor.py -p COM5 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 d:\Works\Project\mod610_sw\build\mxx610-app.elf
usage: idf_monitor - a serial output monitor for esp-idf [-h] [--port PORT] [--baud BAUD] [--make MAKE] [--encrypted]
                                                         [--toolchain-prefix TOOLCHAIN_PREFIX] [--eol {CR,LF,CRLF}]
                                                         [--print_filter PRINT_FILTER] [--decode-coredumps {info,disable}]      
                                                         elf_file
idf_monitor - a serial output monitor for esp-idf: error: argument elf_file: can't open 'esp32': [Errno 2] No such file or directory: 'esp32'
```

JeonghunLee, 2022-11-10:
so resolved this issue by changing  C:\Users\jhlee\.vscode\extensions\espressif.esp-idf-extension-1.5.1\dist\extension.js
removed it below "n.push "--target" 
```
"--toolchain-prefix",this.config.toolchainPrefix];this.config.wsPort&&n.push("--ws","ws://localhost:"+this.config.wsPort),n.push(this.config.elfFilePath);
```

brianignacio5, 2022-11-28:
The issue was that this argument should only be used for ESP-IDF 4.3 or higher. Making a fix in #850

JeonghunLee, 2022-11-29:
I don't understand it, we are developing the products based on v4.2.3 now 

v4.2 ESP-IDF Maintenance period below 
    https://docs.espressif.com/projects/esp-idf/en/v4.4.3/esp32/versions.html#support-periods

Doesn't ESP-IDF v4.2.3 or Lower support anymore?

brianignacio5, 2022-11-29:
Yes we do. Was an bug in the extension itself. Should be fixed with the PR #850

JeonghunLee, 2022-11-29:
Ok I see , Thank you fast response

PR #850 (2022-11-29): fix monitor use target arg idf version
## Description

ESP-IDF version needs to be `>=4.3` to use the `--target` argument in `idf_monitor.py`

Fix #838

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing running the extension Monitor command.

**Test Configuration**:
* ESP-IDF Version: 4.2 and 4.3
* OS (Windows,Linux and macOS): macOS

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Wi [... cut]
closed 2022-11-29 by brianignacio5